cp2004/OctoPrint-WS281x_LED_Status #77: spidev.bufsize setting not picked up (maybe nfs related)

kevinvalk, 2021-01-08:
**Description of the bug**
I am running on a Raspberry Pi 3 Model B Plus Rev 1.3 over NFS (pxe network boot) and somehow the "spidev.bufsize" was not being picked up in the `/boot/cmdline.txt` file (I.e. `cat /sys/module/spidev/parameters/bufsiz` returns 4096).

Instead I created a file at `/etc/modprobe.d/spidev.conf`
```
options spidev bufsiz=32768
```
That did the trick and now `cat /sys/module/spidev/parameters/bufsiz` returns 32768.

However, the OS configuration test still shows "SPI Buffer size is not correct"

**What did you expect to happen**
After changing bufsize via modprobe options to 32768, the OS configuration test should show "SPI Buffer size correct"

**Version of plugin and OctoPrint**
WS281x LED Status (0.6.0)
OctoPrint 1.5.2
Python 3.7.3
OctoPi 0.17

cp2004, 2021-01-09:
I can add that, actually the output is way easier to parse since it is *literally* just the number. As this plugin has got more popular, the wizard is not as suitable for everything. In addition to #71 and #75, I'll have to re-work the wizard and OS configuration test (they share code) backend to make it a bit more universal.

Thanks pointing this one out!

kevinvalk, 2021-01-09:
Nice! Great work on the plugin btw :)

For completeness: removed "maybe nfs related" from title. Originally put that there because I am running this Pi using PXE (network boot) + NFS and I thought maybe that is the reason that `spidev.bufsize` was not being picked up from cmdline.txt.

cp2004, 2021-02-13:
As of the above commit, `/sys/modules/spidev/parameters/bufsiz` is used as a fallback to reading `/boot/cmdline.txt`. It will still check `/boot/cmdline.txt` first, and the 'fix this' button will add the parameter there if it can't find this setting in either file. Ready for 0.7.0 release.

cp2004, 2021-03-05:
Hey @kevinvalk, 0.7.0rc1 is [out now for testing](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/releases/tag/0.7.0rc1) - there's a [feedback ticket here](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/issues/119) for any comments on the release. Any feedback is appreciated.

closed 2021-03-11 by cp2004

climatec, 2021-11-09:
Hi 

I also battle With this now.

guanaco0403, 2022-01-22:
> Hello, I know it should be fixed by you, but i did fresh install v7,2 today and it gave me that error `SPI buffer size increased Error: The file this check expected does not exist ` Without any "fix it" button.
> 
> I even tried upgrate to Dev but nothing changed.
> 
> Edit: I'm sorry, my bad. Restarted whole Pi few times and then suddenly started working.

same please help

commit a987d97d (2021-02-11): :sparkles: Check sys modules for spidev.bufsiz

Closes #77

This implements a secondary check, if the setting is not found in the `/boot/cmdline.txt` file, checking `/sys/module/spidev/paramters/bufsiz`
commit 5c105b82 (2021-02-13): :sparkles: Make backend of OS config test smarter

It will now reply with a reason and what check it is in the status message.

Closes #77 and #75 and will solve some support overhead.