cp2004/OctoPrint-WS281x_LED_Status #78: Switch on Led‘s through SSH

Jack77777777, 2021-01-09:
**What's your question?**

I want to switch on LED's (all white) through command line (ssh). What would be the script, where to place and command?

cp2004, 2021-01-09:
The only option you have to turn it on/off via the CLI is over API commands, using OctoPrint's API. Here's an example of how to use the torch option:

It is all one line, to turn on the torch mode:
```
curl --header "Content-Type: application/json" --header "X-Api-Key: YOURAPIKEY" --request POST --data '{"command":  "activate_torch"}' http://127.0.0.1/api/plugin/ws281x_led_status
```
Make sure you get the API key from OctoPrint, and enter that in the command. It assumes it is running on the same device OctoPrint is on, substitute `127.0.0.1` for the IP of the Pi or `octopi.local` if you want.

To deactivate the torch, you can either use timed torch mode, or you can change the command to `deactivate_torch` instead of activate.

**Please note** that this API is currently undocumented, so it may change in future versions. I am intending to make some adjustments in the next version (0.7.0) that stabilise this, make it more organised and then document it. So it might break.

Jack77777777, 2021-01-10:
Hmmm, Torch is not working as expected. Sometimes the button works but most the click on the torch button has no funtion. Perhaps i configured something wrong.

Jack77777777, 2021-01-10:
Nevermind. Found it.
curl --header "Content-Type: application/json" --header "X-Api-Key: YOURAPIKEY" --request POST --data '{"command":  "toggle_lights"}' http://127.0.0.1/api/plugin/ws281x_led_status

Will test later to get it into system commands and/or custom commands and feedback here

cp2004, 2021-01-10:
Yep that is the one. I've tried to make as many options as possible, there's @ commands for all of them too [documented](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/wiki/Features#host--commands) but like you say some of them only work when the printer is connected.

Once you've confirmed it's all working, I may add it to the [Tips and tricks](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/wiki/Tips-and-Tricks) page. I want that page to be a collection of cool things people have made to work alongside this plugin. Maybe rename it and provide a contents to make them easier to find.

Jack77777777, 2021-01-11:
I don't know how to get following command into shell script. Can you helo me?

curl --header "Content-Type: application/json" --header "X-Api-Key: YOURAPIKEY" --request POST --data '{"command": "toggle_lights"}' http://127.0.0.1/api/plugin/ws281x_led_status

Jack77777777, 2021-01-12:
Created a shell and a python script. Both working from CLI but not through system drop down in octoprint. I wrote the full path in config.yaml but no luck. 

Python script:
#!/usr/bin/env python
import rpi_ws281x as ws281x_led_status
import requests

headers = {
    'Content-Type': 'application/json',
    'X-Api-Key': '253F6F28762E416D961386B8C8F7E93A',
}

data = '{"command": "toggle_lights"}'

response = requests.post('http://127.0.0.1/api/plugin/ws281x_led_status', headers=headers, data=data)

#Execute as /home/pi/oprint/bin/python /home/pi/scripts/camlight.py

config.yaml:
system:
    actions:
    -   action: LED toggle
        command: /home/pi/oprint/bin/python /home/pi/scripts/camlight.py
        confirm: null
        name: LED toggle

Jack77777777, 2021-01-12:
It's curl inside the script what does not work. Cretaed a simple gpio shell scrigt and could run through octoprint without problems. Is there anothee way except curl?

cp2004, 2021-01-12:
Any tool that allows you to send web requests would do, something like wget works as well (though I don't know the syntax).

I don't know why the curl command inside the bash script doesn't work, that Python one you have is probably failing due to the import of `rpi_ws281x` at the top, though you said you run it within the virtualenv. You could try making it executable, change the shebang to `#!/usr/bin/python3` or something and just passing a single `/home/pi/scripts/camlight.py`?

Jack77777777, 2021-01-12:
The python script is working through SSH(CLI). 

I had no errors. In Octoprint it does nothing, also no error message, bitbi have to reboot. The logs show no errors too.

cp2004, 2021-01-12:
I can't really help you beyond that for the custom scripts, maybe you could ask at the [community forums](https://community.octoprint.org) since I've not really used them. Though chances are high it will be me replying there, as soon as you mention WS281x LED Status they leave it to me...

cp2004, 2021-02-08:
@Jack77777777 can I close this now? Did you get it sorted?

Just as a heads-up - the API in the plugin **will be changing** with the next release. This breaking change had to happen as I wanted to make the API stable so that it can be depended on in the future.

It will be documented here afterwards: https://cp2004.gitbook.io/ws281x-led-status/documentation/rest-api

Jack77777777, 2021-02-08:
Hi @cp2004,

I couldn't find a way. Just close it.

cp2004, 2021-02-08:
That's unfortunate, if I knew more getting the system commands up and running I would help. The script works from command line, that's all I have.

Hope the rest of the plugin is working 🙂

closed 2021-02-08 by cp2004