cp2004/OctoPrint-WS281x_LED_Status #81: WS2812 flashing can't controll

ChrisYOLO, 2021-01-12:
the led strips is just flashing green i cant controll it. Its not like the effect its flashing on its own random led

cp2004, 2021-01-12:
Please check the ground connections are exactly as printed in the setup guide, with a very good common ground connection from the LEDs, power supply and Raspberry Pi.

cp2004, 2021-01-22:
Hey @ChrisYOLO,

If I don't hear back in the next week, I'll close this issue. Nothing personal, just trying to keep things tidy around here. Please let me know if you solve the issue so I can close this too. Thanks!

cp2004, 2021-01-30:
Since I haven't received a reply in 18 days, I'm closing this issue now. Please feel free to request a reopen if you are still having issues. Nothing personal, just trying to keep the amount of issues I have tidy and under control. Thanks.

closed 2021-01-30 by cp2004