cp2004/OctoPrint-WS281x_LED_Status #82: heating progress displayed while cooling

supremecool185, 2021-01-12:
im not sure if most printers do this but my printer has a cooling phase before the print is actually "completed" where the bed temperature target will drop to slightly above room temperature and the tool head target will go to 0, once the bed target is met the print completes and the bed target goes to 0 as well

when using both a heating progress and a cooling progress indicator during this cooling phase it will display the heating progress at 100%, once the cooling finishes it THEN displays the cooling progress at around 1% completion (which does not progress since its already at room temperature) before quickly transitioning to my print finished effect

disabling the heating progress will cause it to just show my printing progress instead of heating and does not affect the cooling progress. so there seem to be 2 different problems involved here:
1. heating progress is displayed when it is actually cooling
2. cooling progress is only displayed after the print has finished and it is already cooled

supremecool185, 2021-01-12:
i already have custom Gcode inserted when the cooling begins that turns on a fan, so if there is a way to insert a Gcode that tells the cooling progress to begin i could try that out

cp2004, 2021-01-12:
Please enable the serial.log for OctoPrint, reproduce the issue then upload this log so I can review. I need to know the specific temperatures/gcode that is run, I currently can't tell from that description what (if anything) needs to change or be adjusted for this.

Please note that heating reporting is triggered by a blocking heatup command, and lasts until the next gcode command is sent. Even if this blocking command is lower than the current temperature it should still work, the heating effect going backwards. Cooling progress is *only* triggered at the end of the print, after all gcode has run. This is the only way that I have found to reliably detect cooling where most people/slicers just use non-blocking heating commands to change the temperature - and it would be annoying if mid-print changes in temperature started the cooling effect.

supremecool185, 2021-01-12:
[serial.log](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/files/5803864/serial.log)

cp2004, 2021-01-12:
From the looks of it, your printer/slicer end gcode uses `M190` to cool the bed, and this sets the target temperature to 35. Calculating percentages as `current/target` will always produce over 100% in this instance. Cooldown tracking only functions at the end of the print, on the 'PrintDone' event. I couldn't find a good way to track cooldown outside this, since the same command is used to heat and cool.

Not sure there is a lot I can do - I will be tidying up this detection for the next release but don't intend to start creating more complicated detection of whether the printer is cooling or heating that is more sophisticated - the results are not worth it.

supremecool185, 2021-01-12:
alright thats fine i guess i just wont use the cooling tracker, but do you know why the heating tracker comes up during the cooling?

cp2004, 2021-01-12:
Because it uses the same gcode, `M190` as heating. Actually, I had forgotten that, it would be quite easy to not enter heating tracking if the target temperature is below the current temperature. Harder to implement cooling that way, but this is definitely something that could be changed.

supremecool185, 2021-01-12:
ok thanks for the help! is there any Gcode i can use to mimic the cooling tracker aesthetically by triggering a color wipe at the same point the fan turn on?

cp2004, 2021-01-12:
There's no gcode or other command, it is all based of the 'PrintDone' event.

cp2004, 2021-03-13:
Oops.... sorry about that one. I changed the title and the bot got upset.

cp2004, 2022-02-04:
0.8.0 has been released (finally...!)

closed 2022-02-04 by cp2004