cp2004/OctoPrint-WS281x_LED_Status #83: It does not work with SPI Elegoo 3.5 touchscreen

thomasgoehler, 2021-01-14:
I have set up the plugin according to your instructions. I also did a check of the parameters.SPI is enabled, user pi is added in the gpio group, bufsize is also set and so on. I am using 4 LED strips with 8 LEDs each(WS2812b) which I have connected together.  A short test with an Arduino Nano showed that it works. The strips are powered by an external 5V source, I used a Logiclevel converter. Next I tested the LED strip on the Pi using the ws28xx library and python. With GPIO 18 via PWM it worked. But not with GPIO10 (SPI). Here I found what I was looking for and got an error. SPI is accessed via /dev/spidev0.0, but this is already in use or blocked by my touchscreen. I then removed the touchscreen driver. Without touchscreen your plugin runs perfectly.
Now my question: Can you possibly adapt the script so that it can control the LED bar via PWM(GPIO18) or that one can change the SPI device to e.g. /dev/spidev1.0?

My configuration:
Pi 3b+ with actual Raspberry Pi OS
Octoprint 1.52 installed manually (as described in the official wiki)

cp2004, 2021-01-14:
You can change the GPIO pin to 18 if you wish, though you'll have to run OctoPrint as root which for security concerns is not recommended. See #63 or #64 for details.

There's a section in the troubleshooting guide (which you would have seen before deleting the template hopefully) about SPI touchscreens, and specific instructions where someone had this issue before.

thomasgoehler, 2021-01-14:
thank you for your advice. I will try to enable auxiliary SPI then it should be possible to run two SPI devices at the same time.  I hope I can do that. Otherwise I will get rid of the display and use one with HDMI connection. The touch functionality I don't really need and was nice to have. Your really awesome plugin should stay of course.

closed 2021-01-14 by thomasgoehler