cp2004/OctoPrint-WS281x_LED_Status #84: Install is not Working, how can I fix it?

IonDari, 2021-01-14:
```
E: Unable to locate package https://github.com/cp2004/OctoPrint-WS281x_LED_Status/archive
E: Couldn't find any package by glob 'https://github.com/cp2004/OctoPrint-WS281x_LED_Status/archive'
E: Couldn't find any package by regex 'https://github.com/cp2004/OctoPrint-WS281x_LED_Status/archive'
```
These are the error messages. Please help me to get it working or fix the issue.

cp2004, 2021-01-14:
The URL that you have used to install it is broken. Outline to me exactly what steps you used to get to this place.

IonDari, 2021-01-14:
i typed "sudo apt-get install https://github.com/cp2004/OctoPrint-WS281x_LED_Status/archive/master.zip" because withe the plugin manager it also wasnt working it keept loading and did nothing so i tried this memtmhodm and got me to this errors. thats the url that i found in your setup guide, none butt all of the urls i can find wouldnt work either. Thanks for the quick response.

cp2004, 2021-01-14:
It doesn't say to use `sudo apt-get install` anywhere on that guide, and that will not work. It needs to be installed from the plugin manager (find it in the list of available plugins) or enter that URL into the 'From URL' box within OctoPrint.

If the plugin manager is not working for you, then you should fix that rather than try random commands you think of - you could cause all sorts of problems continuing the way you are.

IonDari, 2021-01-14:
Ive made backups before but when I paste the link in the plugin manager also nothing happens, Ive tried it with 

- [ ] suda apt-get install 

to see if it can get the package but it clouldnt. Its the same with the plugin manager. I spent 4 hours trying to install it because it is exactly what i want.

cp2004, 2021-01-14:
Upload the `octoprint.log` and we will see why the plugin manager will not work.

IonDari, 2021-01-14:
ok just wait a moment

IonDari, 2021-01-14:
[octoprint.log](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/files/5816804/octoprint.log)
[image]

Nothing in the log and I tried it for another time. And yes, I let it run for almost half an hour and it still lokked the same when I came back.

IonDari, 2021-01-14:
It is also a clean OctoPrint installation.
And it has connection to the Internet and I tried to restart the Pi for several times.

cp2004, 2021-01-14:
That log you've uploaded seems to be nearly 10 days old, please try and find a newer one. It is dated 2021-01-05.

If that is the latest one, then I would suggest it doesn't have an internet connection since there is a lot of internet-related errors in the log.

In addition, it is reporting the version as 1.3.12 which is technically too old for this plugin, where the minimum OctoPrint version is 1.4.0. Please try updating OctoPrint to 1.5.2 to see if it fixes your issues.

IonDari, 2021-01-14:
yeah I see its strange because Ive setup octopi today and a completly formatted the sd card XD

IonDari, 2021-01-14:
[octoprint (5).log](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/files/5816926/octoprint.5.log)

Here is the new one

cp2004, 2021-01-14:
It appears that your OctoPrint instance doesn't have an internet connection. Yes, you managed to install plugins previously, but that is probably an issue. There is a bug (since fixed, not released) in OctoPrint where it doesn't handle HTTP errors properly.

If you fully believe that it can access the internet (hint: it can't, it says so in the log) then you can use the following command to do what the plugin manager does, outside it. Please note that if there is no internet connection, then I cannot help you further and that will need to be fixed first.

```
~/oprint/bin/pip install https://github.com/cp2004/OctoPrint-WS281x_LED_Status/releases/latest/download/release.zip
```

IonDari, 2021-01-14:
where du i need to execute the line and yes im sure that i have an internet connection because i can install other plugins before and after i tried yours

IonDari, 2021-01-14:
Ok thank you its installing now, the problem was that the pip version was outdatet, and thank you for your quick answers.

IonDari, 2021-01-14:
Thank you so much for helping me, the pip version was outdated and

closed 2021-01-14 by IonDari

IonDari, 2021-01-14:
I also have a hint for you:
2021-01-14 20:24:46,871 - octoprint.util.connectivity.connectivity_checker - INFO - Connectivity changed from offline to online
what's up with that line?!