cp2004/OctoPrint-WS281x_LED_Status #85: 6 Pin LED Strip

SegmentedAbyss5, 2021-01-15:
Im looking into using this for my enclosure and i have a 6 pin LED strip on hand.
Does this work with a 6 pin Strip (R,G,B,White,Warm white) ? And how would i go about wiring it ?

cp2004, 2021-01-15:
Hey @SegmentedAbyss5!

This plugin will only work with WS281x type LEDs (including WS2811, WS2812 and SK6182 strips, also branded as Neopixels). These LEDs are individually addressable with a single data line, and power and ground.

For non-addressable LED strips, you could look at the [LEDStripControl plugin](https://plugins.octoprint.org/plugins/ledstripcontrol/) or apparently the [OctoPrint Enclosure](https://plugins.octoprint.org/plugins/enclosure/) plugin might be able to do this.

SegmentedAbyss5, 2021-01-15:
Okay great, thanks for the reply !

closed 2021-01-15 by SegmentedAbyss5