cp2004/OctoPrint-WS281x_LED_Status #86: Led stays green

ingo916, 2021-01-19:
**What do you need help with?**
Installed your plugin and it lights up green. Passed the os config test. I can  change the color of the leds in the utility tab. 
however that’s all, bed warming, print progress etc. not working. 

Connected to gpio  10 and power wired as described in your instructions. 

**Log files**

Here are my log files:
* `octoprint.log`: WRITE HERE
* `plugin_ws281x_led_status_debug.log`: WRITE HERE

* [ x] I have read the troubleshooting guide found at https://github.com/cp2004/OctoPrint-WS281x_LED_Status/wiki/Troubleshooting-Guide

ingo916, 2021-01-19:
[image]
[image]

ingo916, 2021-01-19:
found this in the logs
KeyError: 'progress_cooling'
2021-01-19 06:35:21,388 - octoprint.util.comm - ERROR - Error while processing temperatures in ws281x_led_status, skipping
Traceback (most recent call last):

cp2004, 2021-01-19:
That's... a lot of errors. Most of which I have setup better handling if everything fails in my development branch, it really did not like that...

Have you changed anything in `config.yaml` for this plugin? It's complaining that it cannot use the value for delay as a floating point number, which shouldn't be the case. Could you post everything under the `ws281x_led_status:` block from the configuration file? 

If you can't find it, here: https://docs.octoprint.org/en/master/configuration/config_yaml.html

ingo916, 2021-01-19:
I double checked all the settings and found one that Enable idle effect  had no value. i must have removed it while i was getting familiar with the plugin. Added it back rebooted and its working as it should.

Thanks for the great plugin!!

closed 2021-01-19 by ingo916