cp2004/OctoPrint-WS281x_LED_Status #87: No LED output at all

antonyburden, 2021-01-23:
I have a standard WS2812b strip of 24 LEDs, I have proved it works by connecting to a Arudino Nano and run some simple LED code.

I have tried with / without a level shifter.

Common ground (just as you need with the Arduino)

Connected as per Wiki, using GPIO 10

I did have to uncheck "Request exclusive access to the serial port" as this caused issues reconnecting after a SmartPlug power it down and back up. (However, it didn't work with this checked either)

**What did you expect to happen**

**Logs and screenshots**

[octoprint.log](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/files/5860141/octoprint.log)
[plugin_ws281x_led_status_debug.log](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/files/5860142/plugin_ws281x_led_status_debug.log)

**Version of plugin and OctoPrint**

OctoPrint Version 1.5.3 with Klipper Version 0.3.1
OctoPi Version 0.17.0, running on Raspberry Pi 3 Model B Plus Rev 1.3

**(If applicable) Browser version & browser error console**

**Additional context**

antonyburden, 2021-01-23:
Just purchased another brand of ws2812b that apparently work without shifter and they are the same - perfect on Arduino. Nothing with the plugin.

cp2004, 2021-01-24:
Hey @antonyburden! So, what do we know. You've tested the strip works fine, the logs show that the OctoPrint plugin and the effect runner part think everything is OK. OS config test passes, strip is successfully started *according to the plugin*. Thanks for providing a lot of information BTW, usually I have to ask for all of that...

Could you post the contents of the file `/boot/config.txt`? Sometimes there are extra settings that sneak in there, that break things. 

The other thing I'm thinking is that maybe there is something off with the wiring to the Pi, though you sound confident that this is OK. Other than that... I don't see where else the problem could lie. The log shows only a single startup of OctoPrint, so I am going to assume as well that after you setup/configured it, you rebooted? Actually that would show up in the `plugin_ws281x_led_status_debug.log`, since it would fail.

antonyburden, 2021-01-24:
Heya, thanks for getting  back to me - I've just been trying the Enclosure plugin and can't get that working either, he did have a test python script for testing GPIO output and I can see 3v flip flopping on my meter.

Here is config.txt:

dtparam=spi=on
core_freq=250

antonyburden, 2021-01-24:
For interest here is the script

`import RPi.GPIO as GPIO
from time import sleep
import signal
import sys

GPIO.setmode(GPIO.BCM)

GPIO.setwarnings(False)

PIN_NUMBER = 18

GPIO.setup(PIN_NUMBER, GPIO.OUT)

def signal_handler(signal, frame):
    GPIO.output(PIN_NUMBER, True)
    print 'Exiting...'
    sys.exit(0)

signal.signal(signal.SIGINT, signal_handler)

while True:
    print 'Setting low...'
    GPIO.output(PIN_NUMBER, False)
    sleep(5)
    print 'Setting high'
    GPIO.output(PIN_NUMBER, True)
    sleep(5)`

cp2004, 2021-01-24:
>Here is config.txt:
>dtparam=spi=on
>core_freq=250

Is that everything? Both of those items *should* be there, but hopefully there is nothing else causing a problem that has been put in either by mistake or something else in the past.

If you want to try the ws281x test scripts, there is one here:
https://github.com/rpi-ws281x/rpi-ws281x-python/blob/master/examples/strandtest.py

Activate OctoPrint's virtualenv (where the dependency would be installed), change the pin to 10 and test:
* `source ~/oprint/bin/activate`
* `python strandtest.py`

And see if that makes anything happen?

antonyburden, 2021-01-24:
That is all that is in the config.txt

and that script did nothing to the LED's (after changing to PIN_NUMBER = 10)

(oprint) pi@octopi:~ $ python strandtest.py
Press Ctrl-C to quit.
Use "-c" argument to clear LEDs on exit
Color wipe animations.
Theater chase animations.
Rainbow animations.

antonyburden, 2021-01-24:
I use Klipper - no idea if this makes a difference

antonyburden, 2021-01-25:
Out of interest I just disabled ALL plugins bar yours and still nothing.

cp2004, 2021-01-25:
I am suspicious of that being everything in the `/boot/config.txt` file - for referenced, this is what mine looks like:
https://gist.github.com/cp2004/5383a64e13b49b0f59813c88319486de

Unplugging the ground wire - interesting. That implies it could be getting some data, and you need the common ground to keep this signal clean, otherwise you get random flickering as you have seen.

All things you have said point to a problem with the Pi. The strip works well (arduino), the power supply is good enough (since it works not connected to the Pi). The only thing unexplained is the OS software. As I said above, there should be more in `/boot/config.txt`.

> I read a constant 3v (or 5v with shifter) across the green wire (DIN) is that expected?

I can't actually remember what I read, it's been a while since I measured any LEDs. It depends, I don't believe the signal drops for all that long, it is quite high frequency but I don't know the specifics.

Do you have a spare SD card? I would recommend testing new software (maybe OctoPi 0.18?) to rule out any issues with the OS config getting wiped out. Other than that... Keep on trying things to see if you can pinpoint the problem. The OS would be my next thing to target, so that's my suggestion for now.

antonyburden, 2021-01-27:
So... 

Another day...

I got a new SD card, put latest OctoPi on (0.18), did not restore backup, just went through wizard and then installed your plugin.

Went through all of your plugin startup which set the permissions etc. Shutdown and rebooted.

Still nothing at all.

Out of interest, here is my config.txt now.

# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment this if your display has a black border of unused pixels visible
# and your display can output without overscan
#disable_overscan=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Uncomment this to enable infrared communication.
#dtoverlay=gpio-ir,gpio_pin=17
#dtoverlay=gpio-ir-tx,gpio_pin=18

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

[pi4]
# Enable DRM VC4 V3D driver on top of the dispmanx display stack
dtoverlay=vc4-fkms-v3d
max_framebuffers=2

[all]
#dtoverlay=vc4-fkms-v3d
# enable raspicam
start_x=1
gpu_mem=128
dtparam=spi=on
core_freq=250

antonyburden, 2021-01-27:
I just got hold of a 74AHCT125 and a breadboard and wired up exactly as your wiki - still nothing!

antonyburden, 2021-01-28:
Ok, new PI is working! sort of...

The main on / off works, the green startup trail happens on startup, the torch comes on..

However, the torch does not toggle off (yes I've set this as an option), I've turned off ALL effect and the start up green still happens. None of the test buttons work either.

cp2004, 2021-01-28:
Startup I don't think can be disabled, that is a bug/oversight.

For the rest of it, enable 'debug logging' in the features tab, then reproduce the issue and upload the debug log alongside an octoprint.log, thanks.

antonyburden, 2021-01-28:
After a complete shutdown and reboot it's behaving a little better.

@WS_LIGHTSOFF works
@WS_LIGHTSON works
@WS_TORCH_ON works
@WS_TORCH_OFF does not work and the icon can never be toggled off once on.

The torch timer function worked once..

I am in middle of print and the green led's are slowly progressing.

The heating was weird - I've now turned off bed heat tracking so it's just the hotend - should it be tool 0 or 1 ?

I will enable logging after this print.

cp2004, 2021-01-28:
That sounds like you've got an issue with the timer, for sure - `@WS_TORCH_OFF` is disabled in timer mode, since the off is controlled by the timer. The octoprint logs should show the timer on/off and it's corresponding command in the ws281x log.
Heating progress can be a bit confusing if the temperature reporting is confusing to the plugin, since it might jump around as it tries to work out if the tool or bed is heating, and when to stop - there's nothing clear to say 'I am heating, x percent done' - sometimes it has to guess.

>  should it be tool 0 or 1 ?

If you only have one extruder, then it is 0, otherwise I think you would know because they are referred to as 'Tool 0' and 'Tool 1' in the UI.

antonyburden, 2021-01-28:
Still printing but here is the log so far

the first TORCH command was me typing @WS_TORCH - this went out after some random time and back to the progress.
the second TORCH command was me typing @WS_TORCH_ON

Only after a very long random times did it go off!

I could never force it off with the icon or the command.

[image]

```
[2021-01-28 23:00:29,005] INFO: Initialising LED strip
[2021-01-28 23:00:29,008] INFO: Strip successfully initialised
[2021-01-28 23:00:29,012] DEBUG: Current settings:
 | * STRIP SETTINGS *
 | - led_count: 12
 | - led_pin: 10
 | - led_freq_hz: 800000
 | - led_dma: 10
 | - led_invert: False
 | - led_brightness: 102
 | - led_channel: 0
 | - strip_type: WS2812_STRIP
 | - reverse: False
 | * EFFECT SETTINGS *
 | startup
 | - enabled: True
 | - color: #136335
 | - effect: solid
 | - delay: 75.0
 | idle
 | - enabled: False
 | - color: #00ccf0
 | - effect: wipe2
 | - delay: 75.0
 | disconnected
 | - enabled: True
 | - color: #000000
 | - effect: cycle
 | - delay: 25.0
 | progress_print
 | - enabled: True
 | - color: #00ff00
 | - base: #000000
 | progress_heatup
 | - enabled: True
 | - color: #ff0000
 | - base: #0000ff
 | progress_cooling
 | - enabled: True
 | - color: #ff0000
 | - base: #0000ff
 | failed
 | - enabled: True
 | - color: #ff0000
 | - effect: pulse
 | - delay: 10.0
 | success
 | - enabled: True
 | - color: #000000
 | - effect: rainbow
 | - delay: 25.0
 | paused
 | - enabled: True
 | - color: #0000ff
 | - effect: bounce
 | - delay: 40.0
 | printing
 | - enabled: False
 | - color: #ffffff
 | - effect: solid
 | - delay: 1.0
 | torch
 | - enabled: True
 | - color: #ffffff
 | - effect: solid
 | - delay: 1.0
 | * ACTIVE TIMES *
 | - start: None
 | - end: None
[2021-01-28 23:00:29,014] INFO: On message recieved, turning on LEDs
[2021-01-28 23:01:25,107] DEBUG: Recieved message to update progress: progress_print 26
[2021-01-28 23:03:41,917] DEBUG: Recieved message to update progress: progress_print 27
[2021-01-28 23:05:29,643] DEBUG: Recieved message to update progress: progress_print 28
[2021-01-28 23:07:51,106] DEBUG: Recieved message to update progress: progress_print 29
[2021-01-28 23:09:38,616] DEBUG: Recieved message to update progress: progress_print 30
[2021-01-28 23:12:00,051] DEBUG: Recieved message to update progress: progress_print 31
[2021-01-28 23:13:47,533] DEBUG: Recieved message to update progress: progress_print 32
[2021-01-28 23:16:09,299] DEBUG: Recieved message to update progress: progress_print 33
[2021-01-28 23:17:56,606] DEBUG: Recieved message to update progress: progress_print 34
[2021-01-28 23:20:18,420] DEBUG: Recieved message to update progress: progress_print 35
[2021-01-28 23:22:05,395] DEBUG: Recieved message to update progress: progress_print 36
[2021-01-28 23:24:27,447] DEBUG: Recieved message to update progress: progress_print 37
[2021-01-28 23:26:14,075] DEBUG: Recieved message to update progress: progress_print 38
[2021-01-28 23:28:36,369] DEBUG: Recieved message to update progress: progress_print 39
[2021-01-28 23:30:22,750] DEBUG: Recieved message to update progress: progress_print 40
[2021-01-28 23:32:34,245] DEBUG: Recieved message to change effect: torch
[2021-01-28 23:32:45,067] DEBUG: Recieved message to update progress: progress_print 41
[2021-01-28 23:33:23,283] DEBUG: Recieved message to change effect: torch

```

antonyburden, 2021-01-29:
Morning - turned it all off this morning and back on - now the torch is working as a toggle!

```
[2021-01-29 08:42:45,569] DEBUG: Recieved message to change effect: torch
[2021-01-29 08:43:30,692] DEBUG: Recieved message to change effect: startup
[2021-01-29 08:44:21,707] DEBUG: Recieved message to change effect: torch
[2021-01-29 08:44:24,764] DEBUG: Recieved message to change effect: startup
```

antonyburden, 2021-01-29:
I have added @WS_TORCH in my end GCode, this is to light up the print for when a webcam capture is taken.

However it never turns off - even though I set the torch to go off after 30 seconds. I also tried @WS_TORCH_ON

cp2004, 2021-01-29:
I did go through the spaghetti code that was controlling the priorities of effects - it was confusing me, and I wrote it...

Maybe for some reason, it is saving torch as the effect *to go back to* when it switches effect, so when OctoPrint enters idle state this is not saved for when the torch is turned off. I can't guarantee a solution in the short term, but on the [`devel`](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/tree/devel) branch I have made a lot of changes relating to tidying up this code. So, I can say that it should be improving in the future, no guarantees as to when that will come out (since there is still quite a lot of work to go) as it is going to be quite a large update.

If you can figure out the exact times it works/doesn't work precisely, then that would be cool so I can test I have it fixed. Otherwise, don't worry about it and I have a more robust torch setup coming soon. I know that doesn't help, but there's a lot of stability improvements going into the next release (at least, there should be 🙂)

antonyburden, 2021-01-30:
Thanks for at least trying to get to the bottom of it, I am a IT Manager/Programmer of 30+ years and I know the issues you face.

I've also noticed the heat effect kick in briefly during print progress. There does seem to be a priority issue.

I am installing the devel now.

cp2004, 2021-01-31:
Installing devel... brave move lol. Anyway, I have been testing a few things with the manual torch (icon in the navbar) today, and it appears to have the highest priority, at least I can't get anything to stop it. I haven't given it a good test, but it seems fixed. The torch timer however may get cancelled by the next effect - I will have to check on that one, I can't remember now. That might be where this problem lies.

I did have some bugs in heating/cooling progress to work out, since they got re-written but that's getting there too. The startup effect not disabling is still a bug. There's also some with the new fade transitions, which I have fixed locally but not pushed yet, since it needs testing.

Thanks for testing it, I will be publishing a release candidate before 0.7.0 and try and get as many testers as possible to keep it as bug-free as it has been, RC testing has really helped to validate releases in my plugins where there are major changes. Most of the time it goes well.

cp2004, 2021-02-08:
@antonyburden can I close this now? You got the plugin working, now, I don't see anything outstanding beside the next updates.

I'll ping when there's a release candidate available.

cp2004, 2021-02-10:
I've been fixing a fair amount of random issues yesterday and today on the devel branch, some priority stuff has definitely been solved. Closing this issue now since the original problem was solved as well as subsequent discussion. Feel free to carry on testing things and reporting back as things go. Thank you 🙂!

closed 2021-02-10 by cp2004