cp2004/OctoPrint-WS281x_LED_Status #88: Plugin not switching from connecting to idle effect upon startup

trevor229, 2021-01-26:
**What do you need help with?**
I just set up octoprint 0.18.0 and when booting, the connecting effect plays as it should, however when octoprint auto-connects at bootup, the plugin remains in the connecting effect and does not switch to the idle effect. When I manually disconnect, it changes to the disconnected effect and then changes to the idle effect upon clicking connect manually.

I have tried with and without the PortLister plugin and `Request exclusive serial port access`

The only error I see is in `octoprint.log` and it is this section
```
2021-01-26 21:41:09,576 - octoprint.plugin - ERROR - Error while calling plugin ws281x_led_status
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/plugin/__init__.py", line 271, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/__init__.py", line 1890, in wrapper
    return f(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_ws281x_led_status/__init__.py", line 605, in on_event
    self.update_effect(self.supported_events[event])
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_ws281x_led_status/__init__.py", line 560, in update_effect
    if not self.SETTINGS[mode_name][
KeyError: 'idle'
```

**Log files**

Here are my log files:
* [plugin_ws281x_led_status_debug.log](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/files/5876367/plugin_ws281x_led_status_debug.log)
* [octoprint.log](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/files/5876368/octoprint.log)

* [x] I have read the troubleshooting guide found at https://github.com/cp2004/OctoPrint-WS281x_LED_Status/wiki/Troubleshooting-Guide

cp2004, 2021-01-26:
It should be fixed in the upcoming version, I was aware of this issue. Since the connect comes before the plugin is started up, it can't play the effect. In future versions it should (a) not fail and (b) save the connect event so it can be used when the strip etc. is setup.

trevor229, 2021-01-26:
Awesome, thank you for the quick response and info.

cp2004, 2021-03-05:
Hey @trevor229, 0.7.0rc1 is [out now for testing](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/releases/tag/0.7.0rc1) - there's a [feedback ticket here](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/issues/119) for any comments on the release. Any feedback is appreciated.

cp2004, 2021-03-11:
0.7.0 has been released.

closed 2021-03-11 by cp2004