cp2004/OctoPrint-WS281x_LED_Status #89: LEDs stay full white unless pi is performing a separate task

TyropeDeMotorChope, 2021-01-28:
[As seen in this video](https://drive.google.com/file/d/1-aiJGQ5O3VpKG_XGMk8xeULJbUw4IGfl/view?usp=drivesdk) my WS2812 strip wants to stay white unless I interact with Octoprint. While Octoprint does functions, my LEDs return to normal (in this case full green), only to go back white once the function is complete. On startup, I get the correct colors for a few seconds before it flickers 5/6 times, eventually remaining white.

This started after I was testing the strip at first with rainbow cycle at 3ms delay. Was perfect for like 20 minutes until the flickering started. Cleaned/uninstalled/reinstalled the plug-in and uninstalled all other LED plugins but the issue persists.

cp2004, 2021-01-28:
Please fill in the template when opening issues, it's much slower for me because I now have to ask for all the information again:

* octoprint.log
* plugin_ws281x_led_status_debug.log

TyropeDeMotorChope, 2021-01-28:
[plugin_ws281x_led_status_debug.log](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/files/5888568/plugin_ws281x_led_status_debug.log)
[octoprint.log](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/files/5888569/octoprint.log)

Thanks so much for this plugin btw, works flawlessly on my other printer. LEDs are always the first thing my friends mention when I show them the printer.

cp2004, 2021-01-28:
OK the OS configuration test passes, but as I learnt in #75 it doesn't pick up everything. So now I need you to check what is in the config.txt file, it should have this line in the bottom:
```
core_freq_min=500
```
If it doesn't (OS config check says it's there) add it, **or** if the line `core_freq=250` or some other core_freq number is there, remove it. If you could paste the contents of the file as well, so I can check that there's nothing else broken with it, that would be great 🙂

TyropeDeMotorChope, 2021-01-28:
[config.txt](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/files/5888955/config.txt)

The core_freq=250 line is in there, I'll remove it and reboot

TyropeDeMotorChope, 2021-01-28:
I keep getting permission denied when I try to overwrite or delete the file, any suggestions? I'm using WinSCP logged in as 'pi'

cp2004, 2021-01-28:
You need to use a root account, I personally use `sudo nano` on the Pi to do that, not sure how it works in winSCP.

At this point, I think it is a bug that that line is added by the wizard - I have seen it 3 times now. Not a large number, but enough to make me think that there's a bug. I think I got it fixed for the next version, since things work slightly differently.

TyropeDeMotorChope, 2021-01-28:
Got it done using putty and sudo nano. Works like a charm now.

The line may have been added by another LED plugin I was running initially. One feature I would love from your plugin is the ability to control a white (or other) LED that isn't on the neopixel strip. My camera has an LED ring connected to GPIO 23 and now I'm wondering about what the best way to control that would be. Thank you for all your work man, it's awesome.

cp2004, 2021-01-28:
You could look into the OctoLight plugin for that LED, not something I will do from my plugin since I want to keep it 100% focused on the WS281x LEDs, and there are other plugins for that.

Great stuff, enjoy!

closed 2021-01-28 by cp2004