cp2004/OctoPrint-WS281x_LED_Status #90: cracy flikering

CKoblinger, 2021-01-28:
Hello I need help with the software.

**Log files**

Here are my log files:
* `octoprint.log`: [octoprint.log](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/files/5889467/octoprint.log)
* `plugin_ws281x_led_status_debug.log`: [plugin_ws281x_led_status_debug.log](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/files/5889477/plugin_ws281x_led_status_debug.log)

* [YES ] I have read the troubleshooting guide found at https://github.com/cp2004/OctoPrint-WS281x_LED_Status/wiki/Troubleshooting-Guide

cp2004, 2021-01-28:
Could you provide some more details as to what you need help with?

CKoblinger, 2021-01-28:
When I start my power supply for the RGB strip (SK6812_RGBW) I get crazy with flickering even if the switch in the software is turned off (- to - pi connection was made properly; 470 ohm resistor is connected between the data Pin and the strip). If the data pin get disconnected the color stays at the last setting (flashing ca. 1Hz) and it's not flickering anymore, so I think it's a problem with the software

cp2004, 2021-01-28:
Could you post the contents of your `/boot/config.txt` file? I see from the log you previously installed other RGB plugins, there's an issue I know with RGB Status where it puts the wrong line into that file, it may not have been removed. 

If the line `core_freq=250` appears in the `/boot/config.txt` file, remove it.

CKoblinger, 2021-01-29:
Thank you so much that solved the problem. That core_freq=250  was there because I installed the enclosure plugin for a temp sensor. Thank you for the fast support! Have a nice day.

cp2004, 2021-01-29:
Great, enjoy the plugin 🙂!

closed 2021-01-29 by cp2004