cp2004/OctoPrint-WS281x_LED_Status #91: Using white on a SK6812_STRIP_GRBW LED strip

samwiseg0, 2021-02-01:
**What do you need help with?**

I have the plugin setup and I would like to use the white LED that comes with the LED strip.

Is this possible?

I have tried to go through the configuration and see if there was an option to enable white but I only see options for RGB

**Log files**
```
[2021-01-30 19:49:23,787] INFO: Initialising LED strip
[2021-01-30 19:49:23,793] ERROR: Strip failed to initialize, no effects will be run.
[2021-01-30 19:49:23,879] ERROR: Please check your settings.
[2021-01-30 19:49:23,913] ERROR: Here's the exception: ws2811_init failed with code -13 (Unable to initialize SPI)
[2021-01-30 19:49:23,945] INFO: No strip initialised, exiting the effect process.
[2021-01-30 19:51:01,452] INFO: Initialising LED strip
[2021-01-30 19:51:01,457] ERROR: Strip failed to initialize, no effects will be run.
[2021-01-30 19:51:01,495] ERROR: Please check your settings.
[2021-01-30 19:51:01,527] ERROR: Here's the exception: ws2811_init failed with code -13 (Unable to initialize SPI)
[2021-01-30 19:51:01,558] INFO: No strip initialised, exiting the effect process.
[2021-01-30 19:51:44,603] INFO: Initialising LED strip
[2021-01-30 19:51:44,609] INFO: Strip successfully initialised
[2021-01-30 19:51:44,609] INFO: Debug logging not enabled, if you are reporting issues please enable it under 'Features' in the settings page.
[2021-01-30 19:51:44,611] INFO: On message recieved, turning on LEDs
[2021-01-30 19:52:49,221] INFO: Off message recieved, turning off LEDs
[2021-01-30 19:52:51,764] INFO: On message recieved, turning on LEDs
[2021-01-30 19:53:44,189] INFO: Off message recieved, turning off LEDs
[2021-01-30 19:53:45,309] INFO: On message recieved, turning on LEDs
[2021-01-30 19:54:01,764] INFO: Kill message recieved, Bye!
[2021-01-30 19:54:01,810] INFO: Initialising LED strip
[2021-01-30 19:54:01,815] ERROR: Strip failed to initialize, no effects will be run.
[2021-01-30 19:54:01,863] ERROR: Please check your settings.
[2021-01-30 19:54:01,905] ERROR: Here's the exception: ws2811_init failed with code -5 (mmap() failed)
[2021-01-30 19:54:01,936] INFO: No strip initialised, exiting the effect process.
[2021-01-30 19:56:46,492] INFO: Initialising LED strip
[2021-01-30 19:56:46,497] INFO: Strip successfully initialised
[2021-01-30 19:56:46,498] INFO: Debug logging not enabled, if you are reporting issues please enable it under 'Features' in the settings page.
[2021-01-30 19:56:46,499] INFO: On message recieved, turning on LEDs
[2021-01-30 19:56:46,499] INFO: Off message recieved, turning off LEDs
```
* [X] I have read the troubleshooting guide found at https://github.com/cp2004/OctoPrint-WS281x_LED_Status/wiki/Troubleshooting-Guide

cp2004, 2021-02-01:
The plugin doesn't support the W space, only RGB colours - so you won't find that option anywhere. I will make this clear on the wiki for strip configuration.

It won't support RGBW space any time soon, unless someone else contributes that implementation. There's many requests all ready here, and some more things I want to implement myself coupled with the fact this is only my spare time stuff, it is not feasible for me to do it.

samwiseg0, 2021-02-01:
Bummer. I got these LEDs just for that. I will look into what it would take to implement. Maybe I can PR the functionality.

cp2004, 2021-02-01:
I think the biggest issue when I looked was that the HTML colour picker would only do RGB, so it would be difficult to have settings for it. If it can be passed as part of the hex colour through the depths of the plugin (and decoded the other end) it would mostly be front-end stuff.

cp2004, 2021-02-01:
If you do go digging, start with the devel branch. I've tidied up a _lot_ of the mess, it should be a lot easier to understand and work out what goes where.

samwiseg0, 2021-02-02:
Sweet! I will take a look. I understand the color picker might be an issue but maybe just a checkbox to enable it and set a value for brightness 1-255? The color picker can just remain there and the end user can set a value of all 0s?

cp2004, 2021-02-02:
Ideally I would like to keep it as simple as possible, since there already are a *lot* of settings. A global option (strip settings) to enable RGBW, and some way of specifying a W alongside the colour picker.

The underlying library supports passing a W parameter, it defaults to 0:
https://github.com/rpi-ws281x/rpi-ws281x-python/blob/88402d6049220d99b0ad947bde28ac69a7720e49/library/rpi_ws281x/rpi_ws281x.py#L145-L150

Setting the W when RGB are equal is probably the simplest solution, since it could all be done in the hex_to_rgb function or similar pre-processor of colour values, like the colour correction one:
https://github.com/cp2004/OctoPrint-WS281x_LED_Status/blob/e56a72ac45a134fe37dc0946f1a116bef0e3f3fe/octoprint_ws281x_led_status/util.py#L13-L24

samwiseg0, 2021-02-02:
I _think_ the easiest way is to set it in `apply_color_correction`. Something like this. I would just need to add logic for the UI global override. 

```python
def apply_color_correction(settings, r, g, b):
    red = int_0_255(r * (int(settings["red"]) / 100))
    green = int_0_255(g * (int(settings["green"]) / 100))
    blue = int_0_255(b * (int(settings["blue"]) / 100))
    white = 0
    # Use white LEDs when enabled
    if red == green == blue == 255:
        red = 0
        green = 0
        blue = 0
        white = 255
    return red, green, blue, white
 ```

cp2004, 2021-02-02:
You would have to check that everywhere that uses the values of colour correction can handle white - I think it might be able to, depends if it uses `*color` and not `color[0], color[1], colour[2]` patterns in the effects.

Other than that, I think that would make sense - would it work if they were just all equal, not necessarily 255?

samwiseg0, 2021-02-02:
Not with with logic. We could make that happen? I would need to test how that works with the LEDs if lets say the white value is 100.

cp2004, 2021-03-11:
0.7.0 has been released.

PR #93 (2021-02-03): Add functionality to use white LEDs on a RGBW strip
~~I have not tested this yet but here is what I have so far.~~

This version is tested and works. I would like to hide the white brightness bar when the checkbox is not enabled but I could not get that to work? Maybe you can offer some guidance?

Fixes #91
closed 2021-03-11 by cp2004