cp2004/OctoPrint-WS281x_LED_Status #92: Using Rpi 4, WS2812b and I2C bidirectional level shifter and still no LED output

Mauri9391, 2021-02-01:
Hi, 
I installed the plugin, went through the setup and set everything like described.
OS Config Test: Tests completed successfully

Wiring:
external 5 VDC -> LED +
external 5 VDC -> HV (level shifter)
external GND -> LED - / Raspberry Pi GND / two times GND (level shifter)
3.3 VDC Rapbery Pi -> LV (level shifter)
GPIO 10 -> LV1 (level shifter)
HV1 (level shifter) -> LED Din

Octoprint Version 1.5.3
Octopi Version 0.18.0, running on Raspberry Pi 4 Model B Rev 1.1

I also tried this troubleshoot:
https://github.com/cp2004/OctoPrint-WS281x_LED_Status/issues/73

My logs:
[octoprint.log-1.txt](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/files/5906295/octoprint.log-1.txt)
[plugin_ws281x_led_status_debug.log-1.txt](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/files/5906297/plugin_ws281x_led_status_debug.log-1.txt)

It didn't work with or without a level shifter.
And since I'm using the plugin the signal of the raspbery Pi cam I'm using is disturbed (Don't know if this is the plugin)

Mauri9391, 2021-02-02:
I solved the problem with the stream of the picam. I rolled up the ribbon cable. Thats the reason why the singal was disturbed

cp2004, 2021-02-02:
So what do we know is working or isn't working:
* The plugin thinks it is fine - no errors there
* The wiring looks good - I had to draw it out, but it works, all apart from one thing:
  > 3.3 VDC Rapbery Pi -> LV (level shifter)
  
  This doesn't make sense to me, I don't understand why/where this wire is going into the level shifter.
* OS config test is reporting all clear

What is left to work out:
* Does the strip work? Do you have any other devices to test, such as an arduino/ESP board?
* Is the Pi working? Do you have other LEDs to test with?

When there is no output, it is very hard to find out which part is broken. I can't do a lot beyond suggest testing alternatives of each component, because for example we had #87 where it was the Raspberry Pi that was at fault. I know, this is not very helpful but I haven't got anything more advanced to try and test this and it mostly relies on you trying different combinations to verify each component works. If you have any more questions, feel free to say, I'll answer as soon as possible.

cp2004, 2021-02-02:
That's good, that you have some things working, and seem to be on track to working something out. It will work, and it will be great 🙂.

Let me know if you have other problems, or when you get it solved.

Mauri9391, 2021-02-03:
After a few test I'm still frustrated. 
All stripes are working fine. There could be no fault.
My last thought is that the cables from the pi to the first led are to long. (They are ca. 70 cm long and have a cross section of 1 mm²)
Can this be the problem?

cp2004, 2021-02-03:
It can be a problem, yes. I was speaking with someone on the OctoPrint discord the other day that had issues with voltage drop from long cables to the Pi - they had to add one sacrificial LED by the Pi, before the long wires for it to work. No idea why/how that worked for them, but it did... [Here's a link to the message for reference.]( https://discord.com/channels/704958479194128507/708230241285439518/805559523183951893)

Mauri9391, 2021-02-03:
Your discord link isn't working. I get the message that I don't have access to txt channels or the server doesn't exist.

cp2004, 2021-02-03:
Not sure what's up there, sorry about that. It works for me, but it doesn't matter anyway. They were having issues with longer wires and the LEDs were not receiving any signal. They added a single LED on a short cable beside the Pi, then long cables the other side of this LED would work. Not much more than that - I can't say if this will work, but long cables have been known to cause issues.

Mauri9391, 2021-02-03:
I googled it and saw a diagram. I will try it this afternoon and i will report if it works or not

Mauri9391, 2021-02-03:
Now the LED string works fine at my Pi4 and Octoprint.
My mistakes:
I got a loose contact in the data cable by soldering it to the existing data cable of the led string. 
I also found out I mixed up VCC and GND between the strins after reassembling.
At all I was too stupid, I'm soory my bad.
Thanks for your help and the great plugin

cp2004, 2021-02-03:
No problem, enjoy it 🙂!

closed 2021-02-03 by cp2004