cp2004/OctoPrint-WS281x_LED_Status #94: Nothing Happens

nigelboubert, 2021-02-04:
So I'm stumped because nothing works:  So far:

1 Tried two different Pis a 3b and a 4
2) Tried two different LED strips both ws2812
3) The strips work on a different pi using strandtest
4) The strip is powered externally.
5) The strip is connected to GPIO - Pin 10 (din), 14 (grnd)
6) The strip test will not change from Blue. 

System info from Octoprint:

browser.user_agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/88.0.4324.96 Safari/537.36 Edg/88.0.705.56
connectivity.connection_check: 8.8.8.8:53
connectivity.connection_ok: true
connectivity.enabled: true
connectivity.online: true
connectivity.resolution_check: octoprint.org
connectivity.resolution_ok: true
env.hardware.cores: 4
env.hardware.freq: 1500
env.hardware.ram: 1903988736
env.os.bits: 32
env.os.id: linux
env.os.platform: linux
env.plugins.pi_support.model: Raspberry Pi 4 Model B Rev 1.1
env.plugins.pi_support.octopi_version: 0.18.0
env.plugins.pi_support.throttle_state: 0x0
env.python.pip: 20.3.3
env.python.version: 3.7.3
env.python.virtualenv: true
octoprint.safe_mode: false
octoprint.version: 1.5.3

Attaching a number of files here.  If you need more......

[image]
[image]
![Config_test](https://user-images.githubusercontent.com/7
[plugin_ws281x_led_status_debug (6).log](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/files/5926904/plugin_ws281x_led_status_debug.6.log)
[octoprint (5).log](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/files/5926905/octoprint.5.log)

Any suggestions would be greatly apppreciated...

* [ ] I have read the troubleshooting guide found at https://github.com/cp2004/OctoPrint-WS281x_LED_Status/wiki/Troubleshooting-Guide

cp2004, 2021-02-04:
I typed out a reply to this, but having intermittent internet issues so it didn't go through...

The test buttons don't work unless the printer is connected - I can't see it logged if they are pressed, but that might be because it is not logged - I haven't seen them in any logs actually, which could be a shortfall on my part. Even the virtual printer is connected, it is because they require M150 commands. Corrected for the next version, this is no longer an issue.

Could you post the contents of `/boot/config.txt`? - you can find it using SSH, or on the Pi's SD card.

Apparently there can be additional lines in there that break the Pi 4 functionality, such as `core_freq=250`, if this exists remove it.

Could you also explain a little more about the problem? Unless it is literally just that the LEDs come on blue and then will not change to other colours.

nigelboubert, 2021-02-04:
Literally nothing happens at all.   I mentioned the test buttons because I thought (could be wrong) the colour of the torch would change.  It doesn't work either.

core_freq is set to 500

Here is the config.txt file:

# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment this if your display has a black border of unused pixels visible
# and your display can output without overscan
#disable_overscan=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Uncomment this to enable infrared communication.
#dtoverlay=gpio-ir,gpio_pin=17
#dtoverlay=gpio-ir-tx,gpio_pin=18

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

[pi4]
# Enable DRM VC4 V3D driver on top of the dispmanx display stack
dtoverlay=vc4-fkms-v3d
max_framebuffers=2

[all]
#dtoverlay=vc4-fkms-v3d
# enable raspicam
start_x=1
gpu_mem=128
dtparam=spi=on
core_freq_min=500
pi@octopip4:~ $

cp2004, 2021-02-04:
Does it work on the same Pi, as you have it wired now with the strandtest? (you will need to change the pin to 10 to make it use SPI) You mentioned you have it on a different Pi working, just wanted to confirm. I have seen that it can be the Pi that is broken. Unlikely, but I have had one issue opened where that was the solution.

> I thought (could be wrong) the colour of the torch would change.

I understand now, it looks like you are using a theme that has not defined those buttons. By default, they are like this:
[image]
Your theme seems to be overriding them. Not important though, so don't worry about it.

Can you measure the signal voltage when the strip is working, and then compare it to when it doesn't work - I want to work out if there is any signal at all. If you have an oscilloscope or similar to measure the signal that can really help determine if *anything* is working.

nigelboubert, 2021-02-04:
Disabled Themeify and the buttons show properly.  Correct me if I am wrong but my understanding is that the torch button would turn the lights on to whatever color is selected?

[image]

[image]

Sadly I loaned my multimeter out last week.  Have to see if I can get it back.

The 3rd Pi ( the one the leds work on) does not have Octoprint on it.  I was using to play with the LEDs.  Interestingly I don't believe it uses SPI since the leds are on pin 18.  This would be PWM.

Following on the above it seems SPI is the culprit some how.  These are the leds I am working with:
https://www.amazon.ca/gp/product/B01MQ08JH6/ref=ppx_yo_dt_b_asin_title_o06_s01?ie=UTF8&psc=1

cp2004, 2021-02-04:
Apologies for the slow response here, I've been busy with docs this evening.

> Disabled Themeify and the buttons show properly. Correct me if I am wrong but my understanding is that the torch button would turn the lights on to whatever color is selected?

The torch mode is a button to display a specific effect, which can be configured under the 'features' tab. It defaults to solid white.

> The 3rd Pi ( the one the leds work on) does not have Octoprint on it. I was using to play with the LEDs. Interestingly I don't believe it uses SPI since the leds are on pin 18. This would be PWM.

In the strandtest file, you can change the pin number - correct, 18 is PWM. When it is pin 10 the SPI interface is used. Though note that you will need to do the [steps outlined](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/wiki/SPI-Setup) manually on a Pi without the plugin/setup wizard.

nigelboubert, 2021-02-05:
No worries on the response time.  I appreciate the help.    I think tomorrow I'm going to start back at the beginning and reimage the SD card and do a fresh install.  I'll start with just your plugin and take it from there.

nigelboubert, 2021-02-08:
Hi Charlie,

Soooooooo   now I am thoroughly confused.   I started with a fresh image on the Pi 4 and only installed your plugin in Octopi.  Tried a ton of things all weekend to no avail.   This morning I did some reading on SPI and it mentioned MOSI on pin 19.  Moved the DIN lead from pin 10 to pin 19 and........ wait for it...... the torch works.
[image]
[image]

Very confusing for this old brain.........

nigelboubert, 2021-02-08:
More interesting stuff...

[2021-02-08 16:32:33,123] DEBUG: Recieved message to change effect: idle
[2021-02-08 16:32:33,128] DEBUG: Recieved message to update progress: progress_print 0
[2021-02-08 16:32:33,133] DEBUG: Recieved message to update progress: progress_heatup 13
[2021-02-08 16:32:37,190] DEBUG: Recieved message to update progress: progress_heatup 14
[2021-02-08 16:32:39,176] DEBUG: Recieved message to update progress: progress_heatup 16
[2021-02-08 16:32:40,218] DEBUG: Recieved message to update progress: progress_heatup 17
[2021-02-08 16:32:41,259] DEBUG: Recieved message to update progress: progress_heatup 18
[2021-02-08 16:32:42,196] DEBUG: Recieved message to update progress: progress_heatup 20
[2021-02-08 16:32:43,236] DEBUG: Recieved message to update progress: progress_heatup 22
[2021-02-08 16:32:44,172] DEBUG: Recieved message to update progress: progress_heatup 23
[2021-02-08 16:32:45,213] DEBUG: Recieved message to update progress: progress_heatup 25
[2021-02-08 16:32:46,254] DEBUG: Recieved message to update progress: progress_heatup 26
[2021-02-08 16:32:47,192] DEBUG: Recieved message to update progress: progress_heatup 29
[2021-02-08 16:32:48,232] DEBUG: Recieved message to update progress: progress_heatup 30
[2021-02-08 16:32:49,175] DEBUG: Recieved message to update progress: progress_heatup 33
[2021-02-08 16:32:50,216] DEBUG: Recieved message to update progress: progress_heatup 34
[2021-02-08 16:32:51,256] DEBUG: Recieved message to update progress: progress_heatup 36
[2021-02-08 16:32:52,193] DEBUG: Recieved message to update progress: progress_heatup 38
[2021-02-08 16:32:53,234] DEBUG: Recieved message to update progress: progress_heatup 40
[2021-02-08 16:32:54,177] DEBUG: Recieved message to update progress: progress_heatup 42
[2021-02-08 16:32:55,217] DEBUG: Recieved message to update progress: progress_heatup 44
[2021-02-08 16:32:56,258] DEBUG: Recieved message to update progress: progress_heatup 46
[2021-02-08 16:32:57,198] DEBUG: Recieved message to update progress: progress_heatup 48
[2021-02-08 16:32:58,238] DEBUG: Recieved message to update progress: progress_heatup 50
[2021-02-08 16:32:59,186] DEBUG: Recieved message to update progress: progress_heatup 52
[2021-02-08 16:33:00,222] DEBUG: Recieved message to update progress: progress_heatup 54
[2021-02-08 16:33:01,262] DEBUG: Recieved message to update progress: progress_heatup 56
[2021-02-08 16:33:02,198] DEBUG: Recieved message to update progress: progress_heatup 57
[2021-02-08 16:33:03,238] DEBUG: Recieved message to update progress: progress_heatup 59
[2021-02-08 16:33:04,175] DEBUG: Recieved message to update progress: progress_heatup 61
[2021-02-08 16:33:05,215] DEBUG: Recieved message to update progress: progress_heatup 63
[2021-02-08 16:33:06,255] DEBUG: Recieved message to update progress: progress_heatup 65
[2021-02-08 16:33:07,191] DEBUG: Recieved message to update progress: progress_heatup 66
[2021-02-08 16:33:08,231] DEBUG: Recieved message to update progress: progress_heatup 68
[2021-02-08 16:33:09,272] DEBUG: Recieved message to update progress: progress_heatup 70
[2021-02-08 16:33:10,208] DEBUG: Recieved message to update progress: progress_heatup 71
[2021-02-08 16:33:11,248] DEBUG: Recieved message to update progress: progress_heatup 73
[2021-02-08 16:33:12,185] DEBUG: Recieved message to update progress: progress_heatup 75
[2021-02-08 16:33:13,226] DEBUG: Recieved message to update progress: progress_heatup 76
[2021-02-08 16:33:14,267] DEBUG: Recieved message to update progress: progress_heatup 78
[2021-02-08 16:33:15,202] DEBUG: Recieved message to update progress: progress_heatup 80
[2021-02-08 16:33:16,241] DEBUG: Recieved message to update progress: progress_heatup 81
[2021-02-08 16:33:17,183] DEBUG: Recieved message to update progress: progress_heatup 82
[2021-02-08 16:33:18,223] DEBUG: Recieved message to update progress: progress_heatup 84
[2021-02-08 16:33:19,264] DEBUG: Recieved message to update progress: progress_heatup 86
[2021-02-08 16:33:20,200] DEBUG: Recieved message to update progress: progress_heatup 87
[2021-02-08 16:33:21,244] DEBUG: Recieved message to update progress: progress_heatup 89
[2021-02-08 16:33:22,180] DEBUG: Recieved message to update progress: progress_heatup 90
[2021-02-08 16:33:23,220] DEBUG: Recieved message to update progress: progress_heatup 91
[2021-02-08 16:33:24,260] DEBUG: Recieved message to update progress: progress_heatup 92
[2021-02-08 16:33:25,196] DEBUG: Recieved message to update progress: progress_heatup 94
[2021-02-08 16:33:26,241] DEBUG: Recieved message to update progress: progress_heatup 96
[2021-02-08 16:33:27,178] DEBUG: Recieved message to update progress: progress_heatup 97
[2021-02-08 16:33:28,218] DEBUG: Recieved message to update progress: progress_heatup 98
[2021-02-08 16:33:29,259] DEBUG: Recieved message to update progress: progress_heatup 99
[2021-02-08 16:33:30,194] DEBUG: Recieved message to update progress: progress_heatup 101
[2021-02-08 16:33:31,236] DEBUG: Recieved message to update progress: progress_heatup 102
[2021-02-08 16:33:32,181] DEBUG: Recieved message to update progress: progress_heatup 103
[2021-02-08 16:33:34,263] DEBUG: Recieved message to update progress: progress_heatup 104
[2021-02-08 16:33:39,255] DEBUG: Recieved message to update progress: progress_heatup 103
[2021-02-08 16:33:41,231] DEBUG: Recieved message to update progress: progress_heatup 102
[... cut ...]
[2021-02-08 16:34:07,260] DEBUG: Recieved message to update progress: progress_heatup 100
[2021-02-08 16:34:09,237] DEBUG: Recieved message to update progress: progress_heatup 99
[2021-02-08 16:34:09,446] DEBUG: Recieved message to update progress: progress_print 0
[2021-02-08 16:35:39,903] DEBUG: Recieved message to update progress: progress_print 1
[2021-02-08 16:37:22,027] DEBUG: Recieved message to update progress: progress_print 2
[2021-02-08 16:38:51,603] DEBUG: Recieved message to update progress: progress_print 3

cp2004, 2021-02-08:
So it's.... working?
> it mentioned MOSI on pin 19. Moved the DIN lead from pin 10 to pin 19

I think this may be confusing between board numbering and BCM numbering. This is the pin we are talking about: https://pinout.xyz/pinout/pin19_gpio10

I'll clarify this in the docs on https://cp2004.gitbook.io/ws281x-led-status/guides/setup-guide-1/wiring-your-leds to note it is BCM numbering. The diagrams should all show that pin as well, just the difference between 'pin 10' and 'GPIO 10' in most guides.

cp2004, 2021-02-08:
As of 6e33d4e3c9729dc3da8234c5060003461ae1d9ae this has been clarified with an extra hint.

nigelboubert, 2021-02-08:
I don't know how many times I read that wrong.   Lol.   Clarification in the Docs looks good.

Thanks for your help.   When is the update coming out?

cp2004, 2021-02-08:
> When is the update coming out?

If you mean the next update, 0.7.0, good question. It's done when it's done 🙂. As you can see from the issues/[milestone](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/milestone/6), there's a lot that will go into it, I have about half done I reckon.

nigelboubert, 2021-02-08:
I see that.   Personally I would work on the bugs and release as 6.5.    lol    Just from my days as a mainframe programmer.

Anyway, I love the plugin.  Now to mount everything  in the enclosure.

BTW, one thing I found out.  If you delete the debug log, it doesn't seem to start a new one until the next restart.  Or at least Octoprint doesn't show it.

If you ever need beta testers, I'ld be willing to be a guinea pig.  Not much else to do in retirement.    lol

cp2004, 2021-03-11:
0.7.0 has been released - sorry, forgot about this issue to ping you for a beta test! Set the release channel in Software Update to release candidate to be automatically notified of the next release candidate when it comes out.

closed 2021-03-11 by cp2004

nigelboubert, 2021-03-12:
Charlie,
No worries,  My printer has been out of action for a bit due to reorganize the basement.  Maybe next time.