cp2004/OctoPrint-WS281x_LED_Status #95: [Bug] The runner process crashes when using the gradient effect for heating and cooling

samwiseg0, 2021-02-04:
**Description of the bug**
The runner process crashes when using the gradient effect. 

**Steps to reproduce**

1. Change heating and cooling progress effects to Gradient
2. Start or end a print
3. Let the progress get to 100%
4. See error

**Logs and screenshots**
```
[2021-02-04 14:34:12,836] DEBUG: Current settings:
 | * STRIP SETTINGS *
 | - count: 77
 | - pin: 10
 | - freq_hz: 800000
 | - dma: 10
 | - invert: False
 | - channel: 0
 | - reverse: False
 | - type: SK6812_STRIP_GRBW
 | - brightness: 50
 | - adjustment: {'R': 100, 'G': 100, 'B': 100}
 | - white_override: True
 | - white_brightness: 100
 | * EFFECT SETTINGS *
 | startup
 | - enabled: True
 | - effect: Color Wipe
 | - color: #00ff00
 | - delay: 75
 | idle
 | - enabled: True
 | - effect: Color Wipe 2
 | - color: #00eeff
 | - delay: 75
 | disconnected
 | - enabled: True
 | - effect: Bounce
 | - color: #ff0000
 | - delay: 25
 | failed
 | - enabled: True
 | - effect: Pulse
 | - color: #ff0000
 | - delay: 10
 | success
 | - enabled: True
 | - effect: Rainbow Cycle
 | - color: #000000
 | - delay: 25
 | - return_to_idle: 600
 | paused
 | - enabled: True
 | - effect: Bounce
 | - color: #0000ff
 | - delay: 40
 | printing
 | - enabled: True
 | - effect: Solid Color
 | - color: #ffffff
 | - delay: 1
 | torch
 | - enabled: True
 | - effect: Solid Color
 | - color: #ffffff
 | - delay: 1
 | - toggle: True
 | - timer: 15
 | - auto_on_webcam: True
 | progress_print
 | - enabled: False
 | - base: #000000
 | - color: #00ff00
 | - effect: Progress Bar
 | progress_heatup
 | - enabled: True
 | - base: #0000ff
 | - color: #ff0000
 | - tool_enabled: True
 | - bed_enabled: True
 | - tool_key: 0
 | - effect: Gradient
 | progress_cooling
 | - enabled: True
 | - base: #0000ff
 | - color: #ff0000
 | - bed_or_tool: tool
 | - threshold: 30
 | - effect: Gradient
 | * ACTIVE TIMES *
 | - enabled: False
 | - start: 09:00
 | - end: 21:00
 ```

Traceback:
```
[2021-02-04 18:33:51,299] ERROR: Unhandled exception in effect runner process
[2021-02-04 18:33:51,346] ERROR: in method 'ws2811_led_set', argument 3 of type 'uint32_t'
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_ws281x_led_status/runner.py", line 104, in main_loop
    self.parse_q_msg(msg)  # Effects are run from parse_q_msg
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_ws281x_led_status/runner.py", line 122, in parse_q_msg
    self.progress_msg(msg)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_ws281x_led_status/runner.py", line 162, in progress_msg
    self.progress_effect(msg_split[0], float(msg_split[1]))
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_ws281x_led_status/runner.py", line 232, in progress_effect
    reverse=self.reverse,
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_ws281x_led_status/effects/progress.py", line 62, in gradient
    strip.setPixelColorRGB(i, *color)
  File "/home/pi/oprint/lib/python3.7/site-packages/rpi_ws281x/rpi_ws281x.py", line 150, in setPixelColorRGB
    self.setPixelColor(n, Color(red, green, blue, white))
  File "/home/pi/oprint/lib/python3.7/site-packages/rpi_ws281x/rpi_ws281x.py", line 143, in setPixelColor
    self._led_data[n] = color
  File "/home/pi/oprint/lib/python3.7/site-packages/rpi_ws281x/rpi_ws281x.py", line 54, in __setitem__
    return ws.ws2811_led_set(self.channel, pos, value)
OverflowError: in method 'ws2811_led_set', argument 3 of type 'uint32_t'
```

**Version of plugin and OctoPrint**

latest `devel-c2d6aecce51dd80c5a1c462a6ed0d5c2fb3b731e`

**(If applicable) Browser version & browser error console**

N/A

**Additional context**
N/A

cp2004, 2021-02-08:
I think I got it... The `blend_two_colors` doesn't min/max it between 0-255:
```python
>>>blend_two_colors((255, 0, 0), (0, 0, 255), float(101) / 100)
(128, 0, -1)
```

cp2004, 2021-02-08:
Above commit should fix it, if you pull devel again it should be there. I'm going to add some tests for this now as well.

samwiseg0, 2021-02-08:
Ah cool! I was working on trying to replicate it but I was having a hard time and was thinking the virtual printer would not act the same as you just pointed out ;)

Secondly. I may need to put in another issue as in part of my testing if you send prints back to back without letting the cool down or heat up effects run in their entirety then it breaks the printing effect. it acts like it never clears out of the heating or cooling effect even if it its 100%. This maybe solved in part by what we talked about in sending the heating command at the start of a print. I believe there maybe some need for a reset function to clear out any other effects at the start of a new print

cp2004, 2021-02-08:
I take that back, I did this in the wrong place 🤦‍♂️

samwiseg0, 2021-02-08:
> Above commit should fix it, if you pull devel again it should be there. I'm going to add some tests for this now as well.

I will pull it in a few hours when my printer is done. Thanks!

cp2004, 2021-02-10:
Going to close this one, since it was just in devel and is now solved.

closed 2021-02-10 by cp2004

samwiseg0, 2021-02-10:
Can confirm this is fixed!

```
[2021-02-10 22:00:09,948] DEBUG: Changing effect to printing
[2021-02-10 22:01:44,671] DEBUG: Received message to update progress: progress_heatup 99
[2021-02-10 22:01:48,589] DEBUG: Received message to update progress: progress_heatup 98
[2021-02-10 22:01:48,694] DEBUG: Received message to update progress: progress_heatup 99
[2021-02-10 22:01:49,658] DEBUG: Received message to update progress: progress_heatup 98
[2021-02-10 22:01:51,658] DEBUG: Received message to update progress: progress_heatup 99
[2021-02-10 22:01:54,589] DEBUG: Received message to update progress: progress_heatup 100
[2021-02-10 22:01:56,597] DEBUG: Received message to update progress: progress_heatup 101
[2021-02-10 22:01:58,017] DEBUG: Changing effect to printing
```

commit 5cd998d7 (2021-02-08): :bug: Fix issue with blending colours going below 0

Should fix #95
commit 3184cee3 (2021-02-08): :bug: Actually fix #95, in blending colours