cp2004/OctoPrint-WS281x_LED_Status #103: [Bug] Raspberry Pi 4 - no pixels light up at all with plugin - ws2811_init failed with code -3 (Hardware revision is not supported)

rullywowpcb, 2021-02-09:
Uploaded log files
plugin_ws281x_led_status_debug.log

`[2021-02-09 18:27:09,606] INFO: Initialising LED strip
[2021-02-09 18:27:09,608] ERROR: Strip failed to initialize, no effects will be run.
[2021-02-09 18:27:09,608] ERROR: Please check your settings.
[2021-02-09 18:27:09,608] ERROR: Here's the exception: ws2811_init failed with code -3 (Hardware revision is not supported)
[2021-02-09 18:27:09,608] INFO: No strip initialised, exiting the effect process.
[2021-02-09 18:29:22,971] INFO: Initialising LED strip
[2021-02-09 18:29:22,973] ERROR: Strip failed to initialize, no effects will be run.
[2021-02-09 18:29:22,973] ERROR: Please check your settings.
[2021-02-09 18:29:22,973] ERROR: Here's the exception: ws2811_init failed with code -3 (Hardware revision is not supported)
[2021-02-09 18:29:22,973] INFO: No strip initialised, exiting the effect process.
[2021-02-09 18:29:59,292] INFO: Initialising LED strip
[2021-02-09 18:29:59,294] ERROR: Strip failed to initialize, no effects will be run.
[2021-02-09 18:29:59,294] ERROR: Please check your settings.
[2021-02-09 18:29:59,295] ERROR: Here's the exception: ws2811_init failed with code -3 (Hardware revision is not supported)
[2021-02-09 18:29:59,295] INFO: No strip initialised, exiting the effect process.
[2021-02-09 18:31:56,296] INFO: Initialising LED strip
[2021-02-09 18:31:56,298] ERROR: Strip failed to initialize, no effects will be run.
[2021-02-09 18:31:56,298] ERROR: Please check your settings.
[2021-02-09 18:31:56,298] ERROR: Here's the exception: ws2811_init failed with code -3 (Hardware revision is not supported)
[2021-02-09 18:31:56,298] INFO: No strip initialised, exiting the effect process.
[2021-02-09 18:40:29,977] INFO: Initialising LED strip
[2021-02-09 18:40:29,980] ERROR: Strip failed to initialize, no effects will be run.
[2021-02-09 18:40:29,980] ERROR: Please check your settings.
[2021-02-09 18:40:29,980] ERROR: Here's the exception: ws2811_init failed with code -3 (Hardware revision is not supported)
[2021-02-09 18:40:29,980] INFO: No strip initialised, exiting the effect process.
[2021-02-09 19:07:44,582] INFO: Initialising LED strip
[2021-02-09 19:07:44,584] ERROR: Strip failed to initialize, no effects will be run.
[2021-02-09 19:07:44,585] ERROR: Please check your settings.
[2021-02-09 19:07:44,585] ERROR: Here's the exception: ws2811_init failed with code -3 (Hardware revision is not supported)
[2021-02-09 19:07:44,585] INFO: No strip initialised, exiting the effect process.`

octoprint.log

`2021-02-09 18:15:14,767 - octoprint.plugins.discovery - INFO - Registered 'OctoPrint instance on octopi._http._tcp.local.' for _http._tcp.local.
2021-02-09 18:15:14,777 - octoprint.server - INFO - --- Log roll over detected ---------------------------------------------------
2021-02-09 18:15:14,777 - octoprint.server - INFO - OctoPrint 1.5.2
2021-02-09 18:15:14,779 - octoprint.plugin.core - INFO - 17 plugin(s) registered with the system:
|  Action Command Notification Support (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/action_command_notification
|  Action Command Prompt Support (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/action_command_prompt
|  Announcement Plugin (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/announcements
|  Anonymous Usage Tracking (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/tracking
|  Application Keys Plugin (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/appkeys
|  Backup & Restore (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/backup
|  Core Wizard (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/corewizard
|  Discovery (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/discovery
|  Error Tracking (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/errortracking
|  File Check (2020.8.7) (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_file_check
|  Firmware Check (2020.9.23) (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_firmware_check
|  GCode Viewer (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/gcodeviewer
|  Logging (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/logging
|  Pi Support Plugin (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/pi_support
|  Plugin Manager (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/pluginmanager
|  Software Update (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/softwareupdate
|  Virtual Printer (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/virtual_printer
Prefix legend: ! = disabled, # = blacklisted, * = incompatible
2021-02-09 18:15:14,785 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2021-02-09 18:15:14,800 - octoprint.environment - INFO - Detected environment is Python 3.7.3 under Linux (linux). Details:
|  hardware:
|      cores: 4
|      freq: 1500.0
|      ram: 1903988736
|  os:
|      bits: 32
|      id: linux
|      platform: linux
|  plugins:
|      pi_support:
[... 1123 lines cut ...]
2021-02-09 19:07:44,561 - octoprint.plugins.ws281x_led_status - INFO - Ws281x LED Status runner started
2021-02-09 19:08:01,417 - tornado.access - WARNING - 403 GET /api/printer (::ffff:192.168.1.40) 8.18ms
2021-02-09 19:08:01,438 - tornado.access - WARNING - 403 GET /api/job (::ffff:192.168.1.40) 7.92ms
2021-02-09 19:08:01,459 - tornado.access - WARNING - 403 GET /api/job (::ffff:192.168.1.40) 7.80ms
2021-02-09 19:08:01,477 - tornado.access - WARNING - 403 GET /api/job (::ffff:192.168.1.40) 5.12ms
2021-02-09 19:08:01,593 - tornado.access - WARNING - 403 GET /api/printer (::ffff:192.168.1.40) 8.15ms
2021-02-09 19:08:01,611 - tornado.access - WARNING - 403 GET /api/printer (::ffff:192.168.1.40) 6.87ms
2021-02-09 19:08:31,522 - tornado.access - WARNING - 403 GET /api/printer (::ffff:192.168.1.40) 7.86ms
2021-02-09 19:08:31,542 - tornado.access - WARNING - 403 GET /api/job (::ffff:192.168.1.40) 7.93ms
2021-02-09 19:08:31,562 - tornado.access - WARNING - 403 GET /api/job (::ffff:192.168.1.40) 7.89ms
2021-02-09 19:08:31,582 - tornado.access - WARNING - 403 GET /api/job (::ffff:192.168.1.40) 7.96ms
2021-02-09 19:08:31,591 - tornado.access - WARNING - 403 GET /api/printer (::ffff:192.168.1.40) 5.04ms
2021-02-09 19:08:31,609 - tornado.access - WARNING - 403 GET /api/printer (::ffff:192.168.1.40) 6.28ms
`

**Version of plugin and OctoPrint**
OctoPrint: Version 1.5.2
WS281x_LED_Status Version: 0.6.0

**(If applicable) Browser version & browser error console**
n/a

**Additional context**

Strip works fine on same GPIO pin (changing nothing hardware wise) when manually running a test python sketch. I think something broke in the upgrade to python3? Thanks in advance.

cp2004, 2021-02-09:
Please see [this troubleshooting guide entry](https://cp2004.gitbook.io/ws281x-led-status/troubleshooting-guide#no-led-output-on-a-raspberry-pi-4b-rev-1-4-and-the-log-says-unsupported-board) and issue #73 for a solution.

rullywowpcb, 2021-02-09:
This fix worked perfectly!

Thank you for the great plugin (one of my favorites for Octoprint) and the excellent support. Anyone with a Raspberry Pi 4 having issues, follow the fix above until it gets merged in the future. Closing issue.

closed 2021-02-09 by rullywowpcb