cp2004/OctoPrint-WS281x_LED_Status #105: Cooling effect will not end if new print is started before cooldown

cp2004, 2021-02-10:
Secondly. I may need to put in another issue as in part of my testing if you send prints back to back without letting the cool down or heat up effects run in their entirety then it breaks the printing effect. it acts like it never clears out of the heating or cooling effect even if it its 100%. This maybe solved in part by what we talked about in sending the heating command at the start of a print. I believe there maybe some need for a reset function to clear out any other effects at the start of a new print

_Originally posted by @samwiseg0 in https://github.com/cp2004/OctoPrint-WS281x_LED_Status/issues/95#issuecomment-775409358_

cp2004, 2021-02-10:
Sorry bot I edited the title... but not fast enough 🙁

cp2004, 2021-02-10:
OK now I'm confused, I'm sure I looked at this code and thought of course it wouldn't work, now I'm not sure...

cp2004, 2021-02-10:
https://pastebin.com/F2QtVu7b

cp2004, 2021-02-10:
This is the only place I can find that `cooling = True` is being set, but it sure enough flicks that way when it shouldn't...

https://github.com/cp2004/OctoPrint-WS281x_LED_Status/blob/36243ab0191ffe5eca9f31dc651db236d1841e7d/octoprint_ws281x_led_status/__init__.py#L375

```
2021-02-10 16:35:45,165 - octoprint.plugins.ws281x_led_status - INFO - progress_cooling: True
2021-02-10 16:35:47,167 - octoprint.plugins.ws281x_led_status - INFO - progress_cooling: True
2021-02-10 16:35:49,171 - octoprint.plugins.ws281x_led_status - INFO - progress_cooling: True
2021-02-10 16:35:51,173 - octoprint.plugins.ws281x_led_status - INFO - progress_cooling: True
2021-02-10 16:35:53,175 - octoprint.plugins.ws281x_led_status - INFO - progress_cooling: True
2021-02-10 16:35:55,179 - octoprint.plugins.ws281x_led_status - INFO - progress_cooling: True
2021-02-10 16:35:57,181 - octoprint.plugins.ws281x_led_status - INFO - progress_cooling: True
2021-02-10 16:35:59,183 - octoprint.plugins.ws281x_led_status - INFO - progress_cooling: True
2021-02-10 16:36:00,342 - octoprint.util.comm - INFO - Starting job on behalf of user Charlie
2021-02-10 16:36:00,345 - octoprint.util.comm - INFO - Changing monitoring state from "Operational" to "Starting"
2021-02-10 16:36:00,350 - octoprint.events.fire - DEBUG - Firing event: PrinterStateChanged (Payload: {'state_id': 'STARTING', 'state_string': 'Starting'})
2021-02-10 16:36:00,351 - octoprint.events - DEBUG - Sending action to <bound method PrinterStateConnection._onEvent of <octoprint.server.util.sockjs.PrinterStateConnection object at 0xaa54a710>>
2021-02-10 16:36:00,353 - octoprint.plugin - DEBUG - Calling on_event on action_command_notification
2021-02-10 16:36:00,354 - octoprint.plugin - DEBUG - Calling on_event on action_command_prompt
2021-02-10 16:36:00,354 - octoprint.plugin - DEBUG - Calling on_event on announcements
2021-02-10 16:36:00,355 - octoprint.plugin - DEBUG - Calling on_event on file_check
2021-02-10 16:36:00,355 - octoprint.plugin - DEBUG - Calling on_event on firmware_check
2021-02-10 16:36:00,356 - octoprint.plugin - DEBUG - Calling on_event on pluginmanager
2021-02-10 16:36:00,356 - octoprint.plugin - DEBUG - Calling on_event on softwareupdate
2021-02-10 16:36:00,357 - octoprint.plugin - DEBUG - Calling on_event on tracking
2021-02-10 16:36:00,360 - octoprint.plugin - DEBUG - Calling on_event on ws281x_led_status
2021-02-10 16:36:00,366 - octoprint.printer.standard.job - INFO - Print job started - origin: local, path: xyzCalibration_cube.gcode, owner: Charlie, user: Charlie
2021-02-10 16:36:00,367 - octoprint.events.fire - DEBUG - Firing event: PrintStarted (Payload: {'name': 'xyzCalibration_cube.gcode', 'path': 'xyzCalibration_cube.gcode', 'origin': 'local', 'size': 304310, 'owner': 'Charlie', 'user': 'Charlie'})
2021-02-10 16:36:00,375 - octoprint.events - DEBUG - Sending action to <bound method PrinterStateConnection._onEvent of <octoprint.server.util.sockjs.PrinterStateConnection object at 0xaa54a710>>
2021-02-10 16:36:00,398 - octoprint.plugin - DEBUG - Calling on_event on action_command_notification
2021-02-10 16:36:00,405 - octoprint.plugin - DEBUG - Calling on_event on action_command_prompt
2021-02-10 16:36:00,421 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2021-02-10 16:36:00,424 - octoprint.plugin - DEBUG - Calling on_event on announcements
2021-02-10 16:36:00,435 - octoprint.plugin - DEBUG - Calling on_event on file_check
2021-02-10 16:36:00,436 - tornado.access - INFO - 204 POST /api/job (::ffff:192.168.1.173) 127.64ms
2021-02-10 16:36:00,439 - octoprint.util.comm - INFO - Changing monitoring state from "Starting" to "Printing"
2021-02-10 16:36:00,440 - octoprint.plugin - DEBUG - Calling on_event on firmware_check
2021-02-10 16:36:00,441 - tornado.access - INFO - 200 POST /api/job (::ffff:192.168.1.173) 132.91ms
2021-02-10 16:36:00,443 - octoprint.filemanager.analysis - DEBUG - Pausing analysis
2021-02-10 16:36:00,444 - octoprint.plugin - DEBUG - Calling on_event on pluginmanager
2021-02-10 16:36:00,450 - octoprint.plugin - DEBUG - Calling on_event on softwareupdate
2021-02-10 16:36:00,461 - octoprint.plugin - DEBUG - Calling on_event on tracking
2021-02-10 16:36:00,505 - octoprint.plugin - DEBUG - Calling on_event on ws281x_led_status
2021-02-10 16:36:00,505 - octoprint.events.fire - DEBUG - Firing event: PrinterStateChanged (Payload: {'state_id': 'PRINTING', 'state_string': 'Printing'})
2021-02-10 16:36:00,506 - octoprint.events - DEBUG - Sending action to <bound method PrinterStateConnection._onEvent of <octoprint.server.util.sockjs.PrinterStateConnection object at 0xaa54a710>>
2021-02-10 16:36:00,515 - octoprint.plugin - DEBUG - Calling on_event on action_command_notification
2021-02-10 16:36:00,518 - octoprint.plugin - DEBUG - Calling on_event on action_command_prompt
2021-02-10 16:36:00,534 - octoprint.plugin - DEBUG - Calling on_event on announcements
2021-02-10 16:36:00,543 - octoprint.plugin - DEBUG - Calling on_event on file_check
2021-02-10 16:36:00,550 - octoprint.plugin - DEBUG - Calling on_event on firmware_check
2021-02-10 16:36:00,555 - octoprint.plugin - DEBUG - Calling on_event on pluginmanager
2021-02-10 16:36:00,560 - octoprint.plugin - DEBUG - Calling on_event on softwareupdate
2021-02-10 16:36:00,562 - octoprint.plugins.ws281x_led_status - INFO - progress_cooling: False
2021-02-10 16:36:00,563 - octoprint.plugin - DEBUG - Calling on_event on tracking
[... cut ...]
2021-02-10 16:36:05,630 - octoprint.plugins.ws281x_led_status - INFO - progress_cooling: True
2021-02-10 16:36:06,631 - octoprint.plugins.ws281x_led_status - INFO - progress_cooling: True
2021-02-10 16:36:07,198 - octoprint.plugins.ws281x_led_status - INFO - progress_cooling: True
2021-02-10 16:36:07,631 - octoprint.plugins.ws281x_led_status - INFO - progress_cooling: True
2021-02-10 16:36:08,633 - octoprint.plugins.ws281x_led_status - INFO - progress_cooling: True
```

The hunt will continue later, this is just here so I don't forget.

samwiseg0, 2021-02-10:
Ah. I was starting to look at the code. Glad you have something to go off of ;)

cp2004, 2021-02-10:
This implementation of a previous event queue is bugged in general, it's fundamentally broken as far as I can see - a previous @cp2004 didn't think this through enough...

cp2004, 2021-02-10:
With the above commit I believe it should be solved, it seems to work well in my stress testing of it.

samwiseg0, 2021-02-10:
Sweet! I will update my install in about an hour when my print is done and test it out! Thanks for fixing it so quick!

cp2004, 2021-02-10:
Now you say that... I'll see if I can add a section to the software update hook to follow development branches. Might make life a bit easier.

samwiseg0, 2021-02-10:
That would be great as the update modal is very annoying since it thinks there is an update on master even though there is not

cp2004, 2021-02-10:
You can mute updates for individual plugins, by pressing the bell icon in the software update panel. I'm not sure how easy it would work, since it looks like the options are `github_release` *or* `github_commit`. I'll keep on looking.

cp2004, 2021-02-10:
`github_commit` doesn't work in practice, since Github doesn't provide the git metadata when downloading from a branch zip. You have to record current commit, which of course you don't know until after you've commit. Could be grabbed from the git metadata, but that's not provided by Github. Easiest idea is probably to check it out locally and install using `pip install -e .`.

commit 8cc6f8fd (2021-02-10): :bug: Process previous effect, not event

Fixes #105

Previously, the previous event would be saved. This caused multiple issues where additonal hooks for the event were processed, so things like cooling tracking would be restarted when heating progress was finished, if that was the last event to come through.

Now, it records the effect, rather than event that should be run, and bypasses the additonal event handling. Much better...
closed 2021-03-11 by cp2004