cp2004/OctoPrint-WS281x_LED_Status #107: [Bug] [devel] Solid color printing effect causes high CPU usage

samwiseg0, 2021-02-11:
**Description of the bug**

Enabling the "printing effect" as a solid color causes high CPU usage. I did some testing and it seems that the `apply_color_correction()` function is constantly being called when a solid color is set. I added some logging to that function and for some reason it is being called in a loop. Turning off the lights sets the CPU back to what you would expect. 
 
**Steps to reproduce**

1. Enable printing effect as a solid color
2. start a print job
3. Monitor that the CPU usage is locking one CPU to 100%

**What did you expect to happen**
For it not to use 100% CPU.

**Logs and screenshots**
```
[2021-02-11 01:15:59,162] DEBUG: Current settings:
 | * STRIP SETTINGS *
 | - count: 77
 | - pin: 10
 | - freq_hz: 800000
 | - dma: 10
 | - invert: False
 | - channel: 0
 | - reverse: False
 | - type: SK6812_STRIP_GRBW
 | - brightness: 50
 | - adjustment: {'R': 100, 'G': 100, 'B': 100}
 | - white_override: True
 | - white_brightness: 100
 | * EFFECT SETTINGS *
 | startup
 | - enabled: True
 | - effect: Color Wipe
 | - color: #00ff00
 | - delay: 75
 | idle
 | - enabled: True
 | - effect: Color Wipe 2
 | - color: #00eeff
 | - delay: 75
 | disconnected
 | - enabled: True
 | - effect: Bounce
 | - color: #ff0000
 | - delay: 25
 | failed
 | - enabled: True
 | - effect: Pulse
 | - color: #ff0000
 | - delay: 10
 | success
 | - enabled: True
 | - effect: Rainbow Cycle
 | - color: #000000
 | - delay: 25
 | - return_to_idle: 600
 | paused
 | - enabled: True
 | - effect: Bounce
 | - color: #0000ff
 | - delay: 40
 | printing
 | - enabled: True
 | - effect: Solid Color
 | - color: #ffffff
 | - delay: 25
 | torch
 | - enabled: True
 | - effect: Solid Color
 | - color: #ffffff
 | - delay: 1
 | - toggle: True
 | - timer: 15
 | - auto_on_webcam: False
 | progress_print
 | - enabled: False
 | - base: #000000
 | - color: #00ff00
 | - effect: Progress Bar
 | progress_heatup
 | - enabled: True
 | - base: #0000ff
 | - color: #ff0000
 | - tool_enabled: True
 | - bed_enabled: True
 | - tool_key: 0
 | - effect: Gradient
 | progress_cooling
 | - enabled: True
 | - base: #0000ff
 | - color: #ff0000
 | - bed_or_tool: tool
 | - threshold: 30
 | - effect: Gradient
 | * ACTIVE TIMES *
 | - enabled: False
 | - start: 09:00
 | - end: 21:00
 ```
**Version of plugin and OctoPrint**

`devel-6b0c7aa98bbd68b39b1bc6d879944c42602274b5`

**(If applicable) Browser version & browser error console**
N/A

cp2004, 2021-02-11:
Hmm... I think I know why this is happening, but I thought I would have mitigated against it with the commit that broke it. Effects should be run in a thread now, and have complete control of their loop (actually a slight increase in performance, since less code is run...) instead of being restarted every time they go around. 

Will look into it, this is why I don't recommend people run the development branch.... At least you know what you're doing.

samwiseg0, 2021-02-11:
Ha! I know the risks. Thank you for looking into it!

cp2004, 2021-02-11:
Fixed with the above commit - a simple typo when refactoring and instead of `q_poll_sleep` I used `q_poll_milli_sleep` and so it was running 1000 times more often than it should have.

cp2004, 2021-02-11:
Colour wipe 2 was also fixed with 7708109aa6890c609472635ec497107bb7dbe319 - a regression from the same commit.

samwiseg0, 2021-02-11:
Thank you! I will test and post back in a few hours when my printer is free.

samwiseg0, 2021-02-12:
Confirmed fixed! Once again thank you for the fantastic support!

commit c0497f42 (2021-02-11): :bug: Fix high CPU usage on solid colour effects

Regression introduced in 95ea39564fb89e12a7195d211f31fdaeca4a4447

Closes #107
closed 2021-02-12 by samwiseg0