cp2004/OctoPrint-WS281x_LED_Status #108: Turn Off with PSU Shutdown

SaKiEQ, 2021-02-14:
If possible, a shutdown feature when the PSU powers off. 

Not sure if there is a workaround with the ActionCommands plugin and M150 commands, but that would become a 
bit convoluted. 

Similar to the time of the day feature, just 'if PSU-off' then 'lights off'

cp2004, 2021-02-14:
How can I detect that the PSU is off? Please provide some more details as to how this would work.

You can use @ commands and add the command `@WS_LIGHTSOFF` command to any gcode script, from the slicer, other plugins, or from custom buttons etc. This would be a far easier solution than trying to work out if the PSU is on or not from this plugin, which has no idea of the 

[@ commands documentation](https://cp2004.gitbook.io/ws281x-led-status/documentation/host-commands)

*Note:* The bot is still under development, the feature request template should probably be clearer that it needs to find `request` in the title.

Hypertoken, 2021-03-08:
Hey, 
I came here to request this same feature! However, @cp2004 makes a great point; I'm using the PSU Control plugin. So, I've just added  `@WS_LIGHTSON` to "Post On GCode Script" and `@WS_LIGHTSOFF` to ""Post Off GCode Script".

Hope this helps someone.

cp2004, 2021-03-13:
Closing in favour of #125, defining custom triggers. Early days but I intend for it to be in the next release.

closed 2021-03-13 by cp2004