cp2004/OctoPrint-WS281x_LED_Status #109: Sacrificial pixel (led) support

gladiatormax, 2021-02-15:
I use the first led for the 3.3v -> 5v data signal. Poor man's signal shifter.
See example here,

[https://hackaday.com/2017/01/20/cheating-at-5v-ws2812-control-to-use-a-3-3v-data-line/](https://hackaday.com/2017/01/20/cheating-at-5v-ws2812-control-to-use-a-3-3v-data-line/)

Could you add the option to ingnore first pixel(led) as also buing used in de wled project.

cp2004, 2021-02-15:
I probably could, not for the next update (since that needs to be shipped at some point) but I will add it to the backlog.

gladiatormax, 2021-02-15:
> I probably could, not for the next update (since that needs to be shipped at some point) but I will add it to the backlog.

Thx , for the rest great work on the plugin !

cp2004, 2021-07-04:
This has been implemented ready for v0.8.0

cp2004, 2022-02-04:
0.8.0 has been released (finally...!)

commit aedfd80f (2021-07-04): :sparkles: Sacrificial first pixel support

Closes #109

Also fixed a bug in segment implementation
closed 2022-02-04 by cp2004