cp2004/OctoPrint-WS281x_LED_Status #111: [Feature Request] Torch Icon visibility

roboraptor, 2021-02-17:
**Issue**

When using dark themes of octoprint, Torch icon is still black and hard to see if on or off.

**Solution**

Easiest
Which css class/id color I need to change in themify to change color of torch icon? 

Medium
Implement light version of icon (to use with darker modes of octoprint)

Hard?
Implement textfield or colorpicker so user can set color of it.

**Why should this be implemented?**

cp2004, 2021-02-17:
This has been on my mind, under experimentation.

> Easiest
Which css class/id color I need to change in themify to change color of torch icon?

It doesn't really work like that, at least I couldn't get it to work. Explanation below

> Medium
Implement light version of icon (to use with darker modes of octoprint)

This, I can do for the next release of the plugin I think, as a stop-gap solution. Provide one black icon and one white icon.

> Hard?
Implement FontAwesome so user can change how the icon looks like.

FontAwesome free doesn't have a torch icon, which is annoying. The icon is only available in Pro, and even if you get a pro license you can't distribute it from an open source repository... Otherwise this would not be an issue as Themeify would already have covered it. I looked for a *long* time trying to work this out before, even asked them if I could buy just the single icon, and they said no.

> Implement textfield or colorpicker so user can set color of it.

This is my long-term goal, but it is harder than it looks. The SVG icon that is currently used means that you can't change the colour using a class or JavaScript easily. I tried, and couldn't get it to work. If I can find a font style icon (rather than SVG) then this will be the first thing I will implement.

---

The other alternative I have had suggested before, was a toggle switch like the one beside the light bulb icon to show if it is on or off.

roboraptor, 2021-02-17:
Thank you for your quick and explanatory answer! 👍 

> Easiest
Ah... yeah svg... 

> Medium
Is it possible to provide more to choose from?

> Hard?
FontAwesome, maybe changing torch to sun, or torch to whatever user specifies from free FA library would be also interesting possibility.

I guess for changing color of svg you need to append it to html as a code and then with something like moderniser do the magic..

also a switch is good but when we are at it,  would it be possible to choose frome at least some set of icons?
Like... if I can send you some 😄

roboraptor, 2021-02-17:
> Medium

Here I append your original SVGs of torch, Colored.
base colors, mixed colors, discoranged orange color (as i like the theme) 

[torch-icon_colors1.zip](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/files/5998596/torch-icon_colors1.zip)

(can produce more if wanted)

roboraptor, 2021-02-17:
> Medium

Here I append your original SVGs of torch, Colored.

- base colors (R,G,B)
- mixed colors (C,M,Y,K)
- discoranged theme orange
- your green and red (green2, red2)

(also added thinner toggle switch in your green and red)

[torch-icon_colors2.zip](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/files/5998955/torch-icon_colors2.zip)

cp2004, 2021-02-17:
Thank you 🙂

I will take a look at them and see if I can get them into the next release.

roboraptor, 2021-02-18:
You are welcome! I love this plugin! 👍  

I investigated a bit and I guess you need to use the SVG properly in that page so it could be able to catch the 
`style=fill:url(#color)` 
attribute from CSS

When you change the whole icon now, you can possibly change the parameter too 😄 
Also I am not sure if I can provide SVG that will accept the passed parameter, 
after reading the documentation i feel more lost than before :D

cp2004, 2021-02-18:
I wonder, if the SVG was inlined rather than an external image, then a colour picker may be possible. Using the SVG you sent:
```svg
<svg xmlns="http://www.w3.org/2000/svg" width="512" height="512" viewBox="0 0 512 512">
<title>ionicons-v5-f-discoranged</title> <!-- probably unnecessary -->
<linearGradient id="yourcolor">
  <stop stop-color="#000" data-bind="attr: {'stop-color': iconColour() }" />
</linearGradient>
<path style="fill:url(#yourcolor);" d=[...] />
<path style="fill:url(#yourcolor);" d=[...] />
</svg>
```
This might work to allow the colour to be changed using knockout and setting the `iconColour()` or similar variables.

roboraptor, 2021-02-18:
Yeah absolutely!

I used linear gradient to set color from one place for multiple strokes and fills.

You should be able to use something like?
```
<svg xmlns="http://www.w3.org/2000/svg" style="display: none">
  <symbol id="my-first-icon" viewBox="0 0 20 20">
    <title>my-first-icon</title>
    <path fill="var(--color-1)" d="..." />
  </symbol>
</svg>
```
and in css something like
```
.icon-colors {
  --color-1: #c13127;
}
```
source: [https://frontstuff.io/multi-colored-svg-symbol-icons-with-css-variables](url)

roboraptor, 2021-03-15:
*patiently waiting*

cp2004, 2021-03-15:
Sorry! I have a lot of things going on in the space, I'm trying to launch a new plugin as well. 0.7.0 for this was released not that long ago, but I was trying to get it out sooner rather than later, I had to draw the line at some point and that was before this issue.

I'm also considering switching the actual icon itself to the one from Material Icons:
[image]

I'll probably use the rounded variant with that one.

It *will* be done, but I don't work with lots of short releases - so it could be a while. I added a 'Development' release channel in 0.7.0, maybe I can use this if you want to test it as soon as it is done.

roboraptor, 2021-03-15:
If the material icon can have then color changed, I dont see any problem in using this. :)

ThorstenWirth, 2021-03-22:
Why does it has to be a torch-icon?
There is already a toggle-switch icon and a lightbulb icon. Just omit the torch icon and use the lightbulb instead.

roboraptor, 2021-03-22:
Because both has different functions?

roboraptor, 2021-03-26:
> It _will_ be done, but I don't work with lots of short releases - so it could be a while. I added a 'Development' release channel in 0.7.0, maybe I can use this if you want to test it as soon as it is done.

Already switched to that channel 👍 
No need to push it to first thing to do, just let us know when you get to it 🥰 

You are awesome!

cp2004, 2021-07-09:
Now it looks something like this:
[image]

*(This was not the final design, but quite close)*

The colour is configurable and I'm hoping that the icon should be clearer than the tiny variations between the two states previously.

cp2004, 2022-02-04:
0.8.0 has been released (finally...!)

commit 36b8fbc3 (2021-07-09): :sparkles: Use modified material icons torch & configurable colour

Closes #111

User can now configure the colour of the torch icon, more compatible with themes etc.
closed 2022-02-04 by cp2004