cp2004/OctoPrint-WS281x_LED_Status #112: WS2812B Not functioning with Octoprint WS281x_LED_Status

Josiahpox, 2021-02-24:
**What do you need help with?**
So I recently attempted to connect WS2812B Individually Addressable LEDs. I've been trying to set up WS281x_LED_Status, as I have heard it is well made and has good support.

I am currently using a 5v 3amp power supply. I used this on an Arduino and it works just fine. However, when I connected to the pi, it no longer sends signals.

I tried to use a shifter (as I read this might work) but it still doesn't work. The only thing I can think of is the long wires I have connected to the pi. I'm not sure what the problem is and I am still learning.
Any help would be greatly appreciated. 

**Log files**

Here are my log files:
* `octoprint.log`: 2021-02-23 05:16:50,450 - tornado.access - WARNING - 409 GET /api/printer (::ffff:192.168.2.180) 10.63ms
2021-02-23 05:16:52,479 - tornado.access - WARNING - 409 GET /api/printer (::ffff:192.168.2.180) 9.18ms
2021-02-23 05:16:54,490 - tornado.access - WARNING - 409 GET /api/printer (::ffff:192.168.2.180) 9.57ms
2021-02-23 05:16:56,497 - tornado.access - WARNING - 409 GET /api/printer (::ffff:192.168.2.180) 9.66ms
2021-02-23 05:16:58,513 - tornado.access - WARNING - 409 GET /api/printer (::ffff:192.168.2.180) 9.09ms
2021-02-23 05:22:43,215 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2021-02-23 05:22:49,697 - octoprint.plugins.tracking - INFO - Sent tracking event ping, payload: {'octoprint_uptime': 94506}
2021-02-23 05:37:43,216 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2021-02-23 05:37:49,695 - octoprint.plugins.tracking - INFO - Sent tracking event ping, payload: {'octoprint_uptime': 95406}
2021-02-23 05:52:43,218 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2021-02-23 05:52:49,683 - octoprint.plugins.tracking - INFO - Sent tracking event ping, payload: {'octoprint_uptime': 96306}
2021-02-23 06:07:43,219 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2021-02-23 06:07:49,698 - octoprint.plugins.tracking - INFO - Sent tracking event ping, payload: {'octoprint_uptime': 97206}
2021-02-23 06:22:43,221 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2021-02-23 06:22:49,698 - octoprint.plugins.tracking - INFO - Sent tracking event ping, payload: {'octoprint_uptime': 98106}
2021-02-23 06:37:43,223 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2021-02-23 06:37:49,703 - octoprint.plugins.tracking - INFO - Sent tracking event ping, payload: {'octoprint_uptime': 99006}
2021-02-23 06:52:43,224 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2021-02-23 06:52:49,688 - octoprint.plugins.tracking - INFO - Sent tracking event ping, payload: {'octoprint_uptime': 99906}
2021-02-23 07:07:43,225 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2021-02-23 07:07:49,695 - octoprint.plugins.tracking - INFO - Sent tracking event ping, payload: {'octoprint_uptime': 100806}
2021-02-23 07:22:43,227 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2021-02-23 07:22:49,701 - octoprint.plugins.tracking - INFO - Sent tracking event ping, payload: {'octoprint_uptime': 101706}
2021-02-23 07:37:43,228 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2021-02-23 07:37:49,736 - octoprint.plugins.tracking - INFO - Sent tracking event ping, payload: {'octoprint_uptime': 102606}
2021-02-23 07:52:43,230 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2021-02-23 07:52:49,709 - octoprint.plugins.tracking - INFO - Sent tracking event ping, payload: {'octoprint_uptime': 103506}
2021-02-23 07:55:48,829 - tornado.access - WARNING - 409 GET /api/printer (::ffff:192.168.2.180) 17.19ms
2021-02-23 08:07:43,231 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2021-02-23 08:07:49,713 - octoprint.plugins.tracking - INFO - Sent tracking event ping, payload: {'octoprint_uptime': 104406}
2021-02-23 08:22:43,233 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2021-02-23 08:22:49,712 - octoprint.plugins.tracking - INFO - Sent tracking event ping, payload: {'octoprint_uptime': 105306}
2021-02-23 08:37:43,235 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2021-02-23 08:37:49,754 - octoprint.plugins.tracking - INFO - Sent tracking event ping, payload: {'octoprint_uptime': 106206}
2021-02-23 08:52:43,236 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2021-02-23 08:52:49,699 - octoprint.plugins.tracking - INFO - Sent tracking event ping, payload: {'octoprint_uptime': 107106}
2021-02-23 09:07:43,238 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2021-02-23 09:07:49,719 - octoprint.plugins.tracking - INFO - Sent tracking event ping, payload: {'octoprint_uptime': 108006}
2021-02-23 09:22:43,240 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2021-02-23 09:22:49,726 - octoprint.plugins.tracking - INFO - Sent tracking event ping, payload: {'octoprint_uptime': 108906}
2021-02-23 09:37:43,241 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2021-02-23 09:37:49,721 - octoprint.plugins.tracking - INFO - Sent tracking event ping, payload: {'octoprint_uptime': 109806}
2021-02-23 09:52:43,243 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2021-02-23 09:52:49,717 - octoprint.plugins.tracking - INFO - Sent tracking event ping, payload: {'octoprint_uptime': 110706}
2021-02-23 10:07:43,244 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2021-02-23 10:07:49,722 - octoprint.plugins.tracking - INFO - Sent tracking event ping, payload: {'octoprint_uptime': 111606}
2021-02-23 10:22:43,246 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2021-02-23 10:22:49,734 - octoprint.plugins.tracking - INFO - Sent tracking event ping, payload: {'octoprint_uptime': 112506}
2021-02-23 10:37:43,247 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2021-02-23 10:37:49,743 - octoprint.plugins.tracking - INFO - Sent tracking event ping, payload: {'octoprint_uptime': 113406}
2021-02-23 10:52:43,249 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2021-02-23 10:52:49,727 - octoprint.plugins.tracking - INFO - Sent tracking event ping, payload: {'octoprint_uptime': 114306}
2021-02-23 11:07:43,250 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2021-02-23 11:07:49,737 - octoprint.plugins.tracking - INFO - Sent tracking event ping, payload: {'octoprint_uptime': 115206}
2021-02-23 11:22:43,252 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2021-02-23 11:22:49,720 - octoprint.plugins.tracking - INFO - Sent tracking event ping, payload: {'octoprint_uptime': 116106}
2021-02-23 11:37:43,253 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2021-02-23 11:37:49,756 - octoprint.plugins.tracking - INFO - Sent tracking event ping, payload: {'octoprint_uptime': 117006}
2021-02-23 11:52:43,255 - octoprint.server.heartbeat - INFO - Server heartbeat <3
[... 712 lines cut ...]
2021-02-24 00:48:37,733 - octoprint.plugins.ws281x_led_status - INFO - Running OS config test (UI Mode)
2021-02-24 00:48:40,293 - octoprint.plugins.ws281x_led_status - INFO - OS config test complete. Results:
| - adduser: passed
| - spi_enabled: passed
| - spi_buffer_increase: passed
| - set_core_freq: passed
| - set_core_freq_min: passed
2021-02-24 00:48:48,392 - tornado.access - WARNING - 409 POST /api/printer/command (::ffff:192.168.2.180) 24.01ms
2021-02-24 00:49:37,126 - tornado.access - WARNING - 409 POST /api/printer/command (::ffff:192.168.2.180) 17.16ms
2021-02-24 00:49:38,226 - tornado.access - WARNING - 409 POST /api/printer/command (::ffff:192.168.2.180) 17.15ms
2021-02-24 00:49:39,211 - tornado.access - WARNING - 409 POST /api/printer/command (::ffff:192.168.2.180) 17.18ms
2021-02-24 00:49:39,973 - tornado.access - WARNING - 409 POST /api/printer/command (::ffff:192.168.2.180) 17.37ms
2021-02-24 00:50:48,080 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2021-02-24 00:50:52,928 - octoprint.plugins.ws281x_led_status - INFO - Settings refreshed
2021-02-24 00:50:52,930 - octoprint.plugins.ws281x_led_status - INFO - WS281x LED Status runner stopped
2021-02-24 00:50:52,930 - octoprint.plugins.ws281x_led_status - INFO - WS281x LED Status runner stopped
2021-02-24 00:50:52,954 - octoprint.plugins.ws281x_led_status - INFO - Ws281x LED Status runner started
2021-02-24 00:50:54,763 - octoprint.plugins.tracking - INFO - Sent tracking event ping, payload: {'octoprint_uptime': 2706}

I am not sure how to get the debug log (so if anyone lets me know I can upload it).
* `plugin_ws281x_led_status_debug.log`: WRITE HERE

* [X ] I have read the Get Help Guide 

https://cp2004.gitbook.io/ws281x-led-status/guides/get-help-guide

Josiahpox, 2021-02-24:
I also didn't include the entire Octoprint.log file as it apparently was too long, so there about a quarter of the first part missing.

cp2004, 2021-03-01:
Hey @Josiahpox! Sorry about this, somehow this issue slipped by me - I usually catch things like this quickly, and considering the compliment of support at the beginning this feels slightly ironic. Anyway...

Please check the ticket here: #73. Your logs say using RPi 4B rev 1.4 2GB, which I think is one of the models impacted by this change. Please try the instructions detailed there to upgrade the dependency to a patched version.

cp2004, 2021-03-03:
@Josiahpox Continuing the discussion from https://github.com/cp2004/OctoPrint-WS281x_LED_Status/issues/73#issuecomment-789459078, since it is not related to that issue and it is probably easier that it is here.

Please run the OS configuration test, then (because it currently isn't smart enough in the release version of the plugin) open the file at `/boot/config.txt`, and if it contains the line `core_freq=250` remove that line. That is incompatible with the Raspberry Pi 4 however I know that other LED plugins add that because they were not updated to support RPi 4.

Josiahpox, 2021-03-03:
> (comment)

Thank you. I got it working now, I really appreciate your help.
- JP

cp2004, 2021-03-03:
Awesome, enjoy the plugin 😄

closed 2021-03-03 by cp2004