cp2004/OctoPrint-WS281x_LED_Status #115: [Feature Request] Improvement of the print progress effect to show more information. Would it be possible to have the LED's closest to 25%, 50%, and 75% blink as the printing progress effect passes them? For example, if a print is 65% of the way done the 65% of the LED strip length is light up (just as it is coded now), but additionally the nearest LED to 25% and 50% would blink/pulse. From the strip you could then easily tell you are past 20, 50 or 75% printed.

vinster8108, 2021-03-01:
**Is your feature request related to a problem? Please describe.**

**Describe the solution you'd like**

**Why should this be implemented?**

**Additional context**

cp2004, 2021-03-03:
I don't really understand the benefit of this feature - I have never seen a progress bar where part flashes as it is gone, it either animates as a whole or not at all.

What is the significance of knowing 25%, 50% or 75%? As far as I know, those numbers don't mean anything in relation to 3D print in progress, and the actual progress is a lot more useful for that.

vinster8108, 2021-03-06:
Gotcha, I'm new to these addressable LEDs - this idea may be more trouble/coding then it's worth. The simple benefit is more information is displayed via the LEDs. With a vertical progress strip, I suppose it's pretty easy to estimate what your progress percentage is. 
I think we are meaning the same thing with 3D print progress vs. actual progress. The start to finish progress. 

I went a bit extra lol. I have 4 strips in parallel. My longest run is 89 LEDs and meets the second longest run in the middle of the top bar before spiraling at the top ring (pic attached). Progress indicators move up the sides simultaneously, meet in the middle of the cross bar and then fill out the ring. So at a glance, it isn't as easy for me to tell if my progress bar is half way done. A blinking LED at 25, 50 and 75% as the progress passes them would provide that information. I realize most may not find this addition overly beneficial. 
[image]

cp2004, 2021-03-06:
Wow, that's some impressive setup... I like how neatly they fit in the channels of the extrusion.

Try the release candidate for 0.7.0 (#119) - there's 3 new progress effects, one that may be more beneficial is the one that moves from both ends, gradient would probably be a bit boring for that number of LEDs. I can't tell if it would work better or worse.

I would accept a PR that added a new progress effect for this. In the future we may also get per-segment control of the LEDs, it's one of my 'big ideas'. So you could just put the progress bar on the vertical sections.