cp2004/OctoPrint-WS281x_LED_Status #116: [Feature Request] option to limit certain effects to specific parts of the led strip

schieska, 2021-03-02:
I would like to show different effects on different parts of the led strip.
for example:
I would like to set something like
progress bar to led 1 - 10
startup effect on the full strip
ect.

This way I can have a status bar and still have some leds lighting my printbed

cp2004, 2021-03-03:
This is on the big long list. It will come at some point, segment control of the LEDs. However, it will not be anytime soon, since it is a massive development task to work out and convert all of the code to work with segments.

It is tempting to close this, to not set any expectations of when or how this is going to come, because it could be a long time and maybe not happen at all. I would like it to, but you never know what's going to happen.

IzK666, 2021-06-11:
I like the idea and hope to see implemented soon.
I have an inverted U on the front and two lines that goes on the sides and I wish I could use the U for lightning and the sides for printing effects. It's a fraction of what IARI has (#121), but still a few.