cp2004/OctoPrint-WS281x_LED_Status #117: [Feature Request]

dchauran, 2021-03-04:
**Is your feature request related to a problem? Please describe.**

No

**Describe the solution you'd like**

On the utilities screen, I would like a "test RCB color" option which allows me to input specific RGB values to test. Currently if I want to see if I like a color, the easiest way to test is to set torch or idle, save config, go back, repeat.

**Why should this be implemented?**

To improve the customization experience.

**Additional context**

This seems like a straightforward change, so I can look into making it, but i wanted to make sure it wasn't already being worked on.

cp2004, 2021-03-04:
I'd be cool with such a change, it may end up in a later release since I am pushing for a release candidate of 0.7.0 this weekend, and it is in feature freeze. The next release (current devel/0.7.x branch) has *many* improvements, the API for this is now in place you can take advantage of.

dchauran, 2021-03-04:
Thanks, I will take a look and let you know if I'm not able to implement it.

dchauran, 2021-03-05:
@cp2004 I got it done and made a pull request into devel, but the tests in devel are failing, regardless of my changes.

cp2004, 2021-03-05:
I wasn't expecting it to come that fast! Will merge it to the 0.7.x branch and aim to release the first candidate tomorrow, with this included.

About tests.... yeah I'm not that good at writing tests, currently they fail depending on how you run them and a bunch of them are commented out because they are failing. I will get them at some point.

cp2004, 2021-03-05:
Well I was bored, tomorrow turned into today and it got released. Please test the [release candidate](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/releases/0.7.0rc1) and let me know of any problems. [Feedback ticket here](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/issues/119)

dchauran, 2021-03-05:
Thanks, I will load it up to test after this print finishes!

I'm also considering adding a preview for effects, but that will take more time. I'm okay with Python, but not at all familiar with jinja, and adding that requires plumbing through the API, and then making a bunch of changes to the template.

cp2004, 2021-03-11:
0.7.0 has been released.

closed 2021-03-11 by cp2004