cp2004/OctoPrint-WS281x_LED_Status #119: [Feedback] Feedback on the 0.7.0rc1 Release Candidate

cp2004, 2021-03-05:
It's finally here, 0.7.0rc1 Release Candidate!

Be sure to check out the release candidate [release notes here](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/releases/tag/0.7.0rc1) for details of all the changes.

Please test this release, and let me know what you think. If all goes well this will be released at some point next week, so feedback is important to make sure that everything is good! An 'All is working well' is great feedback, since it lets me know people *are* actually testing this release, even if there are no issues to report.

**If you find a bug** please use the bug report template and send across the logs, so that I can get it fixed as quickly as possible.

trevor229, 2021-03-05:
Just tested the RC on my setup, auto connect idle effect changeover works perfectly when the printer and leds are switched on. Thanks for the update!

Gordontgopher, 2021-03-06:
Great work! I can confirm that request #65 works on my set up. Thanks :)

Jester4977, 2021-03-06:
Testing the RC today and so far things are working great.  I really like the single pixel progress (#70), thanks for adding!

cp2004, 2021-03-06:
Thank you to all of you testing so far, hope you are enjoying all the new features 😄

kevinvalk, 2021-03-08:
Currently testing the plugin, great work! Mainly in progress of testing the dedicated white support.

cp2004, 2021-03-08:
0.7.0rc2 is out, not going to bother rotating the feedback ticket since it means updating a bunch of tickets. Keep testing!

Rollout of 0.7.0rc1 was very strong - up to 14 tracked instances ([source](https://data.octoprint.org/export/plugin_stats_7d.json)) at the time of writing- stable release will probably land about the end of the week.

[image]

[See more at OctoPluginStats](https://cp2004.github.io/OctoPluginStats)

cp2004, 2021-03-11:
Stable release will likely come tomorrow evening (for me), if you have any final comments please get them in! It will be good to get this one out of the door.

cp2004, 2021-03-11:
0.7.0 has been released.

closed 2021-03-11 by cp2004