cp2004/OctoPrint-WS281x_LED_Status #120: [Feature Request] Plugin reacting on Extruder / Filament changes (MMU)

Prutsium, 2021-03-06:
This is a feature request see video for purpose.

I would like to have the ability that the Plugin reacts on the current extruder used (1-5) in an MMU setup
So when Extruder 1 is used then leds 1-4 are on (with selected color), extruder 2 uses leds 5-8 (with selected color), and so on

This makes it possible to have selective light in MMU buffer setups.

Last but not least optional via Wled Api so the plugin sends the commands over the api to Wled :)

https://user-images.githubusercontent.com/42968695/110199958-74b3db80-7e5b-11eb-8dfc-feda58c590df.mp4

Prutsium, 2021-03-07:
Seems the video doesnt show here so a link to the video on Youtube:
https://youtu.be/sUbybQ9KC38

cp2004, 2021-03-07:
I'm unlikely to add such a specific configuration, as I said on Discord it will be more likely that there is a combination of segment control and custom events etc. to react to, and you can build this kind of setup yourself. I'm not sure how either of those are going to work or where they are going to turn up, it will be a while and they probably won't come at the same time.. While I would love to spend all day on this project it isn't going to happen unfortunately 😄.

Prutsium, 2021-03-07:
Then we go for Plan B 😊
Mqtt into Nodered as i believe active extruder number is published by
Octoprint on mqtt.

cp2004, 2021-03-13:
Closing in favour of #125, where options for custom triggers are being tracked. There were a few similar issues that are covered by this.

closed 2021-03-13 by cp2004