cp2004/OctoPrint-WS281x_LED_Status #122: [Bug] M150 does not take 0.7.0rc1 dedicated white feature into account

kevinvalk, 2021-03-08:
N/A

**Version of plugin and OctoPrint**

0.7.0rc1

**(If applicable) Browser version & browser error console**

**Additional context**

cp2004, 2021-03-08:
Hmm, the expectations are totally right there, I'll take a look later. A quick check of the code makes me think this should work, but maybe something's wonky in the detection of the 255 value - throwing the debugger on it later should explain.

kevinvalk, 2021-03-08:
Good to hear!

Just tested with `M150 W1`, that turns red, green and blue on on my SK6812_STRIP_RGBW, but not the white LED.

To conclude, with `M150` commands there is no difference in my setup when doing `M150 R1 G1 B1` or `M150 W1`. Both cases behave the same and turn on red, green and blue channels and do not change the dedicated white channel.

cp2004, 2021-03-08:
That's the intended behaviour if white override is not on. With W-O on It should occur if the white is @ 255, or R,G,B are all 255 as well.

kevinvalk, 2021-03-08:
My bad, indeed I just figured that out.... 

But in the case of M150, would a four channel approach make more sense IFF your LED strip has a white channel? In the sense that `M150 R50 W200` would turn on the the dedicated white LED and the red LED? When you do not have a dedicated white LED, the W channel just controls red, green and blue in tandem.

cp2004, 2021-03-08:
I can't entirely reproduce this - my RGB LEDs turn off when `M150 R255 G255 B255` is sent, or `M150 W255`. I can't tell if it is showing anything on the W LEDs, since I don't have them to test it.

Pinging @samwiseg0 to see if he can give this a test, having implemented it, as far as I can see this does work but I want to try and get a second opinion.

@kevinvalk - could you install the latest commit to pre-release which adds more logging of the M150 command (I want to see if there is a rounding/parsing issue), enable debug logging (Features) and upload the `plugin_ws281x_led_status_debug.log` file so that I can see what is going on.
You can download and upload this from the plugin manager, or install using the URL:
[OctoPrint-WS281x LED Status-0.7.0rc1.zip](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/files/6104050/OctoPrint-WS281x.LED.Status-0.7.0rc1.zip)

```
https://github.com/cp2004/OctoPrint-WS281x_LED_Status/files/6104050/OctoPrint-WS281x.LED.Status-0.7.0rc1.zip
```

kevinvalk, 2021-03-08:
Aa damn, sorry that you are spending all this time. Like I tried to say in my previous commit, it currently works as designed (as far as I understand) (original 0.7.0rc1 version).

**Use dedicated white: enabled**
- `M150 W254`: Red, green, blue at 254, white at zero
- `M150 W255`: Red, green, blue at 0, white at 255
- `M150 R255 G255 B254`: Red 255, green 255 blue at 254, white at 0
- `M150 R255 G255 B255`: Red, green blue at 0, white at 255

**Use dedicated white: disabled**
White never turns on

The reason I thought this was "wrong" because I was under the impression that `M150 W<brightness>` would always control the white LED brightness independent from red, green, blue. **This assumption was wrong!**

---

So the open question now is, should the M150 command control the white channel independently? The decision is even more confusing for what happens if you have the "dedicated white" setting on or off and if you hardware (LED strip) even has a dedicated white channel.

Hypothetical example: let's say I want to turn on my white LED's at the beginning and end of the print to 100% (`M150 W255`), but during the print I am fine with a lower brightness (to save energy etc), so I want to do `M150 W100`. But now all of a sudden I have different light as I am using red, green blue and not dedicated white.

samwiseg0, 2021-03-08:
👋 @kevinvalk The issue was that there is no way in the UI to show white in the color picker. This was a "workaround" to be able to use the dedicated white LEDs as my thought was if you have the white LEDs then there is really no reason to run "white" on the RGB side. The code only uses the white channel when the dedicated white override is enabled. 

https://github.com/cp2004/OctoPrint-WS281x_LED_Status/blob/0.7.x/octoprint_ws281x_led_status/util.py#L25

I am curious as to what would be the use case to run white and colors at the same time?

cp2004, 2021-03-08:
It makes sense intuitively, that `M150 W100` would *only* set the white LEDs, if dedicated white is enabled. At the moment, it is only triggered if it is W = 255. However, and I see the miscommunication - I thought there was a problem with this being triggered from M150, it is not but it is still kind of an issue - with the unclear behaviour instead.

With M150 we can bypass the problem of the colour picker, and trust that if the user sets the LEDs using M150 that they are aiming for the W led and not equal RGB. What will *not* happen, is independent control. That's far more difficult to make understandable and actually code.

I'm working on a solution, will probably release RC2 with clearer behaviour here tomorrow. It will go like this:
* `M150 W100` -> White @ 100 intensity
* `M150 W255` -> White @ 255 intensity
* `M150 R255 G255 B255` -> White @ configured intensity (white brightness)
* `M150 R254 G254 B254` -> individual r, g, b values, since they are not in the region to auto-detect we should use white.

*Hopefully*, now we are all on the same page! I think so.

samwiseg0, 2021-03-08:
I agree @cp2004. That was not something I considered when I thought of the workaround.

cp2004, 2021-03-08:
Yeah I hadn't thought of it either. One day there will be too many things implemented I will lose track of how they all work...

cp2004, 2021-03-08:
It was quicker than I thought. [0.7.0rc2 is out now](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/releases/tag/0.7.0rc2),  if it needs more adjustments or doesn't work right let me know!

kevinvalk, 2021-03-09:
Quick work, tested and works as defined!

Thanks 🥇

commit 9d016359 (2021-03-08): :sparkles: Use dedicated white on M150 W

The correct, intuitive behaviour. Closes #122
closed 2021-03-09 by kevinvalk