cp2004/OctoPrint-WS281x_LED_Status #124: [Bug] WS2813 not working

dzervas, 2021-03-10:
**Description of the bug**

I just got a 2 meter (60 leds/m) WS2813 LED strip and it doesn't seem to do anything. I use a Pi 0w (it's not for a 3D printer but for a kind of 2D plotter so even Pi 0 is more than enough) and connected DO to GPIO10 and BO to GND.

**Steps to reproduce**
Tried torch, test red/green/blue/white, nothing.

**What did you expect to happen**

**Logs and screenshots**
`.octoprint/logs/plugin_ws281x_led_status_debug.log`

```
[2021-03-10 18:01:42,590] INFO: Initialising LED strip
[2021-03-10 18:01:42,624] ERROR: Strip failed to initialize, no effects will be run.
[2021-03-10 18:01:42,625] ERROR: Please check your settings.
[2021-03-10 18:01:42,632] ERROR: Here's the exception: ws2811_init failed with code -13 (Unable to initialize SPI)
[2021-03-10 18:01:42,633] INFO: No strip initialised, exiting the effect process.
[2021-03-10 18:10:35,123] INFO: Initialising LED strip
[2021-03-10 18:10:35,154] INFO: Strip successfully initialised
[2021-03-10 18:10:35,176] INFO: Debug logging not enabled, if you are reporting issues please enable it under 'Features' in the settings page.
[2021-03-10 18:10:35,191] INFO: On message recieved, turning on LEDs
[2021-03-10 18:12:47,489] INFO: Kill message recieved, Bye!
[2021-03-10 18:12:47,668] INFO: Initialising LED strip
[2021-03-10 18:12:47,684] INFO: Strip successfully initialised
[2021-03-10 18:12:47,685] INFO: Debug logging not enabled, if you are reporting issues please enable it under 'Features' in the settings page.
[2021-03-10 18:12:47,701] INFO: On message recieved, turning on LEDs
[2021-03-10 18:13:03,184] INFO: Off message recieved, turning off LEDs
[2021-03-10 18:13:05,944] INFO: On message recieved, turning on LEDs
[2021-03-10 18:13:42,731] INFO: Kill message recieved, Bye!
[2021-03-10 18:13:42,914] INFO: Initialising LED strip
[2021-03-10 18:13:42,930] INFO: Strip successfully initialised
[2021-03-10 18:13:42,931] INFO: Debug logging not enabled, if you are reporting issues please enable it under 'Features' in the settings page.
[2021-03-10 18:13:42,936] INFO: On message recieved, turning on LEDs
[2021-03-10 18:13:52,811] INFO: Off message recieved, turning off LEDs
[2021-03-10 18:13:53,765] INFO: On message recieved, turning on LEDs
[2021-03-10 18:13:58,494] INFO: Off message recieved, turning off LEDs
[2021-03-10 18:14:13,499] INFO: On message recieved, turning on LEDs
[2021-03-10 18:14:14,466] INFO: Off message recieved, turning off LEDs
[2021-03-10 18:14:47,541] INFO: Kill message recieved, Bye!
[2021-03-10 18:14:47,724] INFO: Initialising LED strip
[2021-03-10 18:14:47,729] INFO: Strip successfully initialised
[2021-03-10 18:14:47,741] INFO: Debug logging not enabled, if you are reporting issues please enable it under 'Features' in the settings page.
[2021-03-10 18:14:47,747] INFO: Off message recieved, turning off LEDs
[2021-03-10 18:14:57,225] INFO: On message recieved, turning on LEDs
[2021-03-10 18:14:58,329] INFO: Off message recieved, turning off LEDs
[2021-03-10 18:14:58,966] INFO: On message recieved, turning on LEDs
[2021-03-10 18:14:59,445] INFO: Off message recieved, turning off LEDs
[2021-03-10 18:15:01,696] INFO: On message recieved, turning on LEDs
[2021-03-10 18:19:10,490] INFO: Off message recieved, turning off LEDs
[2021-03-10 18:19:11,031] INFO: On message recieved, turning on LEDs
[2021-03-10 18:19:19,643] INFO: Off message recieved, turning off LEDs
[2021-03-10 18:19:20,169] INFO: On message recieved, turning on LEDs
[2021-03-10 18:19:24,023] INFO: Off message recieved, turning off LEDs
[2021-03-10 18:19:24,655] INFO: On message recieved, turning on LEDs
[2021-03-10 18:25:29,614] INFO: Kill message recieved, Bye!
[2021-03-10 18:27:26,862] INFO: Initialising LED strip
[2021-03-10 18:27:26,926] INFO: Strip successfully initialised
[2021-03-10 18:27:26,927] INFO: Debug logging not enabled, if you are reporting issues please enable it under 'Features' in the settings page.
[2021-03-10 18:27:27,029] INFO: On message recieved, turning on LEDs
[2021-03-10 18:31:16,199] INFO: Kill message recieved, Bye!
[2021-03-10 18:31:16,383] INFO: Initialising LED strip
[2021-03-10 18:31:16,399] INFO: Strip successfully initialised
[2021-03-10 18:31:16,401] INFO: Debug logging not enabled, if you are reporting issues please enable it under 'Features' in the settings page.
[2021-03-10 18:31:16,405] INFO: On message recieved, turning on LEDs
[2021-03-10 18:32:40,635] INFO: Kill message recieved, Bye!
[2021-03-10 18:32:40,820] INFO: Initialising LED strip
[2021-03-10 18:32:40,839] INFO: Strip successfully initialised
[2021-03-10 18:32:40,849] DEBUG: Current settings:
 | * STRIP SETTINGS *
[... 211 lines cut ...]
| - set_core_freq_min: passed
2021-03-10 18:49:19,485 - octoprint.plugins.ws281x_led_status - INFO - Settings refreshed
2021-03-10 18:49:19,525 - octoprint.server - INFO - Listening on http://127.0.0.1:5000
2021-03-10 18:49:20,135 - octoprint.plugins.pluginmanager - INFO - Loaded plugin repository data from disk, was still valid
2021-03-10 18:49:32,742 - octoprint.util.pip - INFO - Using "/home/pi/oprint/bin/python3 -m pip" as command to invoke pip
2021-03-10 18:49:32,794 - octoprint.util.pip - INFO - pip installs to /home/pi/oprint/lib/python3.7/site-packages (writable -> yes), --user flag needed -> no, virtual env -> yes
2021-03-10 18:49:32,796 - octoprint.util.pip - INFO - ==> pip ok -> yes
2021-03-10 18:49:32,927 - octoprint.plugins.softwareupdate - INFO - Minimum free storage across all update relevant locations is 3.1GB. That is considered sufficient for updating.
2021-03-10 18:49:33,797 - octoprint.plugins.continuousprint - INFO - Continuous Print Plugin started
2021-03-10 18:49:33,843 - octoprint.plugins.themeify - INFO - Themeify initialized.
2021-03-10 18:49:34,072 - octoprint.plugins.ws281x_led_status - INFO - Ws281x LED Status runner started
2021-03-10 18:49:35,161 - octoprint.plugins.tracking - INFO - Sent tracking event ping, payload: {'octoprint_uptime': 44}
2021-03-10 18:49:36,446 - octoprint.plugins.tracking - INFO - Sent tracking event pong, payload: {'version': '1.5.3', 'os': 'linux', 'bits': 32, 'python': '3.7.3', 'pip': '20.3.3', 'cores': 1, 'freq': 1100.0, 'ram': 386322432, 'pi_model': 'Raspberry Pi Zero W Rev 1.1', 'octopi_version': '0.18.0', 'plugins': 'touchui:0.3.17,themeify:1.2.2,ws281x_led_status:0.6.0,klipper:0.3.2,continuousprint:1.2.1'}
2021-03-10 18:49:37,871 - octoprint.plugins.tracking - INFO - Sent tracking event startup, payload: {'version': '1.5.3', 'os': 'linux', 'bits': 32, 'python': '3.7.3', 'pip': '20.3.3', 'cores': 1, 'freq': 1100.0, 'ram': 386322432, 'pi_model': 'Raspberry Pi Zero W Rev 1.1', 'octopi_version': '0.18.0'}
2021-03-10 18:49:39,791 - octoprint.server.preemptive_cache - INFO - Preemptively caching / (ui _default) for {'base_url': 'http://table.lan/', 'path': '/', 'query_string': 'l10n=en'}
2021-03-10 18:49:44,544 - octoprint.server.util.flask - INFO - Passively logging in user admin from ::ffff:10.13.37.111
2021-03-10 18:49:44,569 - octoprint.access.users - INFO - Logged in user: admin
2021-03-10 18:49:53,863 - octoprint.plugins.pluginmanager - INFO - Loaded notice data from disk, was still valid
2021-03-10 18:49:59,037 - octoprint.plugins.pluginmanager - INFO - Loaded notice data from disk, was still valid
```

**Version of plugin and OctoPrint**

OctoPrint (using OctoPi): 1.5.3
plugin: 0.6.0

cp2004, 2021-03-10:
Hi, WS2813 strips are not supported by this plugin or the underlying library explicitly. You've selected WS2812, presumably because WS2813 is not in the list, the only way it *may* work is if any of the other options (WS2811, SK6182) are compatible.

You can see the list of supported/recommended hardware here: https://cp2004.gitbook.io/ws281x-led-status/guides/setup-guide-1/supported-hardware#led-strips

dzervas, 2021-03-10:
Oh forgot to mention that the underlying library seems to work with WS2813: https://github.com/jgarff/rpi_ws281x/issues/152

dzervas, 2021-03-10:
I connected to the wrong side of the strip -.-
I now have flickering leds but that's probably due to doing nothing about the voltage difference (I'm currently testing so I haven't hooked either a diode or a level shifter)

closed 2021-03-10 by dzervas

dzervas, 2021-03-10:
you're right! I tested by quickly soldering 2 jumper cables on the other side of the 2m strip and that was causing the flicker (I found that out just after I made a circuit with a transistor for the level shifting).
When I properly soldered the 4 cables on the right side (where BI and DI are), the whole strip worked without any level shifting.

out of topic but do you know a plugin where I can just change the led strip effects? the whole setup is for a sand table, so there's no actual printing taking place, thus most features of this plugin do not apply to me, only the torch

awesome plugin, I'll definitely add a strip to my prusa and use this plugin like it's meant to!