cp2004/OctoPrint-WS281x_LED_Status #126: Pulse/Blink pattern glitch [Bug]

kritch83, 2021-03-16:
**Description of the bug**

Since recent update to 0.7.0, when using the Pulse pattern @ idle, when the LEDs get to "off," there is a flash for a split second of the color the pulse is set to.  So for example, you have the Pulse effect set to Red, the pulse fades down, then when it looks to be completely off, all the LEDs burst on for a moment and then back off; then the pulse fades to bright.  If you change the color to blue or whatever it flashes that color of the pulse effect.

The Blink effect doesn't work at all.  Just stays solid.  Looking at the logs there is a error that prevents it from running.

I tried all the different effects for idle, and Pulse & Blink are the only ones having issues.

**Steps to reproduce**
1. Click on octoprint settings.
2. go to WS281X....
3. Select "pulse" for idle effect with any delay.
4. When lights fade down the flash happens and they fade back up.
5. For Blink issue, follow 1 & 2 but instead change effect to "Blink" when idle.
6. Color then just stays on, no blink.

**What did you expect to happen**

For the color to fade down to off and then back on smoothly.

**Logs and screenshots**

See below...

**Version of plugin and OctoPrint**

WS281x LED Status (0.7.0)
OctoPrint Version 1.5.3
OctoPi Version 0.18.0, running on Raspberry Pi 3 Model B Rev 1.2

**(If applicable) Browser version & browser error console**

NA

**Additional context**

I verified the OS tests passed.  I have a very good common ground for the setup.  Its a custom RPi hat for driving the neopixels, powering the printerboard (RUMBA), monitoring voltages & temp and reporting back top Blynk (soon to be displayed to Enclosure Plugin).  System is driven with a 5V @20amp power supply & a 24V @9.2A for the printer board.  All grounds connected.  Data line to pixels has 470(466 actual) ohm resistor in line.  Setup/config hasn't changed with any plugins, octoprint, or system.  Just upgraded the WS281X plugin and issue started.  I did reflash the Marlin FW after issue started (turned on adaptive fan speed in FW) & that had no change of issue.  Logs attached.

[octoprint (2).log](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/files/6145759/octoprint.2.log)
[plugin_ws281x_led_status_debug (2).log](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/files/6145760/plugin_ws281x_led_status_debug.2.log)

Any input would be appreciated!

cp2004, 2021-03-16:
I have a theory for the pulse effect, there may be anywhere between 0-20ms (+ processing time) between the fade effect ending, and the 'Pulse' effect to stop doing it's thing. Because they both try and control the brightness, there is a block on controlling while fading, which seems to be working, except for right at the end when this is released. I vaguely remember seeing this once or twice when developing it, maybe not enough to think anything of it when there are bigger problems. This wouldn't occur with fading off, but the fade is so awesome, so why would you do that 😄 

As for the second with the blink effect, oops. Looks like the `int` conversion is around the wrong thing. In other words, it works with default settings (because they are integers) but not once you have changed them (the frontend works in strings...)

Thank you for the detailed bug report, I try and fix it soon.

kritch83, 2021-03-16:
Thank **YOU** for your time & the great software.  Let me know if you need any other info.

guigro, 2021-03-17:
I have the same bug. Just switched from RGB Status to your plugin (that I now love !) and was trying to setup the pulse effect, that is glitching.
I'll wait the next update to use it ;)

cp2004, 2021-03-18:
I have just published 0.7.1dev1, on the development release channel.

@kritch83 and @guigro, please switch to development channel in the Software update's settings so that you can update and test it out. Once I have verification that the issues are fixed, I will push it out to everyone. Just want to get a 2nd look, so that I know I got *everything*.

[image]

kritch83, 2021-03-18:
Looks good!  I tried both the Blink and the Pulse and they are both working great.  Thanks!

cp2004, 2021-03-18:
0.7.1 has been released.

closed 2021-03-18 by cp2004