cp2004/OctoPrint-WS281x_LED_Status #128: Torch should ignore time settings

antonyburden, 2021-03-23:
This may be a bug, but when the time settings dictate the LED's status settings are off, any TORCH command also fails.

I believe for both to be of use ANY torch command should over-ride/ignore the time settings.

My use is the torch comes on via GCODE and a photo is taken and pushed to my phone, works great unless time settings are activated and my photo is black during the small hours!

cp2004, 2021-03-23:
This is a request, since it was intentionally designed that nothing on the LEDs shows when they are outside the active time. Maybe will sort out something for the next release that makes it work.

cp2004, 2021-11-15:
🎉 Implemented and will be released with 0.8.0. There'll be a release candidate beforehand, switch to the 'Release Candidate' channel in the software update section to get notified when it is available.

cp2004, 2022-02-04:
0.8.0 has been released (finally...!)

commit 370a7419 (2021-11-15): :sparkles: Allow torch mode to ignore active times

This plugin is so highly configurable isn't it

Closes #128
closed 2022-02-04 by cp2004