cp2004/OctoPrint-WS281x_LED_Status #130: [Feature Request] Ambient light

ruudboon, 2021-04-11:
**Is your feature request related to a problem? Please describe.**

I would like to light up my filament box with a solid color when printing. 

**Describe the solution you'd like**

Solution 1
Define a color and the number of pixels can be assigned as ambient light pixels. These led's will be added after the "Number of LEDs" and will have the static configured color. Any effects will run on the "Number of LEDs" and not on the configured ambient pixels.

Solution 2 (probably harder to implement)
Define a color and an array of pixel numbers that can be assigned as ambient light pixels for example (0,1,10,11, 20,21). These defined leds will be skipped when using effects on the strip and show the configured static color.

**Why should this be implemented?**
It will give us some more options to make a great looking nerdy environment.

Thnx for this awesome project

cp2004, 2021-04-11:
Rather than implement lots of very specific features for individual people to get it to do what they want, I prefer to go down a more general route. Solid colour while printing is already an option, taking over from printing progress.

In the future, segmented control over the LEDs will be possible so that people can create what they like. However there is no current timeline for when this will be implemented, I have a lot to do.

closed 2021-04-11 by cp2004

ruudboon, 2021-04-11:
Ok, fair enough. Thank you for the reply.