cp2004/OctoPrint-WS281x_LED_Status #131: [Bug] Compatibility with TPLinkSmartPlug plugin

guigro, 2021-04-20:
**Description of the bug**

I use the TP Link SmartPlug plugin to shut down the printer on idle and put it back on when I upload a file.
When it's powered off and I try to power it on, Octoprint can't connect to the printer again, until I restart Octoprint.

I guess it's the same if I have Octoprint running and then power off manually the printer, the power it back on.

For info : Pi running is a 4, and the printer is a Prusa MK3S.

**Steps to reproduce**

1. When you're on Octoprint with the printer in Idle state (connected and working). Look nicely at the screen and love your devices.
2. Manually power off the printer (from the power source, power unit, power cable) but keep Octoprint runiing.
3. Power back the printer
4. Try to connect
5. There should be an error message
6. To make it work, you have to restart Octoprint

**What did you expect to happen**

I expect the printer to go back on and Octoprint to be able to reconnect to it :)

**Logs and screenshots**
plugin_ws281x_led_status_debug.log : 
`[2021-04-20 17:56:05,858] INFO: Kill message recieved, all effects stopped. Bye!
[2021-04-20 17:56:05,905] INFO: Initialising LED strip
[2021-04-20 17:56:05,908] INFO: Strip successfully initialised
[2021-04-20 17:56:05,909] DEBUG: Current settings:
 | * STRIP SETTINGS *
 | - count: 64
 | - dma: 10
 | - reverse: False
 | - pin: 10
 | - brightness: 100
 | - freq_hz: 800000
 | - invert: False
 | - white_override: False
 | - white_brightness: 50
 | - adjustment: {u'B': 100, u'R': 100, u'G': 100}
 | - type: WS2811_STRIP_GRB
 | - channel: 0
 | * EFFECT SETTINGS *
 | disconnected
 | - color: #ff0000
 | - delay: 20
 | - enabled: False
 | - effect: Color Wipe
 | success
 | - color: #00ff00
 | - delay: 25
 | - enabled: True
 | - effect: Color Wipe
 | - return_to_idle: 300
 | progress_cooling
 | - color: #ff0000
 | - enabled: True
 | - effect: Progress Bar
 | - base: #0000ff
 | - bed_or_tool: tool
 | - threshold: 40
 | torch
 | - color: #ffffff
 | - enabled: True
 | - effect: Solid Color
 | - timer: 15
 | - delay: 1
 | - toggle: True
 | - auto_on_webcam: True
 | startup
 | - color: #00ff00
[... 105 lines cut ...]
2021-04-20 17:56:36,705 - octoprint.util.comm - INFO - Changing monitoring state from "Detecting serial connection" to "Error: No more candidates to test, and no working port/baudrate combination detected."
2021-04-20 17:56:36,708 - octoprint.util.comm - INFO - Changing monitoring state from "Error: No more candidates to test, and no working port/baudrate combination detected." to "Offline (Error: No more candidates to test, and no working port/baudrate combination detected.)"
2021-04-20 17:56:36,712 - octoprint.plugins.excluderegion - INFO - Printing stopped: event=Error
2021-04-20 17:56:36,726 - octoprint.plugins.action_command_notification - INFO - Notifications cleared
2021-04-20 17:57:30,766 - octoprint.server.util.flask - INFO - Passively logging in user guigro from ****
2021-04-20 17:57:30,768 - octoprint.access.users - INFO - Logged in user: guigro
2021-04-20 17:57:30,951 - octoprint.server.util.sockjs - INFO - Client connection closed: ****
2021-04-20 17:57:32,090 - octoprint.server.util.sockjs - INFO - New connection from client: ****
2021-04-20 17:57:32,131 - octoprint.server.util.flask - INFO - Passively logging in user guigro from ****
2021-04-20 17:57:32,133 - octoprint.access.users - INFO - Logged in user: guigro
2021-04-20 17:57:34,087 - octoprint.server.util.sockjs - INFO - User **** logged in on the socket from client ::ffff:192.168.50.100
`

**Version of plugin and OctoPrint**

Plugin 0.7.2
Octoprint : 1.5.3

**(If applicable) Browser version & browser error console**

Not applicable

**Additional context**

I made the same kind of feedback to the TPLinkSmartPlug plugin creator on his Github

cp2004, 2021-04-20:
A duplicate of #113 & #13, yeah I have no idea how to fix this one. I still haven't been able to reproduce the issue here to try and fix it. It obviously does not affect all that many people, since I have 3 reports of it in like 8 months & over 780 instances in the last month.

The latest theory we were able to come up with was that there was a resource being inherited by the sub process the plugin uses to run the LED strip & effects etc. 

There are two current workarounds, necessarily to restart OctoPrint:
* Open the settings dialog -> change a WS281x setting > save. This will cause the process to restart, removing the resource lock.
* Uncheck 'request exclusive access to the serial port' in the serial connection settings

I will look into it again - but I can't guarantee I'll be able to find much. It could be printer/hardware specific, but there is no pattern with the people that have this issue. I spent ages looking through documentation and code, to find nothing of much use...

guigro, 2021-04-20:
Alright you got it, the second tip (the request exclusive) make it work perfectly !

Sorry for the duplicate, and thanks for the amazing work !

closed 2021-04-20 by guigro