cp2004/OctoPrint-WS281x_LED_Status #132: BUG: LED count settings of less than 6 causes flickering of LEDs

Civlo85, 2021-04-23:
I have only 2 LEDs of a WS2812B Strip. If I configure the count to 2 LEDs, the LEDs are flickering when a fade in or out effect is going on. When I test the colors of the LEDs in the GUI, it works, but the effects does not. I have to go up to at least 6 LEDs, then the flickering stops and I have "normal behavior".

cp2004, 2021-04-23:
There is a [bug report template](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/issues/new?assignees=&labels=&template=bug_report.md&title=%5BBug%5D), please fill it in.

Most important is the `plugin_ws281x_led_status_debug.log` **with [debug logging](https://cp2004.gitbook.io/ws281x-led-status/configuration/features#debug-logging) enabled.**

Civlo85, 2021-04-23:
Sorry for that.

**Description of the bug**

I have only 2 LEDs of a WS2812B Strip. If I configure the count to 2 LEDs, the LEDs are flickering when a fade in or out effect is going on. When I test the colors of the LEDs in the GUI, it works, but the effects does not. I have to go up to at least 6 LEDs, then the flickering stops and I have "normal behavior".

**Steps to reproduce**

1. Go to 'OctoPrint Settings -> Plugins / WS281x LED Status -> LED  Strip Settings'
2. Change 'Number of LEDs' to '2' and 'Close' and 'Save'
3. See flickering of LEDs

**What did you expect to happen**

I expect not flockering LEDs. When I change the Settings to a LED Number greater or equals 6, then the LEDs behave like I expect -> Not flickering.

**Logs and screenshots**

Attached:
[octoprint.log](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/files/6365600/octoprint.log)
[plugin_ws281x_led_status_debug.log](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/files/6365601/plugin_ws281x_led_status_debug.log)

**Version of plugin and OctoPrint**

WS281x LED Status (0.7.2)
OctoPrint Version 1.5.3

cp2004, 2021-05-27:
Apologies for not getting back to you faster on this one - I have to develop plugins in cycles and this one has been at the bottom of the list. I'm hoping to get a chance to properly dig into this one next week.

Until then, confirmation that this is a really weird issue, I just setup the development LEDs and I have some really strange effects here, when LEDs are less than 6. Stepping through the debugger shows no issues... So it may take a while to dig into this one and figure out exactly what is wrong here. Either way I'm not looking forward to it 🙂

cp2004, 2021-06-27:
This seems like it may be an upstream bug of the dependency - not fun on any level...

cp2004, 2021-06-29:
A fix/workaround has been implemented for 0.8.0, by tricking the underlying library into thinking it is using 6 LEDs, but only pushing effects to the number configured.

cp2004, 2022-02-04:
0.8.0 has been released (finally...!)