cp2004/OctoPrint-WS281x_LED_Status #133: [Bug]

spacecabbie, 2021-04-24:
Please remove or fix the "pi check script"

`ws281x_led_status - WARNING - No Raspberry Pi detected, will not load plugin`
Really do not understand why one would go through the trouble, to specifically break it on raspberry os. I use the Rtk.GPIO and I have to hack half of the plugin to get it to work.

cp2004, 2021-04-24:
Do you have a Raspberry Pi?

I can only go by what the underlying library supports - so even if I remove that check, the LEDs will fail to run anyway.

spacecabbie, 2021-04-24:
I presume Adafruit ? yes it does. I am having a horrid day i'll respond with bit more info tommorow.
fyi i run: https://uk.pi-supply.com/products/ryanteck-rtk-gpio-pc-gpio-interface in combination with Raspberry Pi OS.
The stuff works with direct tests sensors ic2 etc just all the damn plugins seem to be protected to not allow it to run on anything else then raspberry pi perhaps a licenseing thing ?

cp2004, 2021-04-24:
Please try running the test code here:

https://github.com/jgarff/rpi_ws281x

spacecabbie, 2021-05-04:
> Please try running the test code here:
> 
> https://github.com/jgarff/rpi_ws281x

Okay first thing first. My tone in general was so way of that reading it back I am shocked you even answered.  I do sincerely apologize for that it was rude and just unprofessional. 

I am looking into this but indeed the underlying plugin does not work either.  They all have a check if they are running on a true raspberry pi. I am looking into how to modify Raspian desktop to pretent to be a pi. Cause in essense this board performs and replicates exactly the GPIO's of the pi. But my life is going to shit atm so have put everything on hold. If you wish you may close this ofcourse but if you keep it open I'll update it with my finds eventually

cp2004, 2021-05-04:
I think the underlying library checks the specific hardware because it needs to use said hardware directly. You likely won't be able to trick it into thinking you are on a Pi since it reads the hardware ID and compares against a known list. Yes, it is not the best solution but it is the only solution I have.

It was mentioned to me on Discord recently by someone asking for a similar thing, if it would work on an Orange Pi. We had a look, and there may be a way of creating a generic SPI driver for the LEDs using Python Periphery. However, that would be a long long term thing that could end up not working.

Other simple GPIO based plugins need to use something like Python Periphery to enable cross-platform support. For example, PSU control has recently upgraded to this. But most still use RPI.GPIO. 

Take your time, there is no rush 🙂

cp2004, 2021-05-08:
I am going to close this for now, so that there is no impression that this feature is accepted and will be implemented. It is incredibly difficult to create a cross platform LED control library and probably won't happen for a long time.

closed 2021-05-08 by cp2004