cp2004/OctoPrint-WS281x_LED_Status #135: [Bug] Off message not honored when the printer disconnects.

samwiseg0, 2021-05-01:
**Description of the bug**

I recently changed the ay my lights were being powered. They are now on their own power supply. I wanted to have another plugin issue the off @ command before shutting off the printer. This seems to be successful as the UI shows that the lights are powered off yet they still remain in an on state running the disconnect effect. I suspect this has to do with the fact that the printer goes into a disconnect state before the lights have had a chance to turn off or that effect is overriding the off command. 

**Steps to reproduce**
1. Issue a `@WS_LIGHTSOFF` command
2. Power off the printer immediately after. 
3. Observe that the on/off toggle is showing as off in the UI.
4. Observe that that lights remain on and continue to run the disconnected effect. 

**What did you expect to happen**

I expect when asking the lights to power off that they remain in an off state even if the printer state changes. 

**Logs and screenshots**
```
[2021-05-01 11:00:15,114] INFO: On message received, turning on LEDs to idle
[2021-05-01 11:00:23,797] INFO: Off message received, turning off LEDs (fade: True)
[2021-05-01 11:00:24,640] DEBUG: Changing effect to disconnected
[2021-05-01 11:00:40,632] INFO: On message received, turning on LEDs to disconnected
[2021-05-01 11:00:42,253] INFO: Off message received, turning off LEDs (fade: True)
[2021-05-01 11:00:58,639] DEBUG: Changing effect to idle
[2021-05-01 11:00:58,748] INFO: On message received, turning on LEDs to idle
[2021-05-01 11:01:09,772] INFO: Off message received, turning off LEDs (fade: True)
[2021-05-01 11:01:10,668] DEBUG: Changing effect to disconnected
```
[octoprint-logs.zip](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/files/6409805/octoprint-logs.zip)

**Version of plugin and OctoPrint**

WS281x LED Status (0.7.2)
Octoprint 1.6.0

**(If applicable) Browser version & browser error console**

N/A

**Additional context**

N/A

cp2004, 2021-05-02:
Untested, but going by code analysis I think I have the problem fixed.

You should be able to switch to the 'Development' Release channel and find `0.7.3dev0`, hopefully that works.

samwiseg0, 2021-05-03:
Thanks for putting that out so fast but unfortunately that did not solve the problem.

cp2004, 2021-05-03:
I'll have to give it a proper look and test at some point then. Might take a while, there's a lot on the list

samwiseg0, 2021-05-03:
> I'll have to give it a proper look and test at some point then. Might take a while, there's a lot on the list

No worries. I have a workaround. Disabling the on/off fade works.

cp2004, 2022-02-04:
I think this *should* be fixed with 0.8.0. I can't reproduce it anymore, following adjustments to the way some of this stuff was handled.

closed 2022-02-04 by cp2004