cp2004/OctoPrint-WS281x_LED_Status #138: Control Tab

ShadowOfTheDamn, 2021-05-07:
Add a LED control tab near the control tab inside octopi

this will allow the user to control the LEDs aside the status of the printer. because octopi can be used for other types of devices and not only 3d printers.
the shape of the tab is shown in these images:
[image]

[image]

The implementation of such control makes your application more general and also there are many other projects out there which are using octopi but they are not 3d printers e.g. many inhouse made devices like dynamic sculptors.

cp2004, 2021-05-08:
What would be the benefit of manually controlling LEDs in OctoPrint vs. a dedicated project such as [WLED](https://github.com/Aircoookie/WLED)?

ShadowOfTheDamn, 2021-05-08:
please See this product:
https://sisyphus-industries.com/
maybe someone is going to control his sand table CNC with octopi and the LEDs with the same hardware.

cp2004, 2021-05-08:
Hi, unfortunately I can't work on 'maybe' someone will use this for something completely different - especially not for something that would require as much work as this.

So no, I can't make a custom tab for control now. If you don't use OctoPi, use something like WLED, which would work just fine for that.

closed 2021-05-08 by cp2004

ShadowOfTheDamn, 2021-05-08:
OK man,
maybe in here means I am using this for my DIY project for a sand table controled by octopi and has 240 ws2812 LEDs. But seems you are not interested. No Problem. I would hope for you the bests. I will figure some other way for writing my own plugin. Regards