cp2004/OctoPrint-WS281x_LED_Status #141: [help needed] Blue and green color are switched

andreapede, 2021-05-23:
Strip WS2811B 12V x 28 segments 4A supply
Despite to any possible settings in (RGB, GBR, an so on) the RED is RED the BLUE is GREEN and the GREEN is BLU.
Raspberry 3b+, GPIO10
Common grounding, no flickering, everything works perfectly except I can't switch to the correct color order.

Any hint is welcomed.

[ X] I have read the Get Help Guide 

https://cp2004.gitbook.io/ws281x-led-status/guides/get-help-guide

cp2004, 2021-05-23:
Interesting. All the options for colour order are there, so I would have hoped that one of them met your strip's order. Do they make any difference, or nothing changes?

I'm not really sure what to do about this one - to the best of my knowledge swapping the colour orders worked, but I don't currently have a setup to test this right now. Will think.

andreapede, 2021-05-23:
Yes absolutely no change. I can also select the other kind of led stripe. Still works without change. Is it possible that instead is something related to where the information is stored or saved?

cp2004, 2021-05-23:
Could you enable & upload the `plugin_ws281x_led_status_debug.log` (Settings > Features > debug logging), so we can see what settings are being passed to the LED strip runner.

andreapede, 2021-05-23:
It seems the same as in the settings

2021-05-23 22:11:29,242] INFO: Initialising LED strip [2021-05-23 22:11:29,244] INFO: Strip successfully initialised [2021-05-23 22:11:29,247] DEBUG: Current settings: | * STRIP SETTINGS * | - count: 28 | - pin: 10 | - freq_hz: 800000 | - dma: 10 | - invert: False | - channel: 0 | - reverse: False | - type: WS2811_STRIP_RGB | - brightness: 100 | - adjustment: {'R': '100', 'G': 100, 'B': '101'} | - white_override: False | - white_brightness: 50 | * EFFECT SETTINGS * | startup | - enabled: True | - effect: Color Wipe | - color: #00ff00 | - delay: 75 | idle | - enabled: True | - effect: Color Wipe 2 | - color: #00ff00 | - delay: 75 | - timeout: 0 | disconnected | - enabled: True | - effect: Rainbow Cycle | - color: #000000 | - delay: 25 | failed | - enabled: True | - effect: Pulse | - color: #ff0000 | - delay: 10 | success | - enabled: True | - effect: Rainbow | - color: #000000 | - delay: 25 | - return_to_idle: 0 | paused | - enabled: True | - effect: Bounce | - color: #0000ff | - delay: 40 | printing | - enabled: False | - effect: Solid Color | - color: #ffffff | - delay: 1 | torch | - enabled: True | - effect: Solid Color | - color: #ffffff | - delay: 1 | - toggle: False | - timer: 15 | - auto_on_webcam: True | progress_print | - enabled: True | - base: #000000 | - color: #00ff00 | - effect: Progress Bar | progress_heatup | - enabled: True | - base: #0000ff | - color: #ff0000 | - tool_enabled: True | - bed_enabled: True | - tool_key: 0 | - effect: Progress Bar | progress_cooling | - enabled: True | - base: #0000ff | - color: #ff0000 | - bed_or_tool: tool | - threshold: 40 | - effect: Progress Bar | * ACTIVE TIMES * | - enabled: False | - start: 09:00 | - end: 21:00

And if I change in RBG it seems correctly passed 

2021-05-23 22:52:41,847] DEBUG: Current settings:
 | * STRIP SETTINGS *
 | - count: 28
 | - pin: 10
 | - freq_hz: 800000
 | - dma: 10
 | - invert: False
 | - channel: 0
 | - reverse: False
 | - type: WS2811_STRIP_GBR
 | - brightness: 100
 | - adjustment: {'R': '100', 'G': 100, 'B': '101'}
 | - white_override: False
 | - white_brightness: 50
 | * EFFECT SETTINGS *
 | startup
 | - enabled: True
 | - effect: Color Wipe
 | - color: #00ff00
 | - delay: 75
 | idle
 | - enabled: True
 | - effect: Color Wipe 2
 | - color: #00ff00
 | - delay: 75
 | - timeout: 0
 | disconnected
 | - enabled: True
 | - effect: Rainbow Cycle
 | - color: #000000
 | - delay: 25
 | failed
 | - enabled: True
 | - effect: Pulse
 | - color: #ff0000
 | - delay: 10
 | success
 | - enabled: True
 | - effect: Rainbow
 | - color: #000000
 | - delay: 25
 | - return_to_idle: 0
 | paused
 | - enabled: True
 | - effect: Bounce
 | - color: #0000ff
 | - delay: 40
 | printing
 | - enabled: False
 | - effect: Solid Color
 | - color: #ffffff
 | - delay: 1
 | torch
 | - enabled: True
 | - effect: Solid Color
 | - color: #ffffff
 | - delay: 1
 | - toggle: False
 | - timer: 15
 | - auto_on_webcam: True
 | progress_print
 | - enabled: True
 | - base: #000000
 | - color: #00ff00
 | - effect: Progress Bar
 | progress_heatup
 | - enabled: True
 | - base: #0000ff
 | - color: #ff0000
 | - tool_enabled: True
 | - bed_enabled: True
 | - tool_key: 0
 | - effect: Progress Bar
 | progress_cooling
 | - enabled: True
 | - base: #0000ff
 | - color: #ff0000
 | - bed_or_tool: tool
 | - threshold: 40
 | - effect: Progress Bar
 | * ACTIVE TIMES *
 | - enabled: False
 | - start: 09:00
 | - end: 21:00

andreapede, 2021-05-23:
Ok 👌
My fault. 
I missed the last line in the settings claiming for a save before get working.
Honestly i have to say that having the color test there is a bit misleading.
But it works. Thank you.

cp2004, 2021-05-23:
Cool, good to know it is sorted. Maybe I can add a small note to the test page as well, to say that you have to save for any changes to take effect.

closed 2021-05-23 by cp2004