cp2004/OctoPrint-WS281x_LED_Status #143: Not able to control NeoPixel Ring of 12 LEDs

micheld93, 2021-06-20:
**What do you need help with?**
I have a good working Octopi attached to my Prusa Mini but I'm unable to control my used Neopixel 12 LED-Ring. I used the wiring mentioned here as well as the neopixel ring used in the tutorial: https://towinet.de/raspberry-pi-kamera-clip-mit-integriertem-neopixel-ring.

The log says that the SPI couldn't be initialised but I have no display attached. The only other device connected to the Pi is a Pi-camera. I'm also wondering why the log says "ws2811" because I changed it to "ws2812 Strip". 

Is there anyone who can help me?

**Log files**
[2021-06-20 17:26:42,614] INFO: Initialising LED strip
[2021-06-20 17:26:42,617] ERROR: RuntimeError('ws2811_init failed with code -13 (Unable to initialize SPI)')
[2021-06-20 17:26:42,618] ERROR: Strip failed to initialize, no effects will be run.
[2021-06-20 17:26:42,619] INFO: No strip initialised, exiting the effect process.

* [x ] I have read the Get Help Guide 
But was unable to solve my problem

https://cp2004.gitbook.io/ws281x-led-status/guides/get-help-guide

cp2004, 2021-06-20:
Please upload the full log files, the octoprint.log as well.

micheld93, 2021-06-20:
Sorry, here the complete Octoprint.log:

`2021-06-20 16:30:26,834 - octoprint.plugins.tracking - INFO - Sent tracking event pong, payload: {'version': '1.6.1', 'os': 'linux', 'bits': 32, 'python': '3.7.3', 'pip': '20.3.3', 'cores': 4, 'freq': 1200.0, 'ram': 915722240, 'pi_model': 'Raspberry Pi 3 Model B Rev 1.2', 'octopi_version': '0.18.0', 'plugins': 'prusaslicerthumbnails:1.0.0,ws281x_led_status:0.7.2,simpleemergencystop:1.0.5,prusa_mini_eta:0.1.3,octopod:0.3.0,navbartemp:0.14,heatertimeout:0.0.3,floatingnavbar:0.3.7,dashboard:1.18.2,octolapse:0.4.1,displaylayerprogress:1.26.0'}
2021-06-20 16:30:26,836 - octoprint.server - INFO - --- Log roll over detected ---------------------------------------------------
2021-06-20 16:30:26,837 - octoprint.server - INFO - OctoPrint 1.6.1
2021-06-20 16:30:26,840 - octoprint.plugin.core - INFO - 28 plugin(s) registered with the system:
|  Action Command Notification Support (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/action_command_notification
|  Action Command Prompt Support (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/action_command_prompt
|  Announcement Plugin (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/announcements
|  Anonymous Usage Tracking (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/tracking
|  Application Keys Plugin (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/appkeys
|  Backup & Restore (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/backup
|  Core Wizard (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/corewizard
|  Dashboard (1.18.2) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_dashboard
|  Discovery (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/discovery
|  DisplayLayerProgress Plugin (1.26.0) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_DisplayLayerProgress
|  Error Tracking (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/errortracking
|  File Check (2021.2.23) (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_file_check
|  Firmware Check (2021.5.31) (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_firmware_check
|  Floating Navbar (0.3.7) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_floatingnavbar
|  GCode Viewer (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/gcodeviewer
|  HeaterTimeout (0.0.3) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_HeaterTimeout
|  Logging (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/logging
|  Navbar Temperature Plugin (0.14) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_navbartemp
|  Octolapse (0.4.1) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_octolapse
|  OctoPod Plugin (0.3.0) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_octopod
|  Pi Support Plugin (2021.6.14) (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_pi_support
|  Plugin Manager (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/pluginmanager
|  Prusa Mini ETA Plugin (0.1.3) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_prusa_mini_eta
|  Simple Emergency Stop (1.0.5) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_simpleemergencystop
|  Slicer Thumbnails (1.0.0) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_prusaslicerthumbnails
|  Software Update (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/softwareupdate
|  Virtual Printer (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/virtual_printer
|  WS281x LED Status (0.7.2) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_ws281x_led_status
Prefix legend: ! = disabled, # = blacklisted, * = incompatible
2021-06-20 16:30:26,849 - octoprint.environment - INFO - Detected environment is Python 3.7.3 under Linux (linux). Details:
|  hardware:
|    cores: 4
|    freq: 1200.0
|    ram: 915722240
|  os:
|    bits: 32
|    id: linux
|    platform: linux
|  plugins:
|    pi_support:
|      model: Raspberry Pi 3 Model B Rev 1.2
|      octopi_version: 0.18.0
|      throttle_state: '0x0'
|  python:
|    pip: 20.3.3
|    version: 3.7.3
|    virtualenv: /home/pi/oprint
2021-06-20 16:30:26,850 - octoprint.server - INFO - ------------------------------------------------------------------------------
2021-06-20 16:43:54,004 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2021-06-20 16:43:54,700 - octoprint.plugins.tracking - INFO - Sent tracking event ping, payload: {'octoprint_uptime': 81020, 'printer_state': 'OFFLINE'}
2021-06-20 16:58:54,006 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2021-06-20 16:58:54,716 - octoprint.plugins.tracking - INFO - Sent tracking event ping, payload: {'octoprint_uptime': 81920, 'printer_state': 'OFFLINE'}
2021-06-20 17:02:32,026 - octoprint.server.util.sockjs - INFO - New connection from client: ::ffff:192.168.188.26
2021-06-20 17:02:34,343 - octoprint.access.users - INFO - Logged in user: octopi
2021-06-20 17:02:34,347 - octoprint.server.api - INFO - Actively logging in user octopi from ::ffff:192.168.188.26
2021-06-20 17:02:34,462 - octoprint.server.util.flask - INFO - Passively logging in user octopi from ::ffff:192.168.188.26
2021-06-20 17:02:34,463 - octoprint.access.users - INFO - Logged in user: octopi
[... cut ...]
2021-06-20 19:22:05,609 - octoprint.events - DEBUG - Subscribed listener <bound method PrinterStateConnection._onEvent of <octoprint.server.util.sockjs.PrinterStateConnection object at 0x63fb7190>> for event plugin_softwareupdate_update_failed
2021-06-20 19:22:10,094 - octoprint.plugins.octopod - DEBUG - response from sarge: temp=42.9'C

2021-06-20 19:22:10,095 - octoprint.plugins.octopod - DEBUG - used pattern: "=(.*)'"
2021-06-20 19:22:10,095 - octoprint.plugins.octopod - DEBUG - match: 42.9
2021-06-20 19:22:10,096 - octoprint.plugins.octopod - DEBUG - SoC Temp: 42.9
`

cp2004, 2021-06-20:
Well, that log file doesn't show me the startup of the server so I can't tell what is setup on your environment. And you can just attach logs as full files, it makes them much easier to read, please remember this in future.

You need to enable SPI to use this plugin. Run the OS configuration test, ensure that SPI is enabled, open your `/boot/config.txt` file and check for `dtparam=spi=on` in this file, uncommented.

micheld93, 2021-06-22:
I didn't remember that I can upload full logs, a new one can be found here: 
[octoprint-2.log](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/files/6696474/octoprint-2.log)

SPI is activated. I checked the LEDs with the implemented tests for different colours (in Utilities section) but the colour is not changing. I also executed a reboot with connected LEDs and SPI on but no change.

cp2004, 2021-06-22:
According to the log file, two of the OS configuration tests failed, one for the SPI buffer and the other for the core frequency. They should be fixed, otherwise you may not be able to control the LEDs. You can use the built in OS configuration test for this.

micheld93, 2021-06-26:
I fixed the problems with the built in OS config. I was able to set the colour (red or green) several times with a huge time delay but not every time I tried it. I think I need use another supply for the neo pixel ring. I try it and will report what’s the result.

cp2004, 2021-06-26:
Great, keep me posted - there's not much I can do about the delay, as the only place I can think of is that it is the network/API connection that is slow. Do other things (like pressing buttons in the control tab) have as large of a delay? The power supply may be an issue, as the LEDs might not be picking up the commands right if they are underpowered.

cp2004, 2021-07-09:
It sounds like this issue was solved - if it was not, please let me know.

closed 2021-07-09 by cp2004

micheld93, 2021-07-09:
Sorry. I tried it with a 5V/2A supply and it works very well. It was just not enough power. Thanks for helping.