cp2004/OctoPrint-WS281x_LED_Status #147: [Bug]: Not connecting to printer.

webdiddy, 2021-06-28:
### Description of the bug

Can not connect to printer with this and the PSU control plugin active at the same time.
Disabling either this plugin or psu control plugin resolves the issue.

Here's the messages from terminal.

Changing monitoring state from "Offline" to "Detecting serial connection"
Performing autodetection with 1 port/baudrate candidates: /dev/ttyACM0@115200
Trying port /dev/ttyACM0, baudrate 115200
Connecting to port /dev/ttyACM0, baudrate 115200
Handshake attempt #1 with timeout 10.0s
Connected to: Serial<id=0x691e2e90, open=True>(port='/dev/ttyACM0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Send: N0 M110 N0*125
Recv: Command not found!
Handshake attempt #2 with timeout 10.0s
Send: N0 M110 N0*125
Handshake attempt #3 with timeout 10.0s
Send: N0 M110 N0*125
Changing monitoring state from "Detecting serial connection" to "Error"
Changing monitoring state from "Error" to "Offline after error"
Connection closed, closing down monitor

### Steps to reproduce

Install both plugins and activate them.

### What happened instead?

Will not connect to printer.

### Plugin version

Latest

### OctoPrint version

Latest

### (if relevant) OctoPi version

4B+

cp2004, 2021-06-28:
Could you upload a log where you have a successful connect?

webdiddy, 2021-06-28:
Sure. I can do that later tonight. Would you like a log with your plugin
enabled and psu control disabled or one with psu control enabled and your
plug in disabled?

Best regards
Tony

cp2004, 2021-06-28:
Either would be fine - I just want to see what a successful connect looks like, to compare what it should be doing. I have no idea how either of these plugins would break this, so there may be a lot of back and forth to try and solve the problem. 

Also, one more question - how do you connect to your printer? USB or serial through the GPIO pins?

webdiddy, 2021-06-28:
Hi.

Is this enough? I just copied it from the terminal.

Changing monitoring state from "Offline" to "Detecting serial connection"
Performing autodetection with 1 port/baudrate candidates: ***@***.***
Trying port /dev/ttyACM0, baudrate 115200
Connecting to port /dev/ttyACM0, baudrate 115200
Handshake attempt #1 with timeout 10.0s
Connected to: Serial<id=0x689d9390, open=True>(port='/dev/ttyACM0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Send: N0 M110 N0*125
Recv: \x00start
WARNThe received line contains at least one null byte character at position 0, this hints at some data corruption going on
Changing monitoring state from "Detecting serial connection" to "Operational"
Send: N0 M110 N0*125
Recv: echo: 3.10.0-4990
Recv: echo: Last Updated: May 12 2021 16:57:04 | Author: (none, default config)
Recv: Compiled: May 12 2021
Recv: echo: Free Memory: 2052 PlannerBufferBytes: 1760
Recv: echo:Stored settings retrieved
Recv: adc_init
Recv: Extruder fan type: NOCTUA
Recv: CrashDetect ENABLED!
Recv: FSensor ENABLED (sensor board revision: 0.4 or newer)
Recv: Sending 0xFF
Recv: echo:SD card ok
Recv: echo:busy: processing
Printer seems to support the busy protocol, will adjust timeouts and set busy interval accordingly
Recv: LCD status changed
Recv: ok
Send: N1 M115*39
Recv: FIRMWARE_NAME:Prusa-Firmware 3.10.0 based on Marlin FIRMWARE_URL:https://github.com/prusa3d/Prusa-Firmware PROTOCOL_VERSION:1.0 MACHINE_TYPE:Bondtech MK3S EXTRUDER_COUNT:1 UUID:00000000-0000-0000-0000-000000000000
Recv: Cap:AUTOREPORT_TEMP:1
Recv: Cap:AUTOREPORT_FANS:1
Recv: Cap:AUTOREPORT_POSITION:1
Recv: Cap:EXTENDED_M20:1
Recv: ok
Send: M21
Recv: Sending 0xFF
Recv: echo:SD card ok
Recv: ok
Send: M20
Recv: Begin file list
Recv: FRAGPL~1.GCO 2182551
Recv: APPLE_~1.GCO 4931774
Recv: FRAGPL~2.GCO 3344996
Recv: MK3PET~1.GCO 1614438
Recv: MK3_PL~1.GCO 1586103
Recv: End file list
Recv: ok
Send: M113 S9
Recv: ok
Send: M155 S2
Recv: ok
Send: M105
Recv: ok T:23.5 /0.0 B:24.5 /0.0 T0:23.5 /0.0 @:0 B@:0 P:54.4 A:25.5
Recv: T:24.3 /0.0 B:25.1 /0.0 T0:24.3 /0.0 @:0 B@:0 P:28.0 A:24.8
Recv: T:24.4 /0.0 B:24.9 /0.0 T0:24.4 /0.0 @:0 B@:0 P:25.0 A:24.7

cp2004, 2021-06-29:
I have no idea why this plugin or PSU control would stop you from connecting to your printer. In the initial log, it was connecting to  /dev/ttyACM0, in the second it also connects with /dev/ttyACM0 and the printer responds properly, whereas previously it responded `Recv: Command not found!`. The only thing I can think of is some kind of interference created by both plugins trying to use the GPIO.

Is your printer connected using USB or the GPIO pins?

webdiddy, 2021-06-29:
Yes.

It's very strange. I have octoprint connected to my printer using USB.

cp2004, 2021-07-09:
Still no ideas why this plugin can stop your printer from replying to OctoPrint's commands properly, check that your LED strip setup is not causing interference with the printer's USB connection.

Since I am out of ideas, you may be better asking on the [community forums](https://community.octoprint.org) or the [OctoPrint discord server](https://community.octoprint.org) to see if anyone else has anything to say.

closed 2021-07-09 by cp2004

webdiddy, 2021-07-09:
Okay. No problem dude.
Thanks anyway for your effort and a nice plugin.