cp2004/OctoPrint-WS281x_LED_Status #150: [Errno 2] No such file or directory: 'LICENSE.txt'

raspbounty, 2021-07-13:
### Describe your problem

I am trying to Install the Plugin with the Pluginmanager via the URL. When I do this I get the following output and cannot find out what is causing this error or how to resolve it. Although it says Successfully installed it later doesn't turn on the LEDs so I am curious if that is the reason for my problem.

```
2021-07-13 21:41:53,770 ! creating OctoPrint_WS281x_LED_Status.egg-info
2021-07-13 21:41:53,771 ! writing requirements to OctoPrint_WS281x_LED_Status.egg-info/requires.txt
2021-07-13 21:41:53,772 ! writing OctoPrint_WS281x_LED_Status.egg-info/PKG-INFO
2021-07-13 21:41:53,772 ! writing top-level names to OctoPrint_WS281x_LED_Status.egg-info/top_level.txt
2021-07-13 21:41:53,772 ! writing dependency_links to OctoPrint_WS281x_LED_Status.egg-info/dependency_links.txt
2021-07-13 21:41:53,773 ! writing entry points to OctoPrint_WS281x_LED_Status.egg-info/entry_points.txt
2021-07-13 21:41:53,773 ! writing manifest file 'OctoPrint_WS281x_LED_Status.egg-info/SOURCES.txt'
2021-07-13 21:41:53,774 ! warning: Failed to find the configured license file 'LICENSE.txt'
2021-07-13 21:41:53,774 ! reading manifest file 'OctoPrint_WS281x_LED_Status.egg-info/SOURCES.txt'
2021-07-13 21:41:53,774 ! writing manifest file 'OctoPrint_WS281x_LED_Status.egg-info/SOURCES.txt'
2021-07-13 21:41:53,775 ! Copying OctoPrint_WS281x_LED_Status.egg-info to build/bdist.linux-armv7l/wheel/OctoPrint_WS281x_LED_Status-0.7.2-py2.7.egg-info
2021-07-13 21:41:53,775 ! running install_scripts
2021-07-13 21:41:53,776 ! error: [Errno 2] No such file or directory: 'LICENSE.txt'
2021-07-13 21:41:53,776 ! ----------------------------------------
2021-07-13 21:41:53,776 ! ERROR: Failed building wheel for OctoPrint-WS281x-LED-Status
2021-07-13 21:41:54,509 > Failed to build OctoPrint-WS281x-LED-Status
2021-07-13 21:41:54,509 > Installing collected packages: OctoPrint-WS281x-LED-Status
2021-07-13 21:41:54,510 > Running setup.py install for OctoPrint-WS281x-LED-Status: started
2021-07-13 21:41:56,091 > Running setup.py install for OctoPrint-WS281x-LED-Status: finished with status 'done'
2021-07-13 21:41:56,092 > Successfully installed OctoPrint-WS281x-LED-Status-0.7.2
```

### Please confirm the following

- [X] I have read the Get Help & Troubleshooting guide

### Log files

[octoprint.log](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/files/6811572/octoprint.log)
[plugin_pluginmanager_console.log](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/files/6811581/plugin_pluginmanager_console.log)
[plugin_ws281x_led_status_debug.log](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/files/6811582/plugin_ws281x_led_status_debug.log)

cp2004, 2021-07-13:
If it is successfully installed & **shows up inside OctoPrint** (which it does, it is in the log) then this missing file is not the issue. It is fixed by PR #142 and will be part of the next release. Pip falls back to an older style of installing the plugin, which works just as well.

Since the plugin is loading correctly and all the OS config tests pass, then you should check all the wiring against the guides and try and verify that all of your hardware works - from the software's point of view everything is going fine.

raspbounty, 2021-07-13:
Okay thanks a lot for your answer. Good to know that that's not the problem and I will look again into my wiring.

closed 2021-07-13 by raspbounty

cp2004, 2021-07-13:
Feel free to ask further questions if you need - and be sure to check through all the docs & troubleshooting guides as many common problems are listed. If you find your issue and want to add it to the list, you can contribute or let me know how you solved it, so I can help others going forward.

raspbounty, 2021-07-21:
I figured out what my problem was and I just wired the LED-Strip backwards. And since I didn't think about this as a problem I tried everything else what could be wrong first.

planetar, 2021-09-08:
While this issue report is set to 'closed' the issue - 

"error: [Errno 2] No such file or directory: 'LICENSE.txt'
----------------------------------------
ERROR: Failed building wheel for OctoPrint-WS281x-LED-Status"

persists. I was about to write a new report about it when I finally discovered it with the closed issues.

cp2004, 2021-09-08:
It's solved in the development branch already, I will get round to releasing it at some point. It also does not prevent installing the plugin, so there's no need for a patch release.