cp2004/OctoPrint-WS281x_LED_Status #154: Lights not turning off

frozenemberz, 2021-07-31:
I have a ws2812b light strip connected to 5 volt power supply no level shifting directly to the pie for the data pin and I can't seem to find a way to get the lights to shut off. They turn on fine like torch mode will turn on but will not turn off after 15 seconds like I set in the settings. I disabled all effects and enabled only printing solid white and when the print finishes the lights are still on. I also tried settings the idle effect to shut off after 30 seconds and it just continues forever. Is this because I'm not using level shifting or setting it up wrong?

frozenemberz, 2021-08-01:
I think I might be setting something wrong. I checked the log file and it looks like it's not turning off the lights from the software. At the very end you can see it changed to successful print and never shut off after the specified interval and after 10 minutes I turned on torch mode just to see if it was logging it. Here's the log..
[2021-08-01 00:42:57,748] INFO: On message received, turning on LEDs to blank
[2021-08-01 00:47:08,339] INFO: Initialising LED strip
[2021-08-01 00:47:08,344] INFO: Strip successfully initialised
[2021-08-01 00:47:08,346] DEBUG: Current settings:
 | * STRIP SETTINGS *
 | - count: 90
 | - pin: 10
 | - freq_hz: 800000
 | - dma: 10
 | - invert: False
 | - channel: 0
 | - reverse: False
 | - type: WS2812_STRIP
 | - brightness: 26
 | - adjustment: {'R': 100, 'G': 100, 'B': 100}
 | - white_override: False
 | - white_brightness: 50
 | * EFFECT SETTINGS *
 | startup
 | - enabled: False
 | - effect: Color Wipe
 | - color: #00ff00
 | - delay: 75
 | idle
 | - enabled: False
 | - effect: Bouncy Balls
 | - color: #21a1de
 | - delay: 75
 | - timeout: 61
 | disconnected
 | - enabled: False
 | - effect: Rainbow Cycle
 | - color: #000000
 | - delay: 25
 | failed
 | - enabled: False
 | - effect: Pulse
 | - color: #ff0000
 | - delay: 10
 | success
 | - enabled: True
 | - effect: Rainbow
 | - color: #000000
 | - delay: 25
 | - return_to_idle: 18
 | paused
 | - enabled: False
 | - effect: Bounce
 | - color: #0000ff
 | - delay: 40
 | printing
 | - enabled: True
 | - effect: Solid Color
 | - color: #ffffff
 | - delay: 1
 | torch
 | - enabled: True
 | - effect: Solid Color
 | - color: #ffffff
 | - delay: 1
 | - toggle: False
 | - timer: 15
 | - auto_on_webcam: False
 | progress_print
 | - enabled: False
 | - base: #00ff00
 | - color: #0000ff
 | - effect: Progress Bar
 | progress_heatup
 | - enabled: False
 | - base: #0000ff
 | - color: #ff0000
 | - tool_enabled: True
 | - bed_enabled: True
 | - tool_key: 0
 | - effect: Progress Bar
 | progress_cooling
 | - enabled: False
 | - base: #0000ff
 | - color: #ff0000
 | - bed_or_tool: tool
 | - threshold: 40
 | - effect: Progress Bar
 | * ACTIVE TIMES *
 | - enabled: False
 | - start: 09:00
 | - end: 21:00
[2021-08-01 00:47:08,381] INFO: On message received, turning on LEDs to blank
[2021-08-01 01:11:08,007] DEBUG: Changing effect to printing
[2021-08-01 03:40:04,802] DEBUG: Changing effect to success
[2021-08-01 03:50:08,327] DEBUG: Changing effect to torch

cp2004, 2021-08-01:
Hi, thanks for providing the log file - I'm not sure how you managed to avoid the issue forms from [here](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/issues/new/choose) since I can't select a blank one, but at least I have the log file to look at!

I am currently not sure *exactly* why the timers are not firing. When I was last working on the plugin, I noticed that the timers are cancelled before the checks to see what is enabled, so it could be that there is an 'invisible' effect that is cancelling the timers, but not making it to your strip because it is not enabled. There are certain combinations where I think this can cause problems.

In the coming days I may be able to push something to the development branch that logs the cancelling of the timers, or what else may be causing issues. Stay tuned!

frozenemberz, 2021-08-02:
I'm not very fluent with GitHub yet but I'm using the Android app right now and I just clicked create new issue. To be honest it's probably just the way I'm using it I cleared all the data and said it to defaults. Am I supposed to turn the big red button to the green position? I noticed if I don't turn it on then the effects come on automatically when I print and it did shut off. I'm rather new to rgb's and your plugin but the colors and effects are great and I just wanted to make sure the ws2812b is a good choice for my new printer or there's a better one.

cp2004, 2021-08-16:
Apologies, have been busy the last few days and I think I forgot to reply here.

> Am I supposed to turn the big red button to the green position? I noticed if I don't turn it on then the effects come on automatically when I print and it did shut off.

I think I am understanding what is going on here, but I have discovered a couple of logic issues/forgotten cases with the timers. I have reordered some of the processing and communication with the LED strip in 707fb5d7bd1b2ff3856eeadad7da5e4f68c58ddf which *might* fix this issue, but I think there's more to do here. It is not tested enough for a build to be published, but I am aiming to have any issues with the timers resolved for the 0.8.0 release. Not loads more to do for it, and I might delay some things to get it out sooner rather than waiting more months.

cp2004, 2021-10-02:
If you are still having an issue with this, please try switching to the 'Development' release channel in the software update section, and download & install 0.8.0dev0. I think some of the bugs have gone with some better written code...

cp2004, 2022-02-04:
0.8.0 has been released (finally...!)

closed 2022-02-04 by cp2004