cp2004/OctoPrint-WS281x_LED_Status #156: Problems with the color changing

RiCeyArts, 2021-08-14:
### Describe your problem

Hello,
I have the WS2812B and then I click the test buttons it won´t change the color. Only after a view time of clicking it chances it. I use the SN74AHCT125N as level shifter and wired it like in the document.
I think it is a problem with the installation because there is an error everytime i installed it but i took the changes in the config.txt.
I test the strip with an arduino and it worked perfectly.
Could you please help me!

### Please confirm the following

- [X] I have read the Get Help & Troubleshooting guide

### Log files

Reinstalliere Plugin "WS281x LED Status" von https://github.com/cp2004/OctoPrint-WS281x_LED_Status/releases/latest/download/release.zip...
/home/pi/oprint/bin/python2 -m pip --disable-pip-version-check install file:///tmp/tmpYRwmzi/release.zip --no-cache-dir
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Processing /tmp/tmpYRwmzi/release.zip
Requirement already satisfied: OctoPrint in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint-WS281x-LED-Status==0.7.2) (1.6.1)
Requirement already satisfied: rpi_ws281x>=4.2.5 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint-WS281x-LED-Status==0.7.2) (4.3.0)
Requirement already satisfied: zipstream-new<1.2,>=1.1.8 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-WS281x-LED-Status==0.7.2) (1.1.8)
Requirement already satisfied: Flask-Babel<2,>=1.0 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-WS281x-LED-Status==0.7.2) (1.0.0)
Requirement already satisfied: watchdog==0.10.4 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-WS281x-LED-Status==0.7.2) (0.10.4)
Requirement already satisfied: psutil<6,>=5.7 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-WS281x-LED-Status==0.7.2) (5.7.2)
Requirement already satisfied: frozendict<2,>=1.2; python_version < "3" in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-WS281x-LED-Status==0.7.2) (1.2)
Requirement already satisfied: emoji<1,>=0.5.4 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-WS281x-LED-Status==0.7.2) (0.5.4)
Requirement already satisfied: monotonic<2,>=1.5; python_version < "3" in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-WS281x-LED-Status==0.7.2) (1.5)
Requirement already satisfied: filetype<2,>=1.0.7 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-WS281x-LED-Status==0.7.2) (1.0.7)
Requirement already satisfied: PyYAML<6,>=5.3.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-WS281x-LED-Status==0.7.2) (5.3.1)
Requirement already satisfied: feedparser<6,>=5.2.1; python_version < "3" in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-WS281x-LED-Status==0.7.2) (5.2.1)
Requirement already satisfied: netifaces<1,>=0.10.9 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-WS281x-LED-Status==0.7.2) (0.10.9)
Requirement already satisfied: future<1,>=0.18.2 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-WS281x-LED-Status==0.7.2) (0.18.2)
Requirement already satisfied: enum34<1.2,>=1.1.10; python_version < "3" in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-WS281x-LED-Status==0.7.2) (1.1.10)
Requirement already satisfied: netaddr<1,>=0.7.19 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-WS281x-LED-Status==0.7.2) (0.7.19)
Requirement already satisfied: requests<3,>=2.23.0 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-WS281x-LED-Status==0.7.2) (2.24.0)
Requirement already satisfied: Click<8,>=7.1.2 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-WS281x-LED-Status==0.7.2) (7.1.2)
Requirement already satisfied: itsdangerous<2,>=1.1.0 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-WS281x-LED-Status==0.7.2) (1.1.0)
Requirement already satisfied: blinker<2,>=1.4 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-WS281x-LED-Status==0.7.2) (1.4)
Requirement already satisfied: OctoPrint-PiSupport>=2021.3.26 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-WS281x-LED-Status==0.7.2) (2021.8.2)
Requirement already satisfied: Jinja2<3,>=2.11.2 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-WS281x-LED-Status==0.7.2) (2.11.2)
Requirement already satisfied: Flask-Login<0.6,>=0.5 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-WS281x-LED-Status==0.7.2) (0.5.0)
Requirement already satisfied: flask<2,>=1.1.2 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-WS281x-LED-Status==0.7.2) (1.1.2)
Requirement already satisfied: semantic-version<3,>=2.8.5 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-WS281x-LED-Status==0.7.2) (2.8.5)
Requirement already satisfied: unidecode<0.05,>=0.04.14 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-WS281x-LED-Status==0.7.2) (0.4.21)
Requirement already satisfied: pkginfo<2,>=1.5.0.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-WS281x-LED-Status==0.7.2) (1.5.0.1)
Requirement already satisfied: futures<4,>=3.3; python_version < "3" in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-WS281x-LED-Status==0.7.2) (3.3.0)
Requirement already satisfied: websocket-client<1,>=0.57 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-WS281x-LED-Status==0.7.2) (0.57.0)
Requirement already satisfied: scandir<2,>=1.10; python_version < "3" in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-WS281x-LED-Status==0.7.2) (1.10.0)
Requirement already satisfied: tornado==5.1.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-WS281x-LED-Status==0.7.2) (5.1.1)
Requirement already satisfied: OctoPrint-FileCheck>=2021.2.23 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-WS281x-LED-Status==0.7.2) (2021.2.23)
Requirement already satisfied: pyserial<4,>=3.4 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-WS281x-LED-Status==0.7.2) (3.4)
Requirement already satisfied: markupsafe<2.0,>=1.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-WS281x-LED-Status==0.7.2) (1.1.1)
Requirement already satisfied: markdown<3.2,>=3.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-WS281x-LED-Status==0.7.2) (3.1.1)
Requirement already satisfied: regex!=2018.11.6 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-WS281x-LED-Status==0.7.2) (2019.8.19)
Requirement already satisfied: chainmap<2,>=1.0.3; python_version < "3" in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-WS281x-LED-Status==0.7.2) (1.0.3)
Requirement already satisfied: typing<4,>=3.7.4.1; python_version < "3" in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-WS281x-LED-Status==0.7.2) (3.7.4.1)
Requirement already satisfied: sarge==0.1.5post0 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-WS281x-LED-Status==0.7.2) (0.1.5.post0)
Requirement already satisfied: sentry-sdk<1,>=0.15.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-WS281x-LED-Status==0.7.2) (0.17.6)
Requirement already satisfied: OctoPrint-FirmwareCheck>=2021.2.4 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-WS281x-LED-Status==0.7.2) (2021.8.11)
Requirement already satisfied: pylru<2,>=1.2 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-WS281x-LED-Status==0.7.2) (1.2.0)
Requirement already satisfied: werkzeug<2,>=1.0.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-WS281x-LED-Status==0.7.2) (1.0.1)
Requirement already satisfied: wrapt<2,>=1.12.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-WS281x-LED-Status==0.7.2) (1.12.1)
Requirement already satisfied: Flask-Assets<3,>=2.0 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-WS281x-LED-Status==0.7.2) (2.0)
Requirement already satisfied: cachelib<1,>=0.1 in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint->OctoPrint-WS281x-LED-Status==0.7.2) (0.1)
[... cut ...]
Copying OctoPrint_WS281x_LED_Status.egg-info to build/bdist.linux-armv7l/wheel/OctoPrint_WS281x_LED_Status-0.7.2-py2.7.egg-info
running install_scripts
error: [Errno 2] No such file or directory: 'LICENSE.txt'
----------------------------------------
ERROR: Failed building wheel for OctoPrint-WS281x-LED-Status
Failed to build OctoPrint-WS281x-LED-Status
Installing collected packages: OctoPrint-WS281x-LED-Status
Running setup.py install for OctoPrint-WS281x-LED-Status: started
Running setup.py install for OctoPrint-WS281x-LED-Status: finished with status 'done'
Successfully installed OctoPrint-WS281x-LED-Status-0.7.2

RiCeyArts, 2021-08-14:
> ### Describe your problem
> Hello,
> I have the WS2812B and then I click the test buttons it won´t change the color. Only after a view time of clicking it chances it. I use the SN74AHCT125N as level [...]

cp2004, 2021-08-16:
Could you provide the `octoprint.log` & the `plugin_ws281x_led_status_debug.log` file *with debug logging enabled* as requested in the issue form template?

 The installation error is unrelated, the plugin still installed successfully and it is already fixed for a future version (See #150 #142)

RiCeyArts, 2021-08-16:
[octoprint.log](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/files/6994331/octoprint.log)
[plugin_ws281x_led_status_debug.log](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/files/6994335/plugin_ws281x_led_status_debug.log)
Here are the two files you asked for. I found out that the colors are overwritten by white and when I click on the settings button the trip reacts on it.

cp2004, 2021-08-28:
Apologies for the delay in replying here, I've been away and just got back to have a look at the issue queue.

Is this issue still relevant? From the log file it looks like the strip should be receiving the correct commands. If you are still having an issue, then please check that the OS configuration test passes. If it does, then please post the contents of the file at `/boot/config.txt` - it can be accessed over SSH, or by shutting down the Pi and putting the SD card in another device.

The other thing to look at is to make sure you have a sufficient power supply for the Pi. I have seen that when the Pi throttles itself, it can mess up with the timings of the LEDs and they don't display the proper colours.

closed 2021-09-19 by github-actions[bot]