cp2004/OctoPrint-WS281x_LED_Status #157: ich habe eine frage zur Unterstützung anderer LED streifen

waf1tb, 2021-08-17:
### Describe your problem

Ich habe noch Reste von meinen ws 2801 led´s gibt es eine Möglichkeit die Unterstützung dieser led´s auch noch einzubauen.
Habe sie noch von einem alten Ambiligt über.

mfg
timo

I still have remnants of my ws 2801 leds. There is a possibility to add support for these leds.
Got it from an old Ambiligt.

Kind regards
timo

### Please confirm the following

- [X] I have read the Get Help & Troubleshooting guide

cp2004, 2021-08-17:
You would first have to either find a Python library to talk these LEDs protocols (I am not able to implement support myself) or get support added in in the current library https://github.com/jgarff/rpi_ws281x - which is barely maintained beyond the occasional contributions so I don't think you will get very far.

Sorry, but I haven't got the time, hardware or ability to write custom drivers for these chips.

closed 2021-08-17 by cp2004

waf1tb, 2021-08-17:
ok danke für die schnelle Antwort.
werde mir dann ein paar ws281 besorgen.
welches sind die besten für dieses Projekt.
was könne sie mir empfehlen.

mfg
timo

ok thanks for the quick reply.
will then get me some ws281.
which are the best for this project.
what could she recommend to me.

Mfg
Timo

cp2004, 2021-08-17:
I have no recommendation of LED strips generally, since I buy them directly from China for the cheapest rates at the time.

You can get the branded 'Neopixels', but these tend to be more expensive for the same thing.

I have success with WS2812 LEDs, but any of WS2811, WS2812 or SK6812 LEDs should work with this project.

waf1tb, 2021-08-17:
ok danke
mfg
timo