cp2004/OctoPrint-WS281x_LED_Status #159: Resetting success/failure states back to idle

kdegraaf, 2021-08-30:
### Describe your problem

Hello,

First, thank you for your efforts on this plugin.  It has worked well for me and added lots of value to my OctoPrint experience.  :-)

Perhaps I've missed something, but is there a method of acknowledging a printer success/failure state and transitioning back to the idle state?  That is, if I've collected a good print or cleaned up a failed one, I'd like to get rid of the green/red indication.

I'd be okay with doing this from the Web UI, but would really love some way to trigger that via the Physical Button plugin, as I've made a custom push-button controller.

Thanks!

### Please confirm the following

- [X] I have read the Get Help & Troubleshooting guide

cp2004, 2021-08-30:
> Perhaps I've missed something, but is there a method of acknowledging a printer success/failure state and transitioning back to the idle state? That is, if I've collected a good print or cleaned up a failed one, I'd like to get rid of the green/red indication.

Nothing missed there, it doesn't exist. The only thing that *does* exist is the 'return to idle' timer. Which could be *slightly* dodgy as I think there's some bugs with priority and the timer getting cancelled there.

cp2004, 2021-09-05:
When I get round to implementing #125 you may be able to do this from custom triggers in WS281x LED Status, and then you can setup physical button to use this.

*But*, that may be a fairly long way off. I have planned & done some of the work there but I'm not sure if it will make it into the next release or not. You'll notice when it does get updated, because I will be *plastering* this across the plugin's docs & release notes.

closed 2021-09-28 by github-actions[bot]