cp2004/OctoPrint-WS281x_LED_Status #161: Can only control the first LED on the strip

SearchForSunnyD, 2021-09-18:
### Describe your problem

Ender 3 v2
Raspberry PI 4B w/ Octoprint
WS2812B RGB LED Strip w/ 300 LEDs
I have full control over the one LED, but none of the others are turning on. I have checked that the entire strip has power and there is data continuity throughout the strip. Am confused, pls help.

### Please confirm the following

- [X] I have read the Get Help & Troubleshooting guide

cp2004, 2021-09-19:
> ### Log files
> No response

As it said on the issue form, please provide the log files. Follow the Get Help guide, and enable debug logging & upload the logs.

SearchForSunnyD, 2021-09-22:
Edited with logs

SearchForSunnyD, 2021-09-22:
[plugin_ws281x_led_status_debug (1).log](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/files/7214327/plugin_ws281x_led_status_debug.1.log)