cp2004/OctoPrint-WS281x_LED_Status #163: [Request]: Klipper support

birkanozer, 2021-09-29:
### What would you like implemented?

Is it possible to use his plugin with Klipper? 

### Why should this be implemented?

Klipper is highly sophisticated firmware for 3D printers. But build-in neopixel support isn't as sophisticated as this octoprint plugin.

cp2004, 2021-09-29:
There shouldn't be anything else you need to do, this plugin works with OctoPrint, regardless of what printer/firmware you use.

birkanozer, 2021-09-30:
Hi,
Thanks for the answer. But I didn't mean RaspberryPi interface. Most of the printer motherboard has dedicated NeoPixel interface on PCB. This type of connection is better in terms of power management. Klipper, despite other firmwares, can accept real-time API calls from Pi. We can define the motherboard pin (ex. PA2)  instead of the Pi's GPIO pin and send commands to the mainbord over Klipper API.

cp2004, 2021-09-30:
This plugin communicates directly with the LEDs, using the Pi's SPI hardware. It's able to do this because it goes directly there, it can be fast and not complex code, allowing for more interesting animations and control over the LEDs. It addresses each LED in the strip directly, this would not be possible to do as fast as it can going through any kind of API as far as I understand how Klipper works. The [WLED](https://plugins.octoprint.org/plugins/wled) plugin is an example of how the API approach works, over network this time, but it relies on the ESP device to be able to run the effects, and so something similar would have to be implemented in Klipper. And by that point, you might as well implement the whole thing in Klipper, so no this is not something I or this plugin will be doing.

closed 2021-09-30 by cp2004