cp2004/OctoPrint-WS281x_LED_Status #168: Request to have a function test facility, not only colour but type of pattern to check effect.[Request]:

gnattycole, 2021-10-31:
### What would you like implemented?

Whilst the "RED", "GREEN" and "BLUE" test feature is really useful to check the LEDs are configured correctly, It's still hard to see what the overall effect will be when selecting colours and patterns for various status of the print without actually running the print.

### Why should this be implemented?

I guess the Plugin is a victim of its own success with so many configurable options but it isn't alway clear whether you have the desired effect until you actually run the printer through a complete cycle. Any issues with the configuration, you have to go back in and edit the setting before setting a print off again to check the LED's are behaving as expected. To be able to configure a particular hue, brightness and type of pattern and then have a "Test feature" button, would enable the setting to be verified as the desired effect before committing to saving and rebooting.

cp2004, 2021-11-09:
I guess this would not be too hard to implement - soon the 'test' feature will be full-blown live LED control...!

I'll leave it on the list, but there's not much time at the moment for me to work on most of my projects 🙁. No guarantees of when it might land in the plugin, it could be a very long time.

cp2004, 2022-01-23:
Added for 0.8.0, release candidate to follow shortly - everything that was on the list for it is complete.

gnattycole, 2022-01-25:
Nice one. I'll look forward to checking it out. Thanks for you efforts on this great plug-in.Best regards, Nat.Sent from my Galaxy

cp2004, 2022-02-04:
0.8.0 has been released (finally...!)

commit 78118b61 (2022-01-23): :sparkles: Custom test effect in utilities

Closes #168
closed 2022-02-04 by cp2004