cp2004/OctoPrint-WS281x_LED_Status #170: [Bug]: Documentation webside, broken link

Kyrunix, 2021-11-13:
### Description of the bug

Not a serious bug. But i think nobody noticed it, since.... ever? :D

On the documentation webside (gitbook.io), the link to the plugin listing is broken. Its only a typo.

Current link:
https://plguins.octoprint.org/plugins/ws281x_led_status

Correct link:
https://plugins.octoprint.org/plugins/ws281x_led_status

### Steps to reproduce

1. Open [documentation page](https://cp2004.gitbook.io/ws281x-led-status/)
2. Click on "Plugin Listing" on the top of the page

### What happened instead?

Error 404

### Plugin version

Not necessary

### OctoPrint version

Not necessary

cp2004, 2021-11-13:
Thanks, have fixed it. It could have been there forever....

I was editing some of the stuff on GitBook this morning, and they broke a lot of things recently that almost make me want to move the docs again...

Kyrunix, 2021-11-13:
Thanks for such a fast response and fix. Have a nice day. Will try this plugin today and hope it is perfect for my setup :)

Have a nice weekend 👍

closed 2021-11-13 by Kyrunix